pybamm-team/PyBaMM #1618: Importing pybamm affects style of Matplotlib figures

wigging, 2021-08-24:
**Describe the bug**
I prefer to create my own Matplotlib figures without using the `sim.plot()` function in PyBaMM. Unfortunately, importing `pybamm` will affect the style of the Matplotlib figures even if I don't use PyBaMM to create the plots.  

**To Reproduce**
In this example, `pybamm` is imported. Notice the small font size used for axis labels and the red color applied to the line plot.

```python
import matplotlib.pyplot as plt
import pybamm

_, ax = plt.subplots()
ax.plot([1, 3, 5, 3, 8.4])
ax.set_xlabel('X axis')
ax.set_ylabel('Y axis')

plt.show()
```

[image]

In this example, `pybamm` is not imported. Notice the default (larger) font size for axis labels and the blue color for the line plot.

```python
import matplotlib.pyplot as plt

_, ax = plt.subplots()
ax.plot([1, 3, 5, 3, 8.4])
ax.set_xlabel('X axis')
ax.set_ylabel('Y axis')

plt.show()
```

[image]

**Expected behaviour**
I expect the default figure style to be used when creating plots with Matplotlib.

tinosulzer, 2021-08-24:
`plt.style.use("default")` will reset to matplotlib default. Is this enough or would you prefer pybamm not affect mplstyle at all?

wigging, 2021-08-25:
That works but the reason I'm creating my own plots is to specifically not use whatever style PyBaMM uses. It seems odd to me that just importing the package would affect the plot style. Having a style applied with `sim.plot()` is fine but otherwise the defaults from Matplotlib should be used. I don't know how other users feel about this but I would prefer that `import pybamm` not affect Matplotlib's style settings.

tinosulzer, 2021-08-25:
Sure, let's get rid of it. Let's not even apply it on `sim.plot()` as people might want to specify the mplstyle without it being overridden by sim.plot (that was the reason to first import it in the import call). Could you do this? Might require tweaking the quickplot settings (fontsize, line colors) a bit so it still looks ok

wigging, 2021-08-25:
Yes, I can work on this. I looked at the contributing guidelines. After I fork the repository, which branch should I work from? Do I create a branch from `main` or from `develop`? And from my branch, to which branch do I make the pull request to?

PR #1624 (2021-09-01): Remove Matplotlib style
# Description

Remove the imposed Matplotlib style as discussed in the issue mentioned below. Also adjust the title `fontsize` when using `sim.plot()` for better readability.

Fixes #1618 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up  [... cut]
closed 2021-09-01 by tinosulzer