pybamm-team/PyBaMM #1630: Simulation fails at high current inputs

shant4, 2021-08-29:
**Describe the bug**
I used the Chen2020 chemistry and lithium_ion.DFN model. When I tried to change the current function to 25 A, (5C_rate), it took more than 30 mins to run and the output voltage curve was sharp "V". If I change the model to SPM, the simulation would fail to run. Based on the tutorial, I expected to see the voltage curve ends earlier than the simulated time frame. But it failed to have results if the current is high. Is there a range for the current? 

**To Reproduce**
Steps to reproduce the behaviour:
1. change parameter_values["Current function [A]"] = 25
2. run

**Expected behaviour**
Terminal outputs for DFN model:
At t = 0.0453447, , mxstep steps taken before reaching tout.
At t = 0.0453447, , mxstep steps taken before reaching tout.
At t = 0.0453447, , mxstep steps taken before reaching tout.
Maximum number of decreased steps occurred at t=509.09090909090907. Try solving the model up to this time only or reducing dt_max (currently, dt_max=3.5458164923888056)

Terminal outputs for SPM model:
a\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\numpy\lib\function_base.py", line 488, in asarray_chkfinite
    raise ValueError(
ValueError: array must not contain infs or NaNs

brosaplanella, 2021-08-29:
Hi @shant4! Note that the Chen2020 cell is an energy cell so in practice the maximum recommended C-rate is 1.5C. From the modelling point of view, at high C-rates you will encounter two main issues:
1. The discharge time is very short, and you will need to provide a shorter `t_eval` and shorter `dt_max`, otherwise the solver won't converge.
2. The electrolyte will deplete. For the DFN, this means it will stop when the electrolyte gets close to zero (hence it will stop before reaching the 2.5V cut-off). For the SPM, the simulations should run (once the `dt_max` is reduced) but the results will make no physical sense as they do not include any electrolyte effect.

brosaplanella, 2021-08-31:
This is a very interesting question that can be useful to other people, so I will convert it to a discussion.

closed 2021-08-31 by brosaplanella