pybamm-team/PyBaMM #1633: Suppress warnings within an experiment

ndrewwang, 2021-08-31:
Suppress the warnings that are printed when an experiment ends due to a trigger such as electrolyte depletion instead of the specified duration of the experiment.

There is no warning print necessary when the triggered outcome is desired. For example: when searching for the specific C-rate where electrolyte depletion occurs mid simulation.

Example warning message to suppress:
`Experiment is infeasible: 'event: Zero electrolyte concentration cut-off' was triggered during 'Discharge at 1.5000 C until 3.0 V'. The returned solution only contains the first 0 cycles. Try reducing the current, shortening the time interval, or reducing the period.`

tinosulzer, 2021-08-31:
Do you want to trigger a charge after depletion? Or are you happy with the output and just don't want to see the warning?

ndrewwang, 2021-08-31:
Happy with just the output, for now, for my purposes I just want to see which C-rate causes electrolyte depletion

tinosulzer, 2021-08-31:
Have you tried doing this without using `Experiment`? i.e. just change the "Current function [A]" parameter

tinosulzer, 2021-09-07:
You can suppress this level of warning by changing to `pybamm.set_logging_level("SUCCESS")` or higher https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/logger.py. We should leave this as at least a warning though, since the output of the simulation will be unexpected.

closed 2021-09-07 by tinosulzer