pybamm-team/PyBaMM #1636: Temperature effect on capacity

harish2297, 2021-09-01:
Hi, 
I want to simulate the influence of temperature on capacity. I tried changing parameters like 
'Ambient temperature [K]' = 320.0
'Initial temperature [K]' = 320.0
'Reference temperature [K]' = 320.0
But I am not able to see any difference in results. I even tried with lower temperatures. 
Could I get some help on this ? 

[image]

tinosulzer, 2021-09-01:
Can you post a script?

harish2297, 2021-09-01:
```
Code:
import pybamm

model = pybamm.lead_acid.Full()
param = model.default_parameter_values
param['Nominal cell capacity [A.h]'] = 100.0
param['Number of electrodes connected in parallel to make a cell'] = 15
param['Ambient temperature [K]'] = 320.0
param['Initial temperature [K]'] = 320.0
param['Reference temperature [K]'] = 320.0
param['Number of cells connected in series to make a battery'] = 12.0
param['Negative electrode thickness [m]'] = 0.0025
param['Positive electrode thickness [m]'] = 0.0045

experiment = pybamm.Experiment(
    [
        "Rest for 3 hours"
    ]+
    [
        "Discharge at 10 A for 30 minutes",
        "Rest for 3 hours"
    ]*5
)
sim = pybamm.Simulation(model, parameter_values=param, experiment=experiment)
sim.solve([0, 2400])
param = sim.parameter_values    
sol = sim.solution
pybamm.dynamic_plot(sim, ['Terminal voltage [V]', 'State of Charge', 'Current [A]', 'Electrolyte concentration'])
```
```
Output:
{'1 + dlnf/dlnc': 1.0,
 'Ambient temperature [K]': 320.0,
 'Anion stoichiometry': 1.0,
 'Cation stoichiometry': 1.0,
 'Cation transference number': 0.7,
 'Cell cooling surface area [m2]': 0.154,
 'Cell volume [m3]': 0.00027,
 'Current function [A]': 1.0,
 'Darken thermodynamic factor': <function darken_thermodynamic_factor_Chapman1968 at 0x7f22942f8790>,
 'Edge heat transfer coefficient [W.m-2.K-1]': 0.3,
 'Electrode height [m]': 0.114,
 'Electrode width [m]': 0.065,
 'Electrolyte conductivity [S.m-1]': <function conductivity_Gu1997 at 0x7f22942f8670>,
 'Electrolyte diffusivity [m2.s-1]': <function diffusivity_Gu1997 at 0x7f22942f8700>,
 'Electrolyte viscosity [kg.m-1.s-1]': <function viscosity_Chapman1968 at 0x7f22942f8820>,
 'Electrons in hydrogen reaction': 2.0,
 'Electrons in oxygen reaction': 4.0,
 'Hydrogen diffusivity [m2.s-1]': 4.5e-09,
 'Hydrogen reference OCP vs SHE [V]': 0.0,
 'Initial State of Charge': 1.0,
 'Initial oxygen concentration [mol.m-3]': 0.0,
 'Initial temperature [K]': 320.0,
 'Lower voltage cut-off [V]': 1.73,
 'Maximum porosity of negative electrode': 0.53,
 'Maximum porosity of positive electrode': 0.57,
 'Maximum porosity of separator': 0.92,
 'Maximum temperature [K]': 333.15,
 'Molar mass of anions [kg.mol-1]': 0.097,
 'Molar mass of cations [kg.mol-1]': 0.001,
 'Molar mass of hydrogen molecules [kg.mol-1]': 0.002,
 'Molar mass of oxygen molecules [kg.mol-1]': 0.032,
 'Molar mass of water [kg.mol-1]': 0.01801,
 'Molar volume of lead [m3.mol-1]': 1.82539682539683e-05,
 'Molar volume of lead sulfate [m3.mol-1]': 4.81717011128776e-05,
 'Molar volume of lead-dioxide [m3.mol-1]': 2.54797441364606e-05,
 'Negative current collector density [kg.m-3]': 11300.0,
 'Negative current collector specific heat capacity [J.kg-1.K-1]': 130.0,
 'Negative current collector surface heat transfer coefficient [W.m-2.K-1]': 0.0,
[... 57 lines cut ...]
 'Positive electrode thickness [m]': 0.0045,
 'Positive electrode volumetric capacity [C.m-3]': 2745000000.0,
 'Positive tab centre y-coordinate [m]': 0.147,
 'Positive tab centre z-coordinate [m]': 0.114,
 'Positive tab heat transfer coefficient [W.m-2.K-1]': 10.0,
 'Positive tab width [m]': 0.04,
 'Reference oxygen molecule concentration [mol.m-3]': 1000.0,
 'Reference temperature [K]': 320.0,
 'Separator Bruggeman coefficient (electrolyte)': 1.5,
 'Separator density [kg.m-3]': 1680.0,
 'Separator specific heat capacity [J.kg-1.K-1]': 700.0,
 'Separator thermal conductivity [W.m-1.K-1]': 0.04,
 'Separator thickness [m]': 0.0015,
 'Signed stoichiometry of cations (hydrogen reaction)': 2.0,
 'Signed stoichiometry of cations (oxygen reaction)': 4.0,
 'Signed stoichiometry of hydrogen (hydrogen reaction)': -1.0,
 'Signed stoichiometry of oxygen (oxygen reaction)': 1.0,
 'Signed stoichiometry of water (oxygen reaction)': -1.0,
 'Total heat transfer coefficient [W.m-2.K-1]': 10.0,
 'Typical current [A]': 1.0,
 'Typical electrolyte concentration [mol.m-3]': 5650.0,
 'Typical oxygen concentration [mol.m-3]': 1000.0,
 'Upper voltage cut-off [V]': 2.44,
 'Volume change factor': 1.0}
 ```

tinosulzer, 2021-09-01:
State of charge is just the integral of the current, so temperature won't affect it.
However, if you do a constant current discharge at different temperatures, you will see the cut-off voltage be hit earlier and earlier at lower temperatures due to slower kinetics. Can you try doing something like https://github.com/pybamm-team/PyBaMM/blob/develop/examples/scripts/rate_capability.py but changing temperature instead of C-rate.

Also, I just realized none of the lead-acid parameters are temperature dependent (but there should still be some effect from RT/F). The temperature dependence of diffusivity and conductivity should be: [image] I don't know what the temperature dependences of open circuit potential and exchange current density are.

harish2297, 2021-09-02:
I will try out the example, 
But in general, the usable capacity of any battery will be greater at higher temperatures. But its lifetime will reduce. Since SoC is the measure of usable capacity, will it have any effect ?
Example: if a 100Ah battery is used at a temperature higher than 25 degrees, it will have a usable capacity of say some 10 to 20% more. You can see this image for reference. I am not sure, whether it is applicable for all the battery chemistries. 
[image]

tinosulzer, 2021-09-14:
Can you do a full constant-current (say at 1C) discharge until voltage cut-off? Then you should see some temperature effect, but maybe not as pronounced as in this figure.

harish2297, 2021-09-27:
Thanks @tinosulzer. I will try it out. Can we anyhow get the available capacity of the battery after cycling it many times in this simulation ?

tinosulzer, 2021-09-27:
Why is 1C discharge not possible for lead-acid?

We don't currently have any degradation models implemented for lead-acid. So simulating many times will show no capacity change

harish2297, 2021-09-27:
1C is possible, but we have to take into account Peukert's law and total capacity will be very much less than the actual capacity. This might not be possible in real world.

tinosulzer, 2021-09-27:
I understand now. You need to find information that gives the temperature dependence of open-circuit voltage, then we can reproduce this.

harish2297, 2021-11-05:
Is there any formula related to the ocv-temperature relationship, that can be used for real-world cases ?
Real world scenarios like load patterns, measuring feasibility etc.

tinosulzer, 2021-11-05:
Simplest approximation would be to add a constant term, but you'll need to do more research to see if there are any other formulas out there. If you find the formulas I'll help you implement, but I don't have time to do a literature search on this
[Bode_Lead-acid_OCV.pdf](https://github.com/pybamm-team/PyBaMM/files/7486976/Bode_Lead-acid_OCV.pdf)

harish2297, 2021-11-06:
@tinosulzer Thanks for the information. I will check this out.
Closing this issue, for now..

closed 2021-11-06 by harish2297