pybamm-team/PyBaMM #1637: The simulation terminated before the lower cut-off voltage reached

Leonhardup, 2021-09-02:
When I changed the C-rate, the  terminal voltage can not reach the lower cut-off voltage but the simulation terminated before the setting solve time. 

[image]
 
any ideas about this issues?

Leonhardup, 2021-09-02:
the lower cut-off voltage is 2.5 V, but the voltage terminated above 3.0V.
[image]

tinosulzer, 2021-09-02:
Probably similar to #1632 
Can you print `sim.solution.termination`?

Leonhardup, 2021-09-02:
> Probably similar to #1632
> Can you print `sim.solution.termination`?

Hello, tinosulzer! I printed the sim.solution.termination as "Termination:event: Maximum positive particle surface concentration".
Do you have some ideas on my issue?

tinosulzer, 2021-09-02:
At this high C-rate (for an energy cell which is Chen2020) the simulation stops before the cut-off voltage is reached because the positive particles fill up first. But if you update to the latest version (on the develop branch) it might give you a different result

Leonhardup, 2021-09-02:
> At this high C-rate (for an energy cell which is Chen2020) the simulation stops before the cut-off voltage is reached because the positive particles fill up first. But if you update to the latest version (on the develop branch) it might give you a different result

Great, you mean it can be addressed by updating the latest Pybamm version?

tinosulzer, 2021-09-02:
Possibly, but the pypi release is not working right now, so you will have to clone the repository and install from source

Leonhardup, 2021-09-02:
> Possibly, but the pypi release is not working right now, so you will have to clone the repository and install from source

Thank you! I have cloned the repository by desktop, and how can I install from source?

tinosulzer, 2021-09-02:
https://pybamm.readthedocs.io/en/latest/install/install-from-source.html

tinosulzer, 2021-09-07:
any luck @Leonhardup ?

Leonhardup, 2021-09-07:
> any luck @Leonhardup ?

Hello, Tinosulzer! I have a problem in installing the latest version Pybamm from source in macOS . The CMake got wrong with "Bad CMake executable "". Is it installed or settings contain the correct path (cmake.cmakePath)?" When could  the latest pybamm be released?

tinosulzer, 2021-09-07:
Which step is this?

Leonhardup, 2021-09-07:
> Which step is this?

Manual installing the build-time requirements, doesn't it use VS code?

tinosulzer, 2021-09-07:
If you did `brew install sundials` and installed pybind11 you don't need to do that step. Try just moving on to `tox -e dev`

tinosulzer, 2021-09-14:
Can you try updating to the latest version that was just released on pypi?

Leonhardup, 2021-09-14:
> Can you try updating to the latest version that was just released on pypi?
Thanks for you attention! Is there a get-pip.py file to install the latest version?

Leonhardup, 2021-09-14:
> Can you try updating to the latest version that was just released on pypi?

[image]
Is it successfully installed?

tinosulzer, 2021-09-14:
What does `pip show pybamm` give you?

Leonhardup, 2021-09-14:
> What does `pip show pybamm` give you?

[image]
It is 21.8 version. Right?

tinosulzer, 2021-09-14:
Yes. Does it reach cut-off voltage now?

Leonhardup, 2021-09-14:
> Yes. Does it reach cut-off voltage now?

Yes, it can reach the lower cut-off voltage, but at some C-rate, the discharge terminal curve is a little weird, as the picture shows, the C-rate is 2C. Do you know why?
[image]

tinosulzer, 2021-09-14:
This is probably the point where the electrolyte was depleted, can you plot "Electrolyte concentration"

Leonhardup, 2021-09-25:
> This is probably the point where the electrolyte was depleted, can you plot "Electrolyte concentration"

Hello, Tinosulzer! As for a single particle model, whose parameters set do you think should be the best for simulation?

tinosulzer, 2021-09-25:
You need to choose the appropriate parameter set to match your experimental data. If you don't have any experimental data then Chen2020 is a good place to start. 

It sounds like this issue is fixed, I'm going to close it but can reopen if you have more questions

closed 2021-09-25 by tinosulzer