pybamm-team/PyBaMM #1641: Using jax solver returns error

raghuramshankar, 2021-09-07:
**Describe the bug**
Using Jax solver throws an error.

**To Reproduce**
Minimal reproduce-able example:
```
import pybamm

"""models and parameters"""
model = pybamm.lithium_ion.SPMe()

"""solver"""
sol = pybamm.JaxSolver(method='RK45', atol=1e-6, rtol=1e-6)

"""simulate cell"""
model.convert_to_format = 'jax'
sim = pybamm.Simulation(model, solver=sol)
sim.solve([0, 360]) 
```
Output:
```
/home/raghurams/repos/optimal-battery-charging/tutorials/jax_issue.py
Traceback (most recent call last):
  File "/home/raghurams/repos/optimal-battery-charging/tutorials/jax_issue.py", line 12, in <module>
    sim.solve([0, 360]) 
  File "/home/raghurams/.local/lib/python3.9/site-packages/pybamm/simulation.py", line 399, in solve
    self._solution = solver.solve(
  File "/home/raghurams/.local/lib/python3.9/site-packages/pybamm/solvers/base_solver.py", line 567, in solve
    self.set_up(model, ext_and_inputs, t_eval)
  File "/home/raghurams/.local/lib/python3.9/site-packages/pybamm/solvers/base_solver.py", line 383, in set_up
    model.y0 = init_eval(inputs)
  File "/home/raghurams/.local/lib/python3.9/site-packages/pybamm/solvers/base_solver.py", line 956, in __call__
    return self._function(0, self.y_dummy, inputs=inputs).flatten()
  File "/home/raghurams/.local/lib/python3.9/site-packages/pybamm/expression_tree/operations/evaluate.py", line 459, in evaluate
    result = self._jit_evaluate(self._constants, t, y, y_dot, inputs, known_evals)
ValueError: Non-hashable static arguments are not supported. An error occured while trying to hash an object of type <class 'list'>, [DeviceArray([[0. ],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.8],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [0.6],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ],
             [1. ]], dtype=float64)]. The error was:
TypeError: unhashable type: 'list'
```

**Expected behaviour**
Should be able to solve the experiment using `jax` solver. I read the `pybamm` documentation and saw that the extra code `model.convert_to_format = 'jax'` is necessary while using `jax` solver. Tried searching a lot but not sure if more code is required for the same.

martinjrobins, 2021-09-07:
thanks for the bug report @raghuramshankar. Can you please post the version number of the pybamm, jax and jaxlib modules? i.e.

```bash
→ pip list | grep -E 'pybamm|jax|jaxlib'
jax                           0.2.12
jaxlib                        0.1.65
pybamm                        0.4.0     /home/mrobins/git/PyBaMM
```

raghuramshankar, 2021-09-07:
Hi,

Thank you for your response. Here are my library versions:

```
pip list | grep -E 'pybamm|jax|jaxlib' 
jax                 0.2.19
jaxlib              0.1.70+cuda111
pybamm              0.2.4.post3
```

I am aware that I am not using the latest version of `pybamm`. I had some issues in installing the latest version of `pybamm` and I have reported them in https://github.com/pybamm-team/PyBaMM/issues/1608 and https://github.com/pybamm-team/PyBaMM/issues/1470.

I checked the changelog and found that there were no changes or bug fixes for the `jax_solver` since `v0.2.4`. However, I wonder if not using the latest version might affect the `jax` simulations.

martinjrobins, 2021-09-09:
I see, the `setup.py` in pybamm version 0.2.4 listed the jax dependencies as

"jax>=0.1.68",
"jaxlib>=0.1.47",

however they should have been pinned to the exact version we support. Try installing "jax=0.1.68" and "jaxlib=0.1.47" in your environment and hopefully this should fix it.

i.e.
```bash
pip install -Iv jax==1.1.68
```

raghuramshankar, 2021-09-10:
Hi, 

It seems like only `jaxlib>=0.1.60` is available to be installed via `pip`. Here is the error I got:

```
pip install -Iv jaxlib==0.1.47
Using pip 21.1.3 from /home/raghurams/repos/optimal-battery-charging/venv/lib/python3.9/site-packages/pip (python 3.9)
Non-user install because user site-packages disabled
Created temporary directory: /tmp/pip-ephem-wheel-cache-4y2oaj1z
Created temporary directory: /tmp/pip-req-tracker-ppu80go8
Initialized build tracking at /tmp/pip-req-tracker-ppu80go8
Created build tracker: /tmp/pip-req-tracker-ppu80go8
Entered build tracker: /tmp/pip-req-tracker-ppu80go8
Created temporary directory: /tmp/pip-install-3zjwzvyc
1 location(s) to search for versions of jaxlib:
* https://pypi.org/simple/jaxlib/
Fetching project page and analyzing links: https://pypi.org/simple/jaxlib/
Getting page https://pypi.org/simple/jaxlib/
Found index url https://pypi.org/simple
Looking up "https://pypi.org/simple/jaxlib/" in the cache
Request header has "max_age" as 0, cache bypassed
Starting new HTTPS connection (1): pypi.org:443
https://pypi.org:443 "GET /simple/jaxlib/ HTTP/1.1" 304 0
  Skipping link: none of the wheel's tags (cp36-none-macosx_10_9_x86_64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/b1/1c/4c17ea5e0962c66f37a26572adde35d053e580cd15a1df3987bfdcf769c0/jaxlib-0.1.60-cp36-none-macosx_10_9_x86_64.whl#sha256=eab6715820de2d3be99bcf1bd6e74fc062288eec921aec6ad061eff215fcbd23 (from https://pypi.org/simple/jaxlib/) (requires-python:>=3.6)
  Skipping link: none of the wheel's tags (cp36-none-manylinux2010_x86_64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/dd/d5/fe4aa57ac61929cfc07da094feda8b30c75a975d5fe7cdbb82ad3f3c7396/jaxlib-0.1.60-cp36-none-manylinux2010_x86_64.whl#sha256=21bbe2b328a728accb6ee6d370ec948317becc0e32c4921ff5240260fe21206a (from https://pypi.org/simple/jaxlib/) (requires-python:>=3.6)
  Skipping link: none of the wheel's tags (cp37-none-macosx_10_9_x86_64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/71/75/9709f7b192936f340e1a332e205f55a752508a5971c00be0cde20095816e/jaxlib-0.1.60-cp37-none-macosx_10_9_x86_64.whl#sha256=f74edc451531a7d12c49c18344db09183ee93c7533c58c120d5444f5d5f8da73 (from https://pypi.org/simple/jaxlib/) (requires-python:>=3.6)
  Skipping link: none of the wheel's tags (cp37-none-manylinux2010_x86_64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/79/77/57c2eff18f1cb006fab4d25b074fe9ad68f9a54e365a576651d80e4665f8/jaxlib-0.1.60-cp37-none-manylinux2010_x86_64.whl#sha256=231539132e6322a1d65329bbadcb56cbd2495f584a759437881ee4c163250247 (from https://pypi.org/simple/jaxlib/) (requires-python:>=3.6)
  Skipping link: none of the wheel's tags (cp38-none-macosx_10_9_x86_64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/ae/de/7d67082ad408d592dda52acd9ba90aecbb335ee5345e60da4dcbda386bca/jaxlib-0.1.60-cp38-none-macosx_10_9_x86_64.whl#sha256=4f0accb342d2559b2c199c9d9267ea011c543060213ab9fd03ffb75f6c0150b9 (from https://pypi.org/simple/jaxlib/) (requires-python:>=3.6)
  Skipping link: none of the wheel's tags (cp38-none-manylinux2010_x86_64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/06/80/ce3138bfea4d717e5d529dde01d9387978a3227370ee7d16214e1883249c/jaxlib-0.1.60-cp38-none-manylinux2010_x86_64.whl#sha256=7bff48c253c7bf5ad0986af8e14c981bc5f18855f6eef16f08da6ca915b092f4 (from https://pypi.org/simple/jaxlib/) (requires-python:>=3.6)
  Skipping link: none of the wheel's tags (cp39-none-macosx_10_9_x86_64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/02/11/5729188b0e731c65dd556514c572756f0e19051eec44f30c48718af9d0d4/jaxlib-0.1.60-cp39-none-macosx_10_9_x86_64.whl#sha256=3ebe62f1a389b59e0c1661a6d2c14762b68c5864bc07d7de26d81a1bd29b2060 (from https://pypi.org/simple/jaxlib/) (requires-python:>=3.6)
  Found link https://files.pythonhosted.org/packages/5d/e2/4f06a3198ecafbc92ef619a120950129910a1b456a5151799516344e5e21/jaxlib-0.1.60-cp39-none-manylinux2010_x86_64.whl#sha256=5afbf0a176f0b39e349fe3942fdd06f97093130961ae057d0e567fa247746088 (from https://pypi.org/simple/jaxlib/) (requires-python:>=3.6), version: 0.1.60
  Skipping link: none of the wheel's tags (cp36-none-macosx_10_9_x86_64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/91/a4/1c8ea78495754cf8f1c4c4ed0100486f5bf62f96f778f2fba7cddcb991ce/jaxlib-0.1.61-cp36-none-macosx_10_9_x86_64.whl#sha256=d88866813d6438cbb2f545b6983091525bdbe8633dbbc3ea7842b2e5d561ec6b (from https://pypi.org/simple/jaxlib/) (requires-python:>=3.6)
  Skipping link: none of the wheel's tags (cp36-none-manylinux2010_x86_64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/1b/49/f3e6e9c6e5833fde2be8775ed87b5712beb3620c7ea66ac917d25fbfbaa0/jaxlib-0.1.61-cp36-none-manylinux2010_x86_64.whl#sha256=0ef0449dad105144fde97aaae33585b7186c6fb80390e77771a745cc9a937357 (from https://pypi.org/simple/jaxlib/) (requires-python:>=3.6)
[... cut ...]
    return func(self, options, args)
  File "/home/raghurams/repos/optimal-battery-charging/venv/lib/python3.9/site-packages/pip/_internal/commands/install.py", line 318, in run
    requirement_set = resolver.resolve(
  File "/home/raghurams/repos/optimal-battery-charging/venv/lib/python3.9/site-packages/pip/_internal/resolution/resolvelib/resolver.py", line 136, in resolve
    raise error from e
pip._internal.exceptions.DistributionNotFound: No matching distribution found for jaxlib==0.1.47
```

martinjrobins, 2021-09-10:
hmmm, you are right, looks like they have removed the old versions of `jaxlib`. If you really want to use the jax solver, I think you'll have to upgrade to a newer pybamm version. The newer version is not yet on pypi, so you can clone this repository and install it that way. Follow the developer documentation here: https://pybamm.readthedocs.io/en/latest/install/GNU-linux.html#developer-install

raghuramshankar, 2021-09-11:
Hi,

Thanks. I was able to install the developer version and the latest `jax` and `jaxlib` versions (with gpu) and it seems to be working with some models now.

closed 2021-09-13 by tinosulzer