pybamm-team/PyBaMM #1648: Key error for cycling_ageing example #1505

PippaVossen, 2021-09-09:
**Describe the bug**
When running the example script ``` cycling_ageing.py ``` a key error is raised where ``` 'Lithium metal partial molar volume [m3.mol-1]' not found ```

How can I solve this issue?

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to ``` PyBaMM/examples/scripts/cycling_ageing.py ```
2. Run the file
3. See error below

```
File "C:\Users\pippa\Research\Code\cycling_ageing.py", line 67, in <module>
    sim.solve(solver=pb.CasadiSolver(mode="safe"))
    
KeyError: "'Lithium metal partial molar volume [m3.mol-1]' not found. Best matches are ['Inner SEI partial molar volume [m3.mol-1]', 'Outer SEI partial molar volume [m3.mol-1]', 'Lithium interstitial reference concentration [mol.m-3]']"
```

**Additional context**
A similar issue has been raised for the example script 'calendar_ageing.py' (view the discussion [here](https://github.com/pybamm-team/PyBaMM/issues/1610)), here it seemed to be a version issue between the ``` main ``` and ``` develop ``` branches. However, for me, the examples in both branches raise the same Key Error.

tinosulzer, 2021-09-09:
Which version are you using?

PippaVossen, 2021-09-09:
[image]

tinosulzer, 2021-09-09:
This script should work with v0.4.0: https://github.com/pybamm-team/PyBaMM/blob/v0.4.0/examples/scripts/cycling_ageing_yang.py

closed 2021-09-09 by tinosulzer