pybamm-team/PyBaMM #1655: Error during discretization of 1D model using PyBaMM

ShaunakAJoshi, 2021-09-10:
I'm facing an error ('Multiplication' object has no attribute 'bounds') while creating an independent 1D model on PyBaMM. Could someone please explain the possible reasons behind the error. Image attached. Thank you for your help!
[image]

brosaplanella, 2021-09-10:
Hi! Can you share a screenshot (or copy here) a bit more of your code? It would be especially useful to see the part where you define your model.

ShaunakAJoshi, 2021-09-10:
Sure. I'm following the code format set in the [Creating Models 5. notebook](https://github.com/pybamm-team/PyBaMM/blob/develop/examples/notebooks/Creating%20Models/5-half-cell-model.ipynb). The code lines for defining the model are same as that given in this example. Should I send more code ss/text? The code is quite lengthy due to the presence of 50+ parameters and 20+ variables.

Could this error occur due to inconsistent input boundary conditions? I have attempted to discretize the model with placeholder (arbitrary) boundary conditions as I am yet to figure out what the actual boundary conditions would be.

Thank you for your help!

Copying some lines from the code where geometry is defined:

model = pybamm.BaseModel()
#Defining geometry
x_s = pybamm.SpatialVariable("x_s", domain=["separator"], coord_sys="cartesian")
x_p = pybamm.SpatialVariable("x_p", domain=["positive electrode"], coord_sys="cartesian")
geometry = {
    "separator": {x_s: {"min": 0, "max": L_s}},        
    "positive electrode": {x_p: {"min": L_s, "max": L_s + L_c}},
}
param.process_model(model)
param.process_geometry(geometry)
submesh_types = {
    "separator": pybamm.MeshGenerator(pybamm.Uniform1DSubMesh),
    "positive electrode": pybamm.MeshGenerator(pybamm.Uniform1DSubMesh),
}
var_pts = {x_s: 10, x_p: 20}
mesh = pybamm.Mesh(geometry, submesh_types, var_pts)
spatial_methods = {
    "separator": pybamm.FiniteVolume(),
    "positive electrode": pybamm.FiniteVolume()
}
disc = pybamm.Discretisation(mesh, spatial_methods)
disc.process_model(model);

rtimms, 2021-09-10:
If you do `pybamm.set_logging_level("VERBOSE")` in your script after you import pybamm it should give you some more info. In particular, it looks like the error happens when trying to set the slices of the statevector (line 162 in `discretisation.py`). Changing the logging level will make it print which variable it is trying to do this for, and hopefully point to the problem.

rtimms, 2021-09-10:
Ah sorry I misread the logging message and thought it would print by variable. It looks like it thinks a variable has a multiplication in. Can you share the code where you define your variables?

ShaunakAJoshi, 2021-09-10:
Yes, I do have variables (eg: e1_c11) that are a product of two variables: porosity (e1) (which changes due to precipitation) and concentration (c11) to represent the equation:
[image]

While I tried to code the file first without defining e1_c11, the code returned that initial and boundary conditions are required for e1*c11 (e1*c11 is present in model.rhs). Could this be avoided in some way?

Relevant code lines:
C_1_1 = pybamm.Variable("Concentration of Li(+) in separator [mol.m-3]", domain = "separator")
epsilon_1 = pybamm.Variable("Porosity of separator", domain = "separator")
e1_c11 = epsilon_1*C_1_1
depsilon_1C_1_1dt = -pybamm.div(N_1_1) - R_1_1 #rate of change of porosity of separator*C_1_1 (Li(+) in separator)
model.rhs = {e1_c11: depsilon_1C_1_1dt}
model.boundary_conditions = {e1_c11: {"left": (pybamm.Scalar(0), "Neumann"), "right": (pybamm.Scalar(0), "Neumann")}}

rtimms, 2021-09-10:
The problem here is that pybamm expects the equations and boundary conditions to be for a variable, but in your formulation, they are for a variable multiplied by another variable. We solve the same equation in pybamm. The way we do it is to define a variable that is porosity times concentration, i.e. `e1_c11 = pybamm.Variable("Porosity of separator times concentration of Li(+) in separator [mol.m-3]", domain = "separator")`. Then you write your equation and boundary conditions in terms of `e1_c11`. Then presumably you either have another equation for porosity so that porosity is a variable (`e1 = pybamm.Variable("Porosity of separator", domain = "separator")`), or you have constant porosity so that `e1 = constant`. Finally, you can define `c_11 = e1_c11 / e1`.

ShaunakAJoshi, 2021-09-10:
Thanks a lot for answering @rtimms. This was quite helpful! Appreciate the effort.

closed 2021-09-10 by rtimms