pybamm-team/PyBaMM #1656: building windows wheels fails due to a change in vcpkg

martinjrobins, 2021-09-10:
- PyBaMM version - `$ python -m pip show pybamm`: 21.08
- Python version - `$ python --version`: n/a

**Describe the bug**

The `.github/workflows/build_wheels.yml` workflow fails when building the windows wheel, with the following error:

```
---------- Running CMake for idaklu solver ----------------------------------------
    -- Building for: Visual Studio 16 2019
    -- Running vcpkg install
    Error: sundials:x86-windows-static@5.7.0#1 does not have required feature klu
    
    -- Running vcpkg install - failed
    CMake Error at C:/vcpkg/scripts/buildsystems/vcpkg.cmake:827 (message):
      vcpkg install failed.  See logs for more information:
      C:\Users\runneradmin\AppData\Local\Temp\pip-req-build-_9f138ur\build\temp.win32-3.6\vcpkg-manifest-install.log
    Call Stack (most recent call first):
      C:/Program Files/CMake/share/cmake-3.21/Modules/CMakeDetermineSystem.cmake:124 (include)
      CMakeLists.txt:18 (project)
    
    
    -- Configuring incomplete, errors occurred!
```

**To Reproduce**

merge to main

**Expected behaviour**

pypi wheels are build and deployed to pypi

martinjrobins, 2021-09-10:
perhaps the problem is with the cache step:

```
Run actions/cache@v2
  with:
    path: C:\Users\runneradmin\AppData\Local\vcpkg\archives
    key: Windows-build-vckpg_binary_cache-7fbb1233e518b325804dd94a6631598e7affcb2ebdc79c1a25718b88c68ed9c3
  env:
    pythonLocation: C:\hostedtoolcache\windows\Python\3.9.6\x64
    cache-name: vckpg_binary_cache
Cache not found for input keys: Windows-build-vckpg_binary_cache-7fbb1233e518b325804dd94a6631598e7affcb2ebdc79c1a25718b88c68ed9c3
```

martinjrobins, 2021-09-10:
cache might not exist cause https://github.com/pybamm-team/PyBaMM/blob/main/vcpkg.json has changed

tlestang, 2021-09-10:
That's right - but I don't think a cache miss is a problem, a new cache would be created should the workflow be successful

https://docs.github.com/en/actions/guides/caching-dependencies-to-speed-up-workflows#using-the-cache-action 
> The cache action will attempt to restore a cache based on the key you provide. When the action finds a cache, the action restores the cached files to the path you configure.
> 
> If there is no exact match, the action creates a new cache entry if the job completes successfully. The new cache will use the key you provided and contains the files in the path directory.

I'm more surprised about this line though (within the CMake output)
```
   Error: sundials:x86-windows-static@5.7.0#1 does not have required feature klu
```

The Sundials port in [pybamm-team/sundials-vcpkg-registry](https://github.com/pybamm-team/sundials-vcpkg-registry) has the `klu` feature. Is it now somehow not finding it? Is it because of [bfed1b](https://github.com/pybamm-team/PyBaMM/commit/bfed1bfbb8858c01981a6ebe132f8e2e091e0706) ?

priyanshuone6, 2021-09-10:
Also, see this error https://github.com/pybamm-team/PyBaMM/runs/3566774115?check_suite_focus=true#step:9:114

martinjrobins, 2021-09-10:
@priyanshuone6 , what was the baseline addition in bfed1b? Looks like this was to fix the macosx builds, but I'm confused as I don't think macosx uses vcpkg?

priyanshuone6, 2021-09-10:
The workflow was failing earlier saying that it required field 'baseline' which was missing ([here](https://github.com/pybamm-team/PyBaMM/runs/3536075052?check_suite_focus=true#step:9:855))

PR #1662 (2021-09-13): Update vcpkg configuration
# Description

The current `vcpkg-configuration.json` file causes the vcpkg install to fail. `vcpkg` looks for `sundials` inside the builtin registry instead of the alternative registry at [sundials-vcpkg-registry ](https://github.com/pybamm-team/sundials-vcpkg-registry).

Explicitly specifying the builtin registry to act as the default seems to fix this. In my explorations I also updated the sundials registry, particularly the version keyword used in the various configuration files for the registry and sundials port. See

- [8269dd7b755268e4e0a8e01eb32bb1f140c643d6] [... cut]
closed 2021-09-13 by tinosulzer