pybamm-team/PyBaMM #1669: PyBaMM version 21.08 does not install

wigging, 2021-09-13:
- PyBaMM version 21.08
- Python version 3.9.5

**Describe the bug**
I'm trying to `pip install` the latest version of PyBaMM which is 21.08 but I get several errors and warnings. I'm using the Miniconda installation of Python. I did not have a problem installing the last version PyBaMM on my computer. So there is something wrong with the installation process for version 21.08.

**To Reproduce**
Steps to reproduce the behaviour:
1. `pip install pybamm`
2. See errors and warnings shown below.

```
Collecting pybamm
  Using cached pybamm-21.8.tar.gz (889 kB)
    ERROR: Command errored out with exit status 1:
     command: /opt/miniconda3/bin/python -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/4q/qh3l23cd6jzdclcq6l_bvzm0kyy0fc/T/pip-install-qx2hel0f/pybamm_e6b5ed7ff70c4a8eb7d01b9468cb7681/setup.py'"'"'; __file__='"'"'/private/var/folders/4q/qh3l23cd6jzdclcq6l_bvzm0kyy0fc/T/pip-install-qx2hel0f/pybamm_e6b5ed7ff70c4a8eb7d01b9468cb7681/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /private/var/folders/4q/qh3l23cd6jzdclcq6l_bvzm0kyy0fc/T/pip-pip-egg-info-8ujq764a
         cwd: /private/var/folders/4q/qh3l23cd6jzdclcq6l_bvzm0kyy0fc/T/pip-install-qx2hel0f/pybamm_e6b5ed7ff70c4a8eb7d01b9468cb7681/
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/4q/qh3l23cd6jzdclcq6l_bvzm0kyy0fc/T/pip-install-qx2hel0f/pybamm_e6b5ed7ff70c4a8eb7d01b9468cb7681/setup.py", line 18, in <module>
        import CMakeBuild
    ModuleNotFoundError: No module named 'CMakeBuild'
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/61/0a/f9732746658d728e22282d4b6d2ebd3e50128e6c5d00662fd668e870ccbe/pybamm-21.8.tar.gz#sha256=51af2c85110c948df176f974afe980bb4d34744a3bd03ca0cd29ba965d9d8b64 (from https://pypi.org/simple/pybamm/) (requires-python:>=3.6,<3.10). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
  Using cached pybamm-0.4.0.post1-cp39-cp39-macosx_10_9_x86_64.whl (1.4 MB)
Collecting casadi>=3.5.0
  Using cached casadi-3.5.5.post2-cp39-none-macosx_10_13_x86_64.macosx_10_13_intel.whl (21.2 MB)
Requirement already satisfied: scipy>=1.3 in /opt/miniconda3/lib/python3.9/site-packages (from pybamm) (1.7.1)
Requirement already satisfied: pandas>=0.24 in /opt/miniconda3/lib/python3.9/site-packages (from pybamm) (1.3.2)
Collecting jupyter
  Using cached jupyter-1.0.0-py2.py3-none-any.whl (2.7 kB)
Collecting pybtex
  Using cached pybtex-0.24.0-py2.py3-none-any.whl (561 kB)
Collecting autograd>=1.2
  Using cached autograd-1.3-py3-none-any.whl
Collecting anytree>=2.4.3
  Using cached anytree-2.8.0-py2.py3-none-any.whl (41 kB)
Collecting scikit-fem>=0.2.0
  Using cached scikit_fem-3.2.0-py3-none-any.whl (116 kB)
Requirement already satisfied: numpy>=1.16 in /opt/miniconda3/lib/python3.9/site-packages (from pybamm) (1.20.3)
Collecting jax==0.2.5
  Using cached jax-0.2.5-py3-none-any.whl
Requirement already satisfied: matplotlib>=2.0 in /opt/miniconda3/lib/python3.9/site-packages (from pybamm) (3.4.2)
Collecting pybamm
  Using cached pybamm-0.4.0-cp39-cp39-macosx_10_9_x86_64.whl (1.4 MB)
  Using cached pybamm-0.2.4.post3-py3-none-any.whl (520 kB)
Requirement already satisfied: six>=1.9.0 in /opt/miniconda3/lib/python3.9/site-packages (from anytree>=2.4.3->pybamm) (1.16.0)
Collecting future>=0.15.2
  Using cached future-0.18.2-py3-none-any.whl
Requirement already satisfied: kiwisolver>=1.0.1 in /opt/miniconda3/lib/python3.9/site-packages (from matplotlib>=2.0->pybamm) (1.3.1)
Requirement already satisfied: pillow>=6.2.0 in /opt/miniconda3/lib/python3.9/site-packages (from matplotlib>=2.0->pybamm) (8.3.1)
Requirement already satisfied: cycler>=0.10 in /opt/miniconda3/lib/python3.9/site-packages (from matplotlib>=2.0->pybamm) (0.10.0)
Requirement already satisfied: pyparsing>=2.2.1 in /opt/miniconda3/lib/python3.9/site-packages (from matplotlib>=2.0->pybamm) (2.4.7)
Requirement already satisfied: python-dateutil>=2.7 in /opt/miniconda3/lib/python3.9/site-packages (from matplotlib>=2.0->pybamm) (2.8.2)
Requirement already satisfied: pytz>=2017.3 in /opt/miniconda3/lib/python3.9/site-packages (from pandas>=0.24->pybamm) (2021.1)
Collecting meshio>=4.0.4
  Using cached meshio-5.0.0-py3-none-any.whl (163 kB)
Collecting jupyter-console
  Using cached jupyter_console-6.4.0-py3-none-any.whl (22 kB)
Collecting ipykernel
  Using cached ipykernel-6.4.1-py3-none-any.whl (124 kB)
Collecting nbconvert
  Using cached nbconvert-6.1.0-py3-none-any.whl (551 kB)
Collecting qtconsole
  Using cached qtconsole-5.1.1-py3-none-any.whl (119 kB)
Collecting ipywidgets
  Using cached ipywidgets-7.6.4-py2.py3-none-any.whl (121 kB)
Collecting notebook
  Using cached notebook-6.4.3-py3-none-any.whl (9.9 MB)
[... 71 lines cut ...]
Collecting webencodings
  Using cached webencodings-0.5.1-py2.py3-none-any.whl (11 kB)
Collecting qtpy
  Using cached QtPy-1.11.0-py2.py3-none-any.whl (58 kB)
Installing collected packages: pyrsistent, attrs, pyzmq, nest-asyncio, jupyter-core, jsonschema, entrypoints, webencodings, packaging, nbformat, MarkupSafe, jupyter-client, testpath, pandocfilters, nbclient, mistune, jupyterlab-pygments, jinja2, defusedxml, debugpy, bleach, terminado, Send2Trash, prometheus-client, nbconvert, ipykernel, argon2-cffi, notebook, widgetsnbextension, qtpy, jupyterlab-widgets, qtconsole, meshio, jupyter-console, ipywidgets, future, scikit-fem, jupyter, casadi, autograd, anytree, pybamm
Successfully installed MarkupSafe-2.0.1 Send2Trash-1.8.0 anytree-2.8.0 argon2-cffi-21.1.0 attrs-21.2.0 autograd-1.3 bleach-4.1.0 casadi-3.5.5.post2 debugpy-1.4.3 defusedxml-0.7.1 entrypoints-0.3 future-0.18.2 ipykernel-6.4.1 ipywidgets-7.6.4 jinja2-3.0.1 jsonschema-3.2.0 jupyter-1.0.0 jupyter-client-7.0.2 jupyter-console-6.4.0 jupyter-core-4.7.1 jupyterlab-pygments-0.1.2 jupyterlab-widgets-1.0.1 meshio-5.0.0 mistune-0.8.4 nbclient-0.5.4 nbconvert-6.1.0 nbformat-5.1.3 nest-asyncio-1.5.1 notebook-6.4.3 packaging-21.0 pandocfilters-1.4.3 prometheus-client-0.11.0 pybamm-0.2.4.post3 pyrsistent-0.18.0 pyzmq-22.2.1 qtconsole-5.1.1 qtpy-1.11.0 scikit-fem-3.2.0 terminado-0.12.1 testpath-0.5.0 webencodings-0.5.1 widgetsnbextension-3.5.1
```

According to `pip list`, the version of PyBaMM that is installed appears to be `0.2.4.post3`.

Running a PyBaMM example gives the following module not found error.

```
Traceback (most recent call last):
  File "/Users/g72/Desktop/ORNL/MegaPack/pybamm-examples/example1.py", line 2, in <module>
    import pybamm
  File "/opt/miniconda3/lib/python3.9/site-packages/pybamm/__init__.py", line 105, in <module>
    from .expression_tree.operations.evaluate import (
  File "/opt/miniconda3/lib/python3.9/site-packages/pybamm/expression_tree/operations/evaluate.py", line 13, in <module>
    import jax
ModuleNotFoundError: No module named 'jax'
```

**Expected behaviour**
Expected behavior is to install PyBaMM version 21.08 with pip.

tinosulzer, 2021-09-13:
Was uploaded to pypi accidentally, we are working to remove

closed 2021-09-13 by tinosulzer

tinosulzer, 2021-09-14:
does it work now?

wigging, 2021-09-14:
Yes it works now. Looks like pip installs version `21.8.post1`.

sankhadeeps, 2022-05-22:
Hello,
I am a graduate student, new to mathematical coding of batteries. My research interest is in Lithium-ion battery modeling,
Presently, I have followed the Pybamm installation instructions & videos for the Windows system and accordingly configured the virtual environment in WSL.
However, I am not able to run and plot the battery models. Screenshots are attached for reference. Please suggest.

[image]
[Command Line.txt](https://github.com/pybamm-team/PyBaMM/files/8750404/Command.Line.txt)

tinosulzer, 2022-05-22:
try `sim.solve([0,3700])`. See the error message at the bottom of your screenshot. I suggest looking at some tutorials on pybamm.org/training