pybamm-team/PyBaMM #1690: Automatically generate debugging file

tinosulzer, 2021-09-24:
When running long experiments, it can be hard to debug if the simulation fails a long way in.

If the simulation fails, automatically generate a debugging file with all the information necessary to recreate the failing simulation in just one cycle:
- model including options
- experiment
- parameter values
- discretization
- solver
- starting solution

Could be saved to `pybamm.settings.debug_filename` which defaults to `debug.txt` (and add `debug.txt` to the gitignore)

arunavabasu-03, 2022-03-13:
hey, @tinosulzer I would l like to work on this issue.😀

rtimms, 2022-03-14:
[This PR](https://github.com/pybamm-team/PyBaMM/pull/1880) might be relevant

tinosulzer, 2022-03-14:
The first step for this is to have a nice `__repr__` for the various objects, which shows which arguments and kwargs were passed in, e.g. `pybamm.lithium_ion.DFN(options={"thermal": "x-full"})` (not the whole options dict). scikit-learn does this pretty nicely, see https://github.com/scikit-learn/scikit-learn/blob/f71c0313142c4e5f2f35a0021c36075cf8dba611/sklearn/utils/_pprint.py

This is definitely not a simple first issue! But it doesn't require super in-depth knowledge of the code base at least

jaskiratsingh2000, 2023-02-20:
Is it still opened?

closed 2024-02-19 by tinosulzer