pybamm-team/PyBaMM #1692: Simulate standby usage

harish2297, 2021-09-27:
In most of the UPS industries, batteries are being used in standby conditions, where the battery will be float charged for the maximum amount of time and discharged only when there is a power cut until generator is turned on. 
Can anyone suggest how to simulate this load pattern ?

tinosulzer, 2021-09-27:
So you want to simulate a long rest period followed by a discharge? Or something else?

harish2297, 2021-09-27:
It is not like a long rest period, as here the charger will be connected to the battery always. It will be float charging the battery with say 0.2A constantly to avoid self discharge. Discharge will happen only during a power cut for say 10 to 15 minutes.

tinosulzer, 2021-09-27:
Presumably "float charge" is a constant voltage hold rather than a constant current charge. How about something like this
```python
float1 = 14.5/6
float2 = 13.5/6
pybamm.Experiment([
f"Charge at 0.2A until {float1}V",
f"Hold at {float1}V until 0.05A",
f"Hold at {float2}V for 10h",
"Discharge at 1C for 10 minutes"
])
```

closed 2021-11-04 by tinosulzer

harish2297, 2021-11-06:
Sorry for the delayed reply. I will try this
Thanks for the response.