pybamm-team/PyBaMM #1693: Documentation does not build

brosaplanella, 2021-09-27:
The documentation in readthedocs does not build and all the pages that should be autogenerated only show the header, see [this example](https://pybamm.readthedocs.io/en/latest/source/expression_tree/operations/jacobian.html).

rtimms, 2021-09-27:
lol trust this to happen when we have a workshop on

brosaplanella, 2021-09-27:
Very nicely spotted by some attendees 😃

Saransh-cpp, 2021-09-27:
I was going through the [logs](https://readthedocs.org/projects/pybamm/builds/14832231/) of the latest build and it is full of `jax` and `partial` errors (which are being treated as warnings, hence the docs are passing). For example the `Jacobian` class gives the following error in the logs -

```
WARNING: autodoc: failed to import class 'Jacobian' from module 'pybamm'; the following exception was raised:
Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/pybamm/envs/latest/lib/python3.8/site-packages/sphinx/ext/autodoc/importer.py", line 154, in import_module
    __import__(modname)
  File "/home/docs/checkouts/readthedocs.org/user_builds/pybamm/checkouts/latest/pybamm/__init__.py", line 235, in <module>
    from .solvers.jax_bdf_solver import jax_bdf_integrate
  File "/home/docs/checkouts/readthedocs.org/user_builds/pybamm/checkouts/latest/pybamm/solvers/jax_bdf_solver.py", line 69, in <module>
    @jax.partial(jax.custom_vjp, nondiff_argnums=(0, 1, 2, 3))
AttributeError: module 'jax' has no attribute 'partial'
```

And the docs are building fine locally, so I don't know if these `jax` warnings are the root of this failure

tinosulzer, 2021-09-27:
Ah the jax version isn't pinned in `requirements-docs`

PR #1694 (2021-09-28): Pin `jax` and `jaxlib` versions - fix docs deployment
# Description

Fix the latest docs deployment

Fixes #1693 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No styl [... cut]
closed 2021-09-28 by tinosulzer