pybamm-team/PyBaMM #1695: Temperature dependence in SEI models

tinosulzer, 2021-09-28:
- For the EC reaction limited, reaction limited, and interstitial diffusion limited, multiply the exponent by the following factor (dimensionless 1/T):
```python
T = variables["Negative electrode temperature"]
prefactor = 1 / (1 + self.param.Theta * T)
```
- Add arrhenius dependency to diffusivity in ec reaction limited and solvent diffusion limited (I don't know what the arrhenius factor should be)
- I don't think there is any temperature dependence

soma2000-lang, 2022-02-12:
Working on this issue@tinosulzer

soma2000-lang, 2022-03-13:
@tinosulzer  I will be submitting a proposal for the idea "Documentation" ,for GSOC22 under pybamm.
I have a doubt about how to know what the arrhenius dependency should be.
It would be great to be mentored by you and  other mentors

soma2000-lang, 2022-03-18:
@tinosulzer Can you please tell me where SEI model file is, I am having difficulty in finding them.

tinosulzer, 2022-03-26:
https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/submodels/interface/sei/sei_growth.py

soma2000-lang, 2022-03-27:
@tinosulzer  thanks

PR #2104 (2022-07-27): SEI on cracks
# Description

Adds a new option "SEI on cracks". If set to "true", whichever SEI model is specified by the "SEI" option is run twice; once on the electrode particle surface, and then a second time on the additional surface created by the particle cracking created by the "particle mechanics": "swelling and cracking" option.

Breaking change: "Loss of lithium to SEI on cracks [mol]" is now a summary variable, so a "particle mechanics" submodel must be set, even if it's "none". The examples have been updated to account for this.

Fixes #1695 (in full), #1807 (in part), #2039 (in fu [... cut]
closed 2022-07-27 by tinosulzer