pybamm-team/PyBaMM #1697: Expose parameter management function

rtimms, 2021-09-29:
The parameter [command line interface](https://pybamm.readthedocs.io/en/latest/source/parameters_cli.html?highlight=cli#pybamm.parameters_cli.add_parameter) requires users to use bash. However, there are some python functions defined in `parameters_cli.py` that we could expose so that e.g. instead of 
```
%%bash
pybamm_edit_parameter lithium_ion
```
you can just do 
```python3
pybamm.edit_parameter(["lithium_ion"])
```
directly in python.

rtimms, 2021-09-29:
Update [this notebook](https://github.com/pybamm-team/PyBaMM/blob/develop/examples/notebooks/parameterization/parameter-management.ipynb) too

soma2000-lang, 2022-02-12:
Working on this issue

soma2000-lang, 2022-03-18:
@Saransh-cpp @tinosulzer ,I will have to edit the edit_parameter function in the cli_parameter.py

rtimms, 2022-09-23:
You can already do 
```python3
from pybamm.parameters_cli import edit_parameter
edit_parameter(["lithium_ion"])
```
so it's just a matter of importing `edit_parameter` in the  `__init__.py` in the `pybamm` directory. 

We should also add captions to the [training](https://www.pybamm.org/training) videos where we talk about the CLI showing the new method (which will be useful for windows users).

tinosulzer, 2022-10-05:
`edit_parameter` function will be removed now that all parameters are specified by single python files, which makes it pretty intuitive how to specify your own

closed 2022-10-05 by tinosulzer