pybamm-team/PyBaMM #1699: jax & jaxlib: remove version pinning?

DavidMStraub, 2021-09-30:
- PyBaMM version - v21.08
- Python version -  all

**Describe the bug**
Currently, `jax` and `jaxlib` are version-pinned install requirements. Since PyBaMM is a library, in my opinion pinning should be avoided, as it can easily lead to incompatible version requirements (e.g. when any other library pins the version, too). Can you consider replacing it by a version range?

martinjrobins, 2021-09-30:
For the Jax solver that Pybamm includes, we found that the internal jax api that this uses was changing too rapidly, so we've been pinning the version of jax to ensure that we can still support this, periodically updating the jax version and updating the pybamm solver to match. I can't see a better way to do this, except for perhaps making the jax solver an optional dependency, this would alleviate the problem as long as a user didn't actually want to use the jax solver. But I'd be happy to discuss alternatives, e.g. I'd be happy to support a range, but suspect this would be too narrow to be practically useful

DavidMStraub, 2021-09-30:
I understand. They also don't seem to follow a useful versioning approach where the version number would give any indication about API changes. Making it an optional dependency is worth a thought IMO given that it doesn't work on Windows & Mac anyway, but also fine to leave it as is.

martinjrobins, 2021-09-30:
yea, I'm leaning towards making jax optional for linux, I'll create another issue for this

closed 2021-09-30 by martinjrobins