pybamm-team/PyBaMM #1700: Simulation with explicit experiment and irreversibe plating fails

DavidMStraub, 2021-09-30:
- PyBaMM version: Current `develop`
- Python version: 3.8.5

**Describe the bug**

Error:

```
RuntimeError: .../casadi/core/function_internal.hpp:1257: Input 1 (i1) has mismatching shape. Got 141-by-1. Allowed dimensions, in general, are:
 - The input dimension N-by-M (here 122-by-1)
 - A scalar, i.e. 1-by-1
 - M-by-N if N=1 or M=1 (i.e. a transposed vector)
 - N-by-M1 if K*M1=M for some K (argument repeated horizontally)
 - N-by-P*M, indicating evaluation with multiple arguments (P must be a multiple of 1 for consistency with previous inputs)
```

**To Reproduce**
Steps to reproduce the behaviour:

```python
import pybamm

options = {"lithium plating": "irreversible"}
model = pybamm.lithium_ion.SPMe(options=options)

experiment = pybamm.Experiment(
    [
        "Discharge with 1 C for 30 minutes",
        "Rest for 30 minutes",
    ]
)

solver = pybamm.CasadiSolver("fast with events")

parameter_values = model.default_parameter_values.copy()
parameter_values.update(
    {
        "Lithium metal partial molar volume [m3.mol-1]": 1.30e-5,
        "Exchange-current density for plating [A.m-2]": 0.001,
        "Initial plated lithium concentration [mol.m-3]": 0,
        "Typical plated lithium concentration [mol.m-3]": 1000,
    },
    check_already_exists=False
)

sim = pybamm.Simulation(model, experiment=experiment, solver=solver, parameter_values=parameter_values)
sim.solve()
```

**Expected behaviour**

No error

**Additional context**

- Possibly related to #1547.
- Using only a single step in the experiment removes the error
- Removing lithium plating removes the error

brosaplanella, 2021-09-30:
The issue is with the x-averaging of the plating model. I will open a separate issue, but for the moment you can replace the line where you define the model by
```python3
model = pybamm.lithium_ion.SPMe(build=False)
model.submodels["negative lithium plating"] = pybamm.lithium_plating.IrreversiblePlating(model.param, "Negative", False)
model.build_model()
```

DavidMStraub, 2021-09-30:
Thanks! Fix worked (I had to remove the `"Negative"` from `IrreversiblePlating`).

tinosulzer, 2021-09-30:
It should still work with the x-averaged form, there is just a bug in how multiple steps are set up. I thought I had fixed it but don't know where that branch went

PR #1707 (2021-11-12): fix experiment bugs
# Description

Fix some bugs related to how an experiment is set up

Fixes #1700 
Fixes #1702 
Fixes #479 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

 [... cut]
closed 2021-11-12 by tinosulzer