pybamm-team/PyBaMM #1714: Rename "tortuosity" to "transport efficiency"

tinosulzer, 2021-10-04:
### Discussed in https://github.com/pybamm-team/PyBaMM/discussions/1712

<div type='discussions-op-text'>

<sup>Originally posted by **dion-w** October  3, 2021</sup>
Hello everyone,

i got confused when i recently looked into the code and found that your calculated tortuosities are below 1. As i understood  the concept of tortuosity (or more precisely "tortuosity factor" coming from [this paper](https://doi.org/10.1016/j.coche.2016.02.006)) this property represents the effective transport length in the conductive phase of a medium with obstacles in it. A Tortuosity factor of 1 represents the shortest possible connection between two points, whereas a tortuosity bigger than 1 is a pathway with obstacles in it. 

I think your equations are still correct (i don't know for sure), but it's very confusing to see the property you call "tortuosity" being smaller than 1, as this would mean the transport pathway in your electrode is even shorter than a direct connection. 

My suggestion would be to change the name of the variable to something like "effective transport coefficient" to not confuse it with the concept of tortuosity as a lot of people in the battery community work with tortuosities bigger than 1.

So what do you think about it? 

I'm coming from the battery community and my knowledge is mostly based on my research at [my institute](https://www.iam.kit.edu/et/english) as well as [this paper](https://iopscience.iop.org/article/10.1149/2.0731814jes) and [this publication](https://doi.org/10.1016/j.coche.2016.02.006). Contact details about me can be found [here](https://www.iam.kit.edu/et/english/mitarbeiter_5143.php). 

Best regards,
Dion Wilde
</div>

tinosulzer, 2022-02-22:
Yes please!

soma2000-lang, 2022-03-12:
@tinosulzer  Should all the occurrences of tortuosity" be changed to"transport efficiency" in the submodels folder

tinosulzer, 2022-03-13:
There's already a PR open for this

tinosulzer, 2022-03-13:
Closed by #1950

closed 2022-03-13 by tinosulzer