pybamm-team/PyBaMM #1776: How to simulate a battery with given macroscopic 3d geometry

ehtec, 2021-11-03:
Is there a way to simulate a lithium-ion battery with a given arbitrary 3d macroscopic geometry (e.g. cylindrical, rectangular-shaped) and dimensions? I found a few sources that suggest it is possible with the current PyBaMM functionalities:

https://pybamm.readthedocs.io/en/latest/source/spatial_methods/finite_volume.html

https://github.com/pybamm-team/PyBaMM/blob/269629fc83774dbdea2288f466195b8805e72df2/pybamm/geometry/battery_geometry.py#L7

https://github.com/pybamm-team/PyBaMM/blob/269629fc83774dbdea2288f466195b8805e72df2/pybamm/meshes/scikit_fem_submeshes.py#L11

I was also able to use the "arbitrary" geometry of the "lumped" thermal submodel with the DFN model and calculate area and volume according to my geometry, but I was not able to find any usage example on how to adapt the rest of the model to an arbitrary macroscopic 3d geometry.

Could you give me a hint how to approach this problem?

rtimms, 2021-11-03:
At the moment you can only do single layer pouch cells in 3D by doing e.g.
```python3
dfn = pybamm.lithium_ion.DFN(
    {"current collector": "potential pair", "dimensionality": 2, "thermal": "x-lumped"},
)
```
this will solve a "2+1D" model where the temperature doesn't vary across the thickness of the pouch but does vary in the other two dimensions. See [this paper](https://iopscience.iop.org/article/10.1149/1945-7111/abbce4) for details.

The arbitrary option in the lumped thermal model uses an electrochemical model coupled to a lumped thermal model which has a cooling term that has a factor area/volume in it. The point of this option is to let the user specify the surface area available for cooling and volume for their cell, rather than letting the code calculate it from the other dimensions assuming a pouch cell.

ehtec, 2021-11-03:
Thank you for the quick reply. For my use case, I need to simulate macroscopic 3d lithium ion cells of cylindrical and rectangular-shaped geometry using the DFN model with the thermal, SEI, lithium plating and swelling + cracking submodels. What modifications will I have to make to PyBaMM to add support for my geometries?

rtimms, 2021-11-03:
You would need to add a new geometry for the cylindrical cell and update the spatial methods to work in cylindrical coordinates. I have an issue open #1311 to add some homogenised cylindrical models, which I'm hoping to work on soon, but it has been open for a year haha... other things have gotten in the way.

Do you actually need spatially resolved simulations? I.e. do you care about, say, how SEI thickness varies with macroscale r and theta? Or do you just want a 1D model that represents a cylindrical cell and tells you how quantities vary across the thickness of the electrode/separator/electrode sandwich? If it is the latter, the Chen2020 set is for a cylindrical cell, so would be a good starting point to see how to add your parameters.

ehtec, 2021-11-03:
Yes, I need spatially resolved simulations, ideally I should be able to create them for an arbitrary geometry. I would not expect the SEI thickness to vary with theta, but with r, eventually. And also the temperature will be dependent on r. I need to write a submodel that produces T(r, theta, z), or T(x, y, z), also with external cooling options, and swelling / SEI / lithium plating variables in a similar way.

I will have a look at the Chen2020 set, thanks.

rtimms, 2021-11-03:
A good starting point might be to just get the cylindrical geometry and spatial methods set up to solve something easy like the heat equation with a prescribed source term. Then once that is all set up you can figure out how to couple the electrical and thermal problem at the cell scale to the local electrochemistry. Solving something like a DFN at each point in space in the jelly roll might be computationally expensive, and you might find the homogenised models described in #1311 are enough.

We also did some work coupling pybamm with openpnm to solve jelly roll problems, see [here](https://iopscience.iop.org/article/10.1149/1945-7111/aba44b) and [here](https://iopscience.iop.org/article/10.1149/1945-7111/abd44f).

rtimms, 2021-11-03:
But the rectangular model should work already for a single-layer pouch. There are ways you can make the single-layer model behave like it represents a multi-layer pouch, e.g. by scaling the parameters as in [this paper](https://www.sciencedirect.com/science/article/pii/S0378775320300902), or by doing homogenisation (hoping to add this in the future, will open an issue now so you can track progress).

ehtec, 2021-11-03:
Thanks, I will take some time to have a look through all of this information.

tinosulzer, 2021-11-09:
Can we close this now we have #1777?

ehtec, 2021-11-15:
Ok I close this now... I will reopen in case it is necessary.

closed 2021-11-15 by ehtec