pybamm-team/PyBaMM #1778: Add option to include heat of mixing

rtimms, 2021-11-03:
Add an option to include heat of mixing (irreversible) and use the expression for the reversible heating from [this paper](https://arxiv.org/abs/2104.01590)

srinathkp, 2022-04-14:
Hey @rtimms , I'm interested in PyBamm for GSoC22 and I wanna work on this issue. I went through the abstract and few parts of the paper that explains the equations of importance- Eq. (26) and (27) But, I don't have a background on battery physics, so it's a bit complex for me to identify where the heat of mixing comes into picture in the equation and the corresponding file/class that needs changing. Can you help me with that

brosaplanella, 2022-04-14:
You need to incorporate the extra terms in `base_thermal.py` (in `_get_standard_coupled_variables` l90). You basically need to define the extra terms, which correspond to equations (20)-(21) of the [published version](https://www.sciencedirect.com/science/article/abs/pii/S0013468621011993) (let me know if you can't access it). Note that you need to handle full cell and half-cell differently (like some other terms already do).

brosaplanella, 2023-03-06:
# Hackathon info
This will be one of the projects for the first PyBaMM Hackathon. Here you can find some information (summarising other points made on this issue) which should be a useful starting point. Note that this post will evolve, even during the Hackathon day, so make sure to check it regularly.

brosaplanella, 2023-03-29:
https://github.com/abillscmu/git-practice

@smitasahu2 @abillscmu @Afgr1087 @ikorotkin @geslina

brosaplanella, 2023-03-29:
https://github.com/abillscmu/PyBaMM

PR #2837 (2024-06-21): Issue 1778 - heat of mixing
# Description
implemented heat of mixing

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context. List any dependencies that are required for this change.

Fixes #1778 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [ ] [... cut]
closed 2024-06-21 by brosaplanella