pybamm-team/PyBaMM #1787: How to add a custom parameter function that requires variables to be evaluated

ehtec, 2021-11-04:
I have written some smooth code that can train a model using machine learning methods (for example random forest regression) from supplied measured parameter values from a CSV file and save it to a PKL file. The code works with any number of independent variables, as long as there are enough measurements. I want to add this code to PyBaMM so that parameter value functions can easily be loaded from the PKL file. Here is the class taking care of that:

class PybammStandardParameterClass:

    def __init__(self, pkl_filename):

        with open(pkl_filename, 'rb') as modelfile:
            self.model = pickle.load(modelfile)

    def pybamm_standard_parameter_function(self, params):

        print("params: {0}".format(params))

        params = np.array([[params]])

        res = float(self.model.predict(params))

        return res

I can call the function successfully using

    my_pspc = PybammStandardParameterClass(PATH_TO_PKL)

    my_pspc.pybamm_standard_parameter_function(0.5)

for example.

If I try now to use this parameter, for example like this:

    pv = pb.ParameterValues({'Negative electrode volume change': my_pspc.pybamm_standard_parameter_function})

I get an error when running the simualtion:

    TypeError: float() argument must be a string or a number, not 'Division'

When I look at the output from my debug message, I see this:

    params: integral dr ['negative particle'](Negative particle concentration) / integral dr ['negative particle'](broadcast(1.0))

It looks like Pybamm is supplying my function with some abstract expression tree values instead of numerical values, and the model from the PKL file does not know how to use these abstract expressions, of course.

I played around with the `pb.Function` class, trying to fix this. I also had a look at this file, which has a similar functionality

https://pybamm.readthedocs.io/en/latest/_modules/pybamm/expression_tree/interpolant.html#Interpolant

But I could not figure out a working solution.

How do I need to handle the input parameter inside my function to be able to evaluate it?

tinosulzer, 2021-11-04:
pybamm has its own symbolic representation so your model would need to be able to be expressed in terms of these functions. This is necessary for conversion to casadi and jacobian calculations. Can you give an example of what `self.model` looks like?

ehtec, 2021-11-04:
```python
import numpy as np
from sklearn.ensemble import RandomForestRegressor
import pickle
import os

def train_model(csvpath, pkl_filename, choice_number=None):
    input_data = np.genfromtxt(csvpath, delimiter=',')

    if bool(choice_number):
        input_data = input_data[np.random.choice(input_data.shape[0], choice_number, replace=False)]

    regressor = RandomForestRegressor(n_estimators=100, random_state=0)

    regressor.fit(input_data[:, 0:-1], input_data[:, -1])

    with open(pkl_filename, 'wb') as file:
        pickle.dump(regressor, file)

if __name__ == "__main__":

    csvpath = input("Path to CSV data file: ")

    bare_path = os.path.splitext(csvpath)[0]

    default_pkl_filename = bare_path + '.pkl'

    pkl_filename = input("Path to output model (leave empty to use {0}): ".format(default_pkl_filename))

    if not bool(pkl_filename):
        pkl_filename = default_pkl_filename

    choice_number = input("Number of values from the CSV to use (leave empty to use all): ")

    if not bool(choice_number):
        choice_number = None

    else:
        choice_number = int(choice_number)

    train_model(csvpath, pkl_filename, choice_number=choice_number)
```

This is how the model file is generated.

ehtec, 2021-11-04:
I was able to get a PyBaMM-like output with this code

```python
class PybammFunctionConverter(pb.Function):

    def __init__(self, func, child):
        super().__init__(func, child)

thefunc = lambda x: PybammFunctionConverter(my_pspc.pybamm_standard_parameter_function, x)
```

Calling `thefunc` with a number

```python
thefunc(0.9)
```

resulted in

```
PybammFunctionConverter(0x7662feb79d48a836, function (pybamm_standard_parameter_function), children=['0.9'], domain=[], auxiliary_domains={})
```

Evaluating

```python
thefunc(0.1).evaluate()
```

resulted in

```
params: 0.1

0.018294096502836133
```

But also when assigning `thefunc` as parameter, it did not work. It resulted in an error because in the end, only `NaN` values arrived as arguments for `model.predict` for some reason.

tinosulzer, 2021-11-04:
The problem isn't the pybamm representation (since we can always write a new one), it's the casadi representation. For the interpolant, it works because we convert to casadi's interpolant class. I don't know how scikit-learn represents its random forest, maybe it's possible to convert to a casadi format?

tinosulzer, 2021-11-04:
It's times like this I wish we had written pybamm in julia

tinosulzer, 2021-11-04:
Relevant functions: https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/expression_tree/operations/convert_to_casadi.py
and tests: https://github.com/pybamm-team/PyBaMM/blob/develop/tests/unit/test_expression_tree/test_operations/test_convert_to_casadi.py

ehtec, 2021-11-04:
I am currently trying to get the predict function into bare python code using this

https://github.com/konstantint/SKompiler

I will send the result tomorrow it looks like this needs some time to run. Is it possible to convert a decision tree consisting of if-else statements to casadi?

ehtec, 2021-11-04:
By the way, does the `Interpolant` support 2d interpolation? Because the `interp2d` function is used in the code, although the docstring clearly states the class is there to interpolate data in 1D.

tinosulzer, 2021-11-04:
(Some) if-else statements can be handled via heavisides:
```python
if x > 1:
    a = 2
else:
    a = 3
```
is equivalent to `a = (x>1)*2 + (x<=1)*3`.

The 2D interpolant is only half done and untested. I think it should be possible to get that finished fairly easily, as long as casadi's interpolant can do 2D. See #1101

ehtec, 2021-11-05:
It looks like Casadi can do 2D. I'll have a look into this as well.

http://casadi.sourceforge.net/api/html/de/dbe/group__interpolant.html

ehtec, 2021-11-05:
[regressor_code_compiled.txt](https://github.com/pybamm-team/PyBaMM/files/7481944/regressor_code_compiled.txt)

Here is the "compiled" model code. Is the best way now to convert this to a Casadi expression tree to write a regex replacement script to convert the if-else statements like you suggested in your message above or can you suggest a better way?

tinosulzer, 2021-11-05:
Can't think of a better way than regex. It might also end up being slow to solve the model, since that will be a very large expression, but I guess there's only one way to find out

ehtec, 2021-11-05:
Ok I'll try that. But first I will do the 2D interpolant I guess... I suppose I will need to edit this file to add support for the 2D interpolant

https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/expression_tree/operations/convert_to_casadi.py

Where is the parsing of the [data] tag happening? Should I add support for 2d in the `Interpolant` class or is something else needed as well? I remember there was already `interp2d` used in this class, you said it was half-finished. Can you give me more details on what is already done and what is still missing? Then I will complete that

tinosulzer, 2021-11-05:
Things to do are 
1. process 2D data or add an example on how to set 2D data
2. check that a model with a 2D interpolant can be converted to casadi and solved

1. [data] tag processed here
https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/parameters/parameter_values.py#L309-L325
and then here
https://github.com/pybamm-team/PyBaMM/blob/93ead88da806438b1268f7305776342241b1d841/pybamm/parameters/parameter_values.py#L639-L662

Bear in mind it is harder to provide a csv file for 2D data than for 1D data. For 1D data it is just a two-column csv. How do we do 2D? We could just leave it to the user to define the `pybamm.Interpolant` in python.

2. Just needs editing the convert_to_casadi file and then adding a test to make sure it works

ehtec, 2021-11-05:
May I ask why you raise a `ValueError` here on line 53?

https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/expression_tree/interpolant.py

If `x` is multi-dimensional (which is the case when we have two independent variables) `y` will still be one-dimensional right?

tinosulzer, 2021-11-05:
No, if `x[0]` has length m and `x[1]` has length n then `y` will be 2D with shape (m,n) (unless it is reshaped to a (mxn,) vector, but then it is unclear whether that should be transformed to a 2D array row-first or column-first)

tinosulzer, 2021-11-05:
this is the test https://github.com/pybamm-team/PyBaMM/blob/7b748f8f0122c8edf3096ad5b9fee749a551b0c9/tests/unit/test_expression_tree/test_interpolant.py#L79-L89

ehtec, 2021-11-05:
Ah I get now what you do. And I understand now why you said it is harder to provide a CSV file in 2D. I think this should be changed to the way I did it for the random forest regression. The user should be able to supply a CSV with 3 columns (x1, x2, y) and the Interpolant should be able to work based on that. Using a two-dimensional approach instead of the CSV file is a nice approach in theory, but it doesn't work in practice because nobody will be able to supply a CSV file with measurements in a perfect grid. You will rather get a list of measurements with differing distances for both x1 and x2.

ehtec, 2021-11-05:
Ok I think I found a solution. I take the non-uniform input data and use my Random Forest Regressor on it, and then I build a uniform grid with `linspace` and use it as argument for `model.predict` to get the result as a uniform grid which I can pass to the Interpolant.

Most of the stuff is done but I still have a problem with the Casadi conversion. This is what I am passing to the casadi interpolant (text from my console output):

```
CONVERTED CHILDREN: [MX(y_diff[641:1241]), MX((298.15+y_diff[1301])[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0])]
```

And I get this error:

```python
NotImplementedError: Wrong number or type of arguments for overloaded function 'Function_call'.
  Possible prototypes are:
    call(self,dict:DM,bool,bool)
    call(self,[DM],bool,bool)
    call(self,[SX],bool,bool)
    call(self,dict:SX,bool,bool)
    call(self,dict:MX,bool,bool)
    call(self,[MX],bool,bool)
  You have: '(Function,([MX]))'
```

According to the casadi docs I linked above, I need to pass a list as a single argument to the interpolant and this is what I intend to do, but it doesn't interpret it correctly. If you know the solutionto this, please let me know. Otherwise I will continue with this on Monday. I have forked PyBaMM and pushed all my changes, you can have a look.

ehtec, 2021-11-09:
This issue turned out to be more difficult to resolve than expected. I have tried multiple things but it still does not work. 

I create the casadi interpolant like this, according to the documentation:

```python
LUT = casadi.interpolant(
    "LUT", "bspline", symbol.x, symbol.y.ravel(order='F')
)
```

Hardcoding the converted children using numbers works:

```python
res = LUT([0.5, 300.0])
```

Supplying the converted children as arguments does not work:

```python
res = LUT(*converted_children)
```

```
RuntimeError: .../casadi/core/function_internal.hpp:1241: Assertion "arg.size()==n_in_" failed:
Incorrect number of inputs: Expected 1, got 2
```

Supplying them as a list does also not work:

```python
res = LUT(converted_children)
```

```
NotImplementedError: Wrong number or type of arguments for overloaded function 'Function_call'.
Possible prototypes are:
  call(self,dict:DM,bool,bool)
  call(self,[DM],bool,bool)
  call(self,[SX],bool,bool)
  call(self,dict:SX,bool,bool)
  call(self,dict:MX,bool,bool)
  call(self,[MX],bool,bool)
You have: '(Function,([MX]))'
```

Supplying them as two lists also does not work:

```python
res = LUT([converted_children[0]], [converted_children[1]])
```

```
NotImplementedError: Wrong number or type of arguments for overloaded function 'Function_call'.
Possible prototypes are:
  call(self,dict:DM,bool,bool)
  call(self,[DM],bool,bool)
  call(self,[SX],bool,bool)
  call(self,dict:SX,bool,bool)
  call(self,dict:MX,bool,bool)
  call(self,[MX],bool,bool)
You have: '(Function,([MX],[MX]))'
```

Supplying them as two lists surrounded by another list also does not work:

```python
res = LUT([[converted_children[0]], [converted_children[1]]])
```

```
NotImplementedError: Wrong number or type of arguments for overloaded function 'Function_call'.
Possible prototypes are:
  call(self,dict:DM,bool,bool)
  call(self,[DM],bool,bool)
  call(self,[SX],bool,bool)
  call(self,dict:SX,bool,bool)
  call(self,dict:MX,bool,bool)
  call(self,[MX],bool,bool)
You have: '(Function,([[MX]]))'
```

I am running out of ideas how I could pass the children otherwise.

tinosulzer, 2021-11-09:
What is `converted_children` here?

ehtec, 2021-11-09:
See line 101 of `convert_to_casadi.py`.

```python
converted_children = [
                self.convert(child, t, y, y_dot, inputs) for child in symbol.children
            ]
```

This is not my code, all of this is already in the `convert_to_casadi.py` file. I also tested it successfully with the 1D interpolant.

This is what you do with the 1D interpolant

```python
return casadi.interpolant(
                        "LUT", "bspline", symbol.x, symbol.y.flatten()
                    )(*converted_children)
```

I looked at this and the casadi docs

http://casadi.sourceforge.net/api/html/de/dbe/group__interpolant.html

when trying to implement the same for 2D.

tinosulzer, 2021-11-09:
Hmm strange, do you have a casadi-only MWE for this?

ehtec, 2021-11-09:
With casadi only it works, yes. The only change you have to make is to replace the `converted_children` with some numbers, like this:

```python
res = LUT([0.5, 300.0])
```

This works. But as soon as I replace the numbers with `converted_children`, I get errors.

ehtec, 2021-11-09:
This is the content of the `converted_children` variable at the time of the error

```
[MX(y_diff[641:1241]), MX((298.15+y_diff[1301])[0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0])]
```

ehtec, 2021-11-09:
> 
> 
> With casadi only it works, yes. The only change you have to make is to replace the `converted_children` with some numbers, like this:
> 
> ```python
> res = LUT([0.5, 300.0])
> ```
> 
> This works. But as soon as I replace the numbers with `converted_children`, I get errors.

Actually this is the only change that needs to be made to get the whole simulation to work. Everything works, from `pkl` file reading to plotting. So the only thing that fails here is to pass the children to the casadi 2D interpolant

tinosulzer, 2021-11-09:
Could you create an example, including creating the converted_children, that can be run using only casadi, without needing to hardcode converted_children. That might elucidate what's going on here (i.e. we need to create the converted_children differently). The casadi developers won't be able to help without it

ehtec, 2021-11-09:
So what else should I use instead of hardcoded values or values from the PyBaMM simulation then? I am currently running a full cycle aging simulation (that is working in general) and try now to import parameters from a csv using my method (I created the csv using the existing function).

ehtec, 2021-11-09:
Something interesting I just found out:

I create the casadi interpolant like this:

```python
import casadi
xgrid = np.linspace(-5,5,11)
ygrid = np.linspace(-4,4,9)
X,Y = np.meshgrid(xgrid,ygrid,indexing='ij')
R = np.sqrt(5*X**2 + Y**2)+ 1
data = np.sin(R)/R
data_flat = data.ravel(order='F')
# data_flat = data.flatten()
LUT = casadi.interpolant('name','bspline',[xgrid,ygrid],data_flat)
```

now:

```python
print(LUT([[0.5,1], [0.5,1]]))
```

yields an error:

```
NotImplementedError: Wrong number or type of arguments for overloaded function 'Function_call'.
  Possible prototypes are:
    call(self,dict:DM,bool,bool)
    call(self,[DM],bool,bool)
    call(self,[SX],bool,bool)
    call(self,dict:SX,bool,bool)
    call(self,dict:MX,bool,bool)
    call(self,[MX],bool,bool)
  You have: '(Function,([[float|int]]))'
```

If I wrap the parameters in a numpy array, it works:

```python
print(LUT(np.array([[0.5,1], [0.5,1]])))
```

yields

```
[[0.421411, -0.0878551]]
```

Which means it takes the first row of the numpy array as `x`-values and the second as `y`-values.

ehtec, 2021-11-11:
I managed to produce a Casadi-only example:

```python
# Python
import casadi
xgrid = np.linspace(-5,5,11)
ygrid = np.linspace(-4,4,9)
X,Y = np.meshgrid(xgrid,ygrid,indexing='ij')
R = np.sqrt(5*X**2 + Y**2)+ 1
data = np.sin(R)/R
data_flat = data.ravel(order='F')
# data_flat = data.flatten()
LUT = casadi.interpolant('name','bspline',[xgrid,ygrid],data_flat)
print(LUT([0.5,1]))
```

With numbers as argument, it works.

Now we try to put matrices as argument:

```python
from casadi import MX, SX, DM

# a = SX.sym("a")
# b = SX.sym("b")

# a = DM()
# b = DM()

a = MX.sym("a")
b = MX.sym("b")

LUT([a, b])
```

This yields exactly the error we experienced before:

```
NotImplementedError: Wrong number or type of arguments for overloaded function 'Function_call'.
  Possible prototypes are:
    call(self,dict:DM,bool,bool)
    call(self,[DM],bool,bool)
    call(self,[SX],bool,bool)
    call(self,dict:SX,bool,bool)
    call(self,dict:MX,bool,bool)
    call(self,[MX],bool,bool)
  You have: '(Function,([MX]))'
```

Using DM or SX instead of MX yields a similar error.

Interestingly, if we only supply a single matrix as argument, we don't get an error:

```python
LUT(a)
```

yields

```
MX(name(a[0, 0]){0})
```

tinosulzer, 2021-11-11:
Ah, that's interesting, so we may need to somehow concatenate the two matrices into a single one before passing them in? Can you try with two vectors (e.g., `DM([1,2,3,4]`, `DM([5,6,7,8])`), and if that doesn't work try concatenating with `hcat`?

ehtec, 2021-11-11:
Yes I am trying with hcat, this resolved the original error but gave anotherone regarding the shape is not suitable (it gave some more info. I think I might need to transpose it or so). I'll continue with that tomorrow. I think the worst is behind me now.

ehtec, 2021-11-12:
Looks like I finally did it.

```python
res = LUT(casadi.hcat(converted_children).T).T
```

works.

The whole workflow looks now like this:

1. Run a file called `csv_model_trainer.py` which converts a CSV with measurements in a .json and .pkl.
2. Use a `[ml data]` tag, similar to the `[data]` tag, to load the data from the .pkl file and use it as a parameter.

The CSV data is of the form

```
independent variable 1, independent variable 2, dependent variable
```

I have also written a script which helps to create CSV pseudo-measurements from functions (for testing).

And I have a Jupyter notebook to test the functionality as well.

Where in the PyBaMM directory should I put these files?

Another question... do you maybe know a way to parallelize / otherwise speed up operations? When using the 2D interpolant with 1000x1000 measurements, the execution time increases from around 10 minutes to 1 hour.

And if you give me some more notebooks with ready-to use simulations that use 2D parameters I could execute some more tests to make sure everything works as expected.

tinosulzer, 2021-11-12:
Nice! Please open a PR with wherever you have put the files so I can see. Depending on how robust it is, it might be that we just keep it as an example for people to follow rather than a feature. 
See https://github.com/pybamm-team/PyBaMM/blob/develop/examples/notebooks/solvers/speed-up-solver.ipynb for speeding up tests, but the casadi solver is known to really struggle with interpolants. I'm surprised it took 10 minutes in the first place though, usually most models solve in a few seconds.
Tests should be added to spm and dfn here: https://github.com/pybamm-team/PyBaMM/tree/develop/tests/integration/test_models/test_full_battery_models/test_lithium_ion

ehtec, 2021-11-15:
Pull request: https://github.com/pybamm-team/PyBaMM/pull/1805

The solution is very robust and implemented into the files you told me to edit, I was only asking where to put the data preparation part (the script that will convert a .csv into .json and .pkl, which can be read by PyBaMM then). I added it to examples now.

ehtec, 2021-11-16:
By the way only the bspline interpolant took so long, the linear one is a lot quicker.

tinosulzer, 2021-12-08:
fixed by #1805

closed 2021-12-08 by tinosulzer