pybamm-team/PyBaMM #1788: Drive cycle in experiment syntax?

TomTranter, 2021-11-06:
Is this the right syntax?
it results in the following error

  File "c:\users\tom\code\pybamm\pybamm\simulation.py", line 350, in set_up_model_for_experiment_new
    if op_cond["electric"] not in self.op_conds_to_model_and_param:

TypeError: unhashable type: 'numpy.ndarray

```
import pybamm
import os
import pandas as pd
os.chdir(pybamm.__path__[0]+'/..')

# import drive cycle from file
drive_cycle = pd.read_csv("pybamm/input/drive_cycles/US06.csv",
                          comment="#", header=None).to_numpy()

experiment = pybamm.Experiment(operating_conditions=['Run US06 (A)'],
                               drive_cycles={'US06':  drive_cycle})
chemistry = pybamm.parameter_sets.Chen2020
parameter_values = pybamm.ParameterValues(chemistry=chemistry)

sim = pybamm.Simulation(model=pybamm.lithium_ion.SPM(),
                        parameter_values=parameter_values,
                        experiment=experiment)
```

tinosulzer, 2021-11-06:
That is the intended API, but it has not yet been fully implemented (I think the experiment can be set up but not solved), and I'm open to suggestions for a different API

TomTranter, 2021-11-07:
I'm happy with the api, just wondered if I'd got it wrong because of the error. I can take a look at getting it to work in pybamm. Was trying to use it for liionpack but might just circumvent the experiment and allow you to pass drive-cycle in directly

PR #1793 (2021-11-10): Solve experiment with drive cycle
# Description

Experiments now allow to solve for drive cycles.

Fixes #1788 and includes #1524

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist [... cut]
closed 2021-11-10 by brosaplanella