pybamm-team/PyBaMM #1802: on pybamm import RuntimeWarning: invalid value encountered in sign   return self.function(*evaluated_children)

TomTranter, 2021-11-13:
This is coming from 
[[](url)](https://github.com/pybamm-team/PyBaMM/blob/4aa4776285d7a418d7b21f9ab33902073dee1b4b/pybamm/parameters/electrical_parameters.py#L66)

TomTranter, 2021-11-13:
evaluated children is nan  - where does this come from? I see other examples ignoring the error state for nan
# don't raise RuntimeWarning for NaNs
with np.errstate(invalid="ignore"):

tinosulzer, 2021-11-13:
You're getting this warning from `import pybamm`?

TomTranter, 2021-11-14:
Yes

tinosulzer, 2021-11-14:
That is strange. Can you post the full stacktrace?

TomTranter, 2021-11-15:
Ok so I inserted this into electrical parameters:
`    def __init__(self):

        # Get geometric parameters
        self.geo = pybamm.geometric_parameters

        # Set parameters
        self._set_dimensional_parameters()
        import warnings
        warnings.simplefilter("error")
        self._set_dimensionless_parameters()
`
and ran this:
`
import traceback

try:
    import pybamm
except RuntimeWarning:
    print(traceback.format_exc())
`
and got this:
Traceback (most recent call last):
  File "c:\users\tom\code\pybamm\pybamm\expression_tree\symbol.py", line 751, in evaluate_for_shape
    return self._saved_evaluate_for_shape
AttributeError: 'Function' object has no attribute '_saved_evaluate_for_shape'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\tom\code\pybamm_liionpack\examples\scripts\untitled5.py", line 14, in <module>
    import pybamm
  File "c:\users\tom\code\pybamm\pybamm\__init__.py", line 172, in <module>
    from .parameters.electrical_parameters import (
  File "c:\users\tom\code\pybamm\pybamm\parameters\electrical_parameters.py", line 73, in <module>
    electrical_parameters = ElectricalParameters()
  File "c:\users\tom\code\pybamm\pybamm\parameters\electrical_parameters.py", line 28, in __init__
    self._set_dimensionless_parameters()
  File "c:\users\tom\code\pybamm\pybamm\parameters\electrical_parameters.py", line 67, in _set_dimensionless_parameters
    self.dimensional_current_with_time
  File "c:\users\tom\code\pybamm\pybamm\expression_tree\symbol.py", line 516, in __mul__
    return pybamm.simplified_multiplication(self, other)
  File "c:\users\tom\code\pybamm\pybamm\expression_tree\binary_operators.py", line 1018, in simplified_multiplication
    if left.shape_for_testing == right.shape_for_testing and all(
  File "c:\users\tom\code\pybamm\pybamm\expression_tree\symbol.py", line 948, in shape_for_testing
    evaluated_self = self.evaluate_for_shape()
  File "c:\users\tom\code\pybamm\pybamm\expression_tree\symbol.py", line 753, in evaluate_for_shape
    self._saved_evaluate_for_shape = self._evaluate_for_shape()
  File "c:\users\tom\code\pybamm\pybamm\expression_tree\functions.py", line 201, in _evaluate_for_shape
    return self._function_evaluate(evaluated_children)
  File "c:\users\tom\code\pybamm\pybamm\expression_tree\functions.py", line 204, in _function_evaluate
    return self.function(*evaluated_children)
RuntimeWarning: invalid value encountered in sign

closed 2021-11-17 by tinosulzer