pybamm-team/PyBaMM #1804: Parameterization issue

manjunathnilugal, 2021-11-15:
Dear all, 

I have a set of parameters for simulation of the lithium ion batteries, which I have collectively arranged in the "param_file.csv" file.
I used to fetch the parameter values directly from the csv file for simulation (PFA).
Now I have upgraded the pybamm to newer version. I am getting error stating : **ModuleNotFoundError: No module named 'E:'**
In case, if my simulation is wrong kindly let me know.

Regards,
[image]
[image]

Manjunath N

tinosulzer, 2021-11-16:
Which pybamm version causes this error? And do other versions work?

Saransh-cpp, 2021-11-16:
I can't say anything about the `ModuleNotFoundError` but absolute paths in windows should be written like `r"E:\User\Some\Folder\then\File"` (with `r` because of the backslashes).

manjunathnilugal, 2021-11-17:
Pybamm version 0.4.0 work perfectly fine with .csv file.
Recently I upgraded to pybamm version 21.10 it stopped working. I verified with pybamm version 21.08 it still didn't work.
So, I reverted back to older pybamm version 0.4.0.
Kindly help me to resolve this issue.

tinosulzer, 2021-11-17:
Did you try `r"E:\User\Some\Folder\then\File"` ?

tinosulzer, 2021-11-17:
Can you try the branch I just pushed?

manjunathnilugal, 2021-11-18:
I tried the particular simulation using the "r", but still the issue is persisting.
[image]

tinosulzer, 2021-11-18:
Did you try with the branch I pushed that is referenced above?

manjunathnilugal, 2021-11-18:
Dear sir, 
I have written in this format and tried to execute.

import pybamm
import os
model = pybamm.lithium_ion.DFN()
param = pybamm.ParameterValues(os.path.join("E\Manjunath\Model development\P2D\PybaMm\Parameters\Li-ion_New\param_file.csv"))
sim = pybamm.Simulation(model, parameter_values = param)
sim.solve([0,3600])
sim.plot()

But, I am not sure if I have written it correctly, Kindly correct if I am wrong, ( I am very new to python)

manjunathnilugal, 2021-11-22:
Sir, would you please confirm, whether the above written code is correct or not.
Adding parameter in the .csv format is very comfortable in working with large data sets.

tinosulzer, 2021-11-24:
Can you do `pip install git+https://github.com/pybamm-team/pybamm.git@issue-1804-parameterization` and then test again?

manjunathnilugal, 2021-11-26:
Sir, I tried installing the above mentioned steps. Restarted the kernel and simulation.
Still the issue persists.
Kindly help me to fix this issue.

brosaplanella, 2021-11-26:
Did you try uninstalling the previous version and then installing the one Tino recommended?

Saransh-cpp, 2021-11-26:
Try running `pip install git+https://github.com/pybamm-team/pybamm.git@issue-1804-parameterization --force-reinstall` in your environment to get rid of the previous version

manjunathnilugal, 2021-11-26:
I tried to run the above mentioned command, it successfully ran.
But the issue still persists.
Please find the attachment.
[image]

Saransh-cpp, 2021-11-26:
Can you post the full stack trace?

manjunathnilugal, 2021-11-26:
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-7-c6bc9da8f251> in <module>
      3 import sys
      4 model = pybamm.lithium_ion.DFN()
----> 5 param = pybamm.ParameterValues("E:\Manjunath\Model development\P2D\PybaMm\Parameters\LIONSIMBA\param_file.csv")
      6 #param = pybamm.ParameterValues(os.path.join("E:\Manjunath\Model development\P2D\PybaMm\Parameters\Li-ion_New\param_file.csv "))
      7 sim = pybamm.Simulation(model, parameter_values = param)

c:\users\manjunath\appdata\local\programs\python\python39\lib\site-packages\pybamm\parameters\parameter_values.py in __init__(self, values, chemistry)
     83                 path = ""
     84             # Don't check parameter already exists when first creating it
---> 85             self.update(values, check_already_exists=False, path=path)
     86 
     87         # Initialise empty _processed_symbols dict (for caching)

c:\users\manjunath\appdata\local\programs\python\python39\lib\site-packages\pybamm\parameters\parameter_values.py in update(self, values, check_conflict, check_already_exists, path)
    303             if isinstance(value, str):
    304                 if value.startswith("[function]"):
--> 305                     loaded_value = pybamm.load_function(os.path.join(path, value[10:]))
    306                     self._dict_items[name] = loaded_value
    307                     values[name] = loaded_value

c:\users\manjunath\appdata\local\programs\python\python39\lib\site-packages\pybamm\util.py in load_function(filename)
    285     path = root_path.replace("/", ".")
    286     path = path.replace("\\", ".")
--> 287     module_object = importlib.import_module(path)
    288 
    289     return getattr(module_object, tail)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\__init__.py in import_module(name, package)
    125                 break
    126             level += 1
--> 127     return _bootstrap._gcd_import(name[level:], package, level)
    128 
    129 

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)
[... 5 lines cut ...]
c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

ModuleNotFoundError: No module named 'E:'

tinosulzer, 2021-11-26:
Can you update again with the same command then add `pybamm.set_logging_level("DEBUG")` to the start of your script and report the full result

tinosulzer, 2021-11-26:
Also please copy and paste the text, not a screenshot

manjunathnilugal, 2021-11-26:
2021-11-26 23:15:38,661 - [INFO] base_battery_model.build_model(697): Start building Doyle-Fuller-Newman model
2021-11-26 23:15:38,661 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for external circuit submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,661 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for porosity submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,661 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for negative active material submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,661 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for positive active material submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,685 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for electrolyte tortuosity submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,686 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for electrode tortuosity submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,687 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for transverse convection submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,690 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for through-cell convection submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,692 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for negative interface submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,693 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for positive interface submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,694 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for negative oxygen interface submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,695 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for positive oxygen interface submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,697 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for negative particle submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,704 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for positive particle submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,710 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for negative electrode potential submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,713 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for positive electrode potential submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,716 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for electrolyte diffusion submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,722 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for electrolyte conductivity submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,729 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for thermal submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,732 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for current collector submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,733 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for sei submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,737 - [DEBUG] base_battery_model.build_fundamental_and_external(583): Getting fundamental variables for lithium plating submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,741 - [DEBUG] base_battery_model.build_fundamental_and_external(597): Getting external variables for external circuit submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,743 - [DEBUG] base_battery_model.build_fundamental_and_external(597): Getting external variables for porosity submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,744 - [DEBUG] base_battery_model.build_fundamental_and_external(597): Getting external variables for negative active material submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,745 - [DEBUG] base_battery_model.build_fundamental_and_external(597): Getting external variables for positive active material submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,746 - [DEBUG] base_battery_model.build_fundamental_and_external(597): Getting external variables for electrolyte tortuosity submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,747 - [DEBUG] base_battery_model.build_fundamental_and_external(597): Getting external variables for electrode tortuosity submodel (Doyle-Fuller-Newman model)
2021-11-26 23:15:38,748 - [DEBUG] base_battery_model.build_fundamental_and_external(597): Getting external variables for transverse convection submodel (Doyle-Fuller-Newman model)
[... cut ...]
c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

ModuleNotFoundError: No module named 'E:'

tinosulzer, 2021-11-26:
You didn't update properly. I don't see this line in the stacktrace https://github.com/pybamm-team/PyBaMM/blob/544a833bfabb0daecb0acc698658479b9e9645c3/pybamm/util.py#L292

Maybe the command I shared isn't working. Can you clone the repository and [install from source](https://pybamm.readthedocs.io/en/latest/install/install-from-source.html) then switch to the branch `issue-1804-parameterization`?

manjunathnilugal, 2021-11-26:
Sir, I tried un-installing the pybamm, again reinstalled from repository. Then cloned the _issue-1804-parmetrization_ .
Still the issue persists. The Line 292 is not visible.

tinosulzer, 2021-11-26:
If your local file doesn't look like this then you haven't installed it correctly https://github.com/pybamm-team/PyBaMM/blob/issue-1804-parameterization/pybamm/util.py
Obviously the issue will persist if you don't correctly install the version with the fix.

tinosulzer, 2021-11-26:
If your local file doesn't look like this then you haven't installed it correctly https://github.com/pybamm-team/PyBaMM/blob/issue-1804-parameterization/pybamm/util.py
Please install this branch and then check, and if not report the debug log

manjunathnilugal, 2021-11-27:
Sir, i have cloned the repository util.py in my local directory. PFA
Error:

ModuleNotFoundError                       Traceback (most recent call last)
C:\Users\MANJUN~1\AppData\Local\Temp/ipykernel_4328/1506530120.py in <module>
      4 import sys
      5 model = pybamm.lithium_ion.DFN()
----> 6 param = pybamm.ParameterValues("E:\Manjunath\Model development\P2D\PybaMm\Parameters\LIONSIMBA\param_file.csv")
      7 sim = pybamm.Simulation(model, parameter_values = param)
      8 sim.solve([0,3600])

c:\users\manjunath\appdata\local\programs\python\python39\lib\site-packages\pybamm\parameters\parameter_values.py in __init__(self, values, chemistry)
     95                     path = ""
     96                 # Don't check parameter already exists when first creating it
---> 97                 self.update(values, check_already_exists=False, path=path)
     98 
     99         # Initialise empty _processed_symbols dict (for caching)

c:\users\manjunath\appdata\local\programs\python\python39\lib\site-packages\pybamm\parameters\parameter_values.py in update(self, values, check_conflict, check_already_exists, path)
    297             if isinstance(value, str):
    298                 if value.startswith("[function]"):
--> 299                     loaded_value = pybamm.load_function(os.path.join(path, value[10:]))
    300                     self._dict_items[name] = loaded_value
    301                     values[name] = loaded_value

c:\users\manjunath\appdata\local\programs\python\python39\lib\site-packages\pybamm\util.py in load_function(filename)
    293         f"Importing function '{tail}' from file '{filename}' via path '{path}'"
    294     )
--> 295     module_object = importlib.import_module(path)
    296 
    297     return getattr(module_object, tail)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\__init__.py in import_module(name, package)
    125                 break
    126             level += 1
--> 127     return _bootstrap._gcd_import(name[level:], package, level)
    128 
    129 

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)
[... 7 lines cut ...]
c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

ModuleNotFoundError: No module named 'E:'

Saransh-cpp, 2021-11-27:
> i have cloned the repository

Did you install the cloned repository? Can you make a new environment and install it there?

manjunathnilugal, 2021-11-27:
I installed from the repository and then cloned the 1804-parmatermization.

tinosulzer, 2021-11-27:
please set logging level to "DEBUG" and run again

manjunathnilugal, 2021-11-28:
Sir, please find the debug report.

2021-11-28 10:36:08,886 - [INFO] base_battery_model.build_model(741): Start building Doyle-Fuller-Newman model
2021-11-28 10:36:08,891 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for external circuit submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,893 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for porosity submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,896 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for negative active material submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,901 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for positive active material submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,906 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for electrolyte tortuosity submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,907 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for electrode tortuosity submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,908 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for transverse convection submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,912 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for through-cell convection submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,915 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for negative interface submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,916 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for positive interface submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,917 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for negative oxygen interface submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,918 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for positive oxygen interface submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,918 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for negative particle submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,921 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for positive particle submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,924 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for negative electrode potential submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,926 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for positive electrode potential submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,928 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for electrolyte diffusion submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,930 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for electrolyte conductivity submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,935 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for negative surface potential difference submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,936 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for separator surface potential difference submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,936 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for positive surface potential difference submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,937 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for thermal submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,940 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for current collector submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,940 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for sei submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,943 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for lithium plating submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,944 - [DEBUG] base_battery_model.build_fundamental_and_external(641): Getting external variables for external circuit submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,945 - [DEBUG] base_battery_model.build_fundamental_and_external(641): Getting external variables for porosity submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,945 - [DEBUG] base_battery_model.build_fundamental_and_external(641): Getting external variables for negative active material submodel (Doyle-Fuller-Newman model)
2021-11-28 10:36:08,946 - [DEBUG] base_battery_model.build_fundamental_and_external(641): Getting external variables for positive active material submodel (Doyle-Fuller-Newman model)
[... cut ...]
c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

ModuleNotFoundError: No module named 'E:'

tinosulzer, 2021-12-01:
The problem is trying to import the path `E:.Manjunath.Model development.P2D.PybaMm.Parameters.LIONSIMBA.graphite_mcmb2528_diffusivity_Dualfoil1998`. Can someone with Windows check how this should be formatted instead?

npaulson, 2021-12-03:
I am having the same issue, and I believe in a simpler situation. I am running in a new conda environment with Python 3.9.7 and pybamm 21.11 on Ubuntu for Windows.

Here is the code generating the error:
```
import pybamm
pybamm.set_logging_level('NOTICE')
model = pybamm.lithium_ion.DFN()
param = model.default_parameter_values
```

And here is the trace:
```
Traceback (most recent call last):
  File "/mnt/c/codes/2021_12_02_pybamm/debug.py", line 6, in <module>
    param = model.default_parameter_values
  File "/home/npaulson/miniconda3/envs/pybamm/lib/python3.9/site-packages/pybamm/models/full_battery_models/lithium_ion/base_lithium_ion_model.py", line 52, in default_parameter_values
    return pybamm.ParameterValues("Marquis2019")
  File "/home/npaulson/miniconda3/envs/pybamm/lib/python3.9/site-packages/pybamm/parameters/parameter_values.py", line 87, in __init__
    self.update_from_chemistry(values)
  File "/home/npaulson/miniconda3/envs/pybamm/lib/python3.9/site-packages/pybamm/parameters/parameter_values.py", line 211, in update_from_chemistry
    self.update(
  File "/home/npaulson/miniconda3/envs/pybamm/lib/python3.9/site-packages/pybamm/parameters/parameter_values.py", line 299, in update
    loaded_value = pybamm.load_function(os.path.join(path, value[10:]))
  File "/home/npaulson/miniconda3/envs/pybamm/lib/python3.9/site-packages/pybamm/util.py", line 292, in load_function
    module_object = importlib.import_module(path)
  File "/home/npaulson/miniconda3/envs/pybamm/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 984, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'pybamm.lithium_ion'
```

tinosulzer, 2021-12-03:
Is there a `lithium_ion` directory in your root pybamm directory?

npaulson, 2021-12-03:
It appears not. Only under `models/full_battery_models/lithium_ion`.

Saransh-cpp, 2021-12-04:
I see the full problem now (with the original issue) -

1. If I am running my script in the same drive in which I have my parameters (running the script in `C` drive and the parameters are present in the `C` drive) then PyBaMM works (It removes the problematic `C:` thing) -
```
2021-12-04 13:15:01,762 - [DEBUG] util.load_function(292): Importing function 'copper_heat_capacity_CRC' from file 'C:\Users\Saransh\Desktop\params\copper_heat_capacity_CRC' via path 'Desktop.params.copper_heat_capacity_CRC'
2021-12-04 13:15:01,762 - [DEBUG] util.load_function(292): Importing function 'aluminium_heat_capacity_CRC' from file 'C:\Users\Saransh\Desktop\params\aluminium_heat_capacity_CRC' via path 'Desktop.params.aluminium_heat_capacity_CRC'
2021-12-04 13:15:01,763 - [DEBUG] util.load_function(292): Importing function 'copper_thermal_conductivity_CRC' from file 'C:\Users\Saransh\Desktop\params\copper_thermal_conductivity_CRC' via path 'Desktop.params.copper_thermal_conductivity_CRC'
```

2. If I am running my script from some drive and the parameter values are in another drive (the case here, say the script is running in `C` drive and the parameters are present in the `E` drive), I get the same error from `utils.py` -
```
2021-12-04 13:16:38,419 - [VERBOSE] parameter_values.find_parameter(949): Using absolute path: 'C:\Users\Saransh\Desktop\params\parameters.csv'
2021-12-04 13:16:38,422 - [DEBUG] util.load_function(292): Importing function 'copper_heat_capacity_CRC' from file 'C:\Users\Saransh\Desktop\params\copper_heat_capacity_CRC' via path 'C:.Users.Saransh.Desktop.params.copper_heat_capacity_CRC'
```

I'll try to fix this. @manjunathnilugal, if it is too urgent, then you can move your parameter files to the same drive for now.

Saransh-cpp, 2021-12-04:
@npaulson, could you also install the fix branch using - `pip install git+https://github.com/pybamm-team/pybamm.git@issue-1804-parameterization` and run your script after setting the logging level to "DEBUG" - `pybamm.set_logging_level("DEBUG")`.

Saransh-cpp, 2021-12-04:
@manjunathnilugal, can you try running your script after installing the branch again?

npaulson, 2021-12-04:
@Saransh-cpp I have done so and here is the output:
```
2021-12-04 16:47:51,351 - [INFO] base_battery_model.build_model(741): Start building Doyle-Fuller-Newman model
2021-12-04 16:47:51,352 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for external circuit submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,355 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for porosity submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,359 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for negative active material submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,365 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for positive active material submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,375 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for electrolyte tortuosity submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,384 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for electrode tortuosity submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,394 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for transverse convection submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,398 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for through-cell convection submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,400 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for negative interface submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,401 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for positive interface submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,402 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for negative oxygen interface submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,403 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for positive oxygen interface submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,403 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for negative particle submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,412 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for positive particle submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,440 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for negative electrode potential submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,444 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for positive electrode potential submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,450 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for electrolyte diffusion submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,471 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for electrolyte conductivity submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,481 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for negative surface potential difference submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,492 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for separator surface potential difference submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,506 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for positive surface potential difference submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,508 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for thermal submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,553 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for current collector submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,561 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for sei submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,582 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for lithium plating submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,585 - [DEBUG] base_battery_model.build_fundamental_and_external(641): Getting external variables for external circuit submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,586 - [DEBUG] base_battery_model.build_fundamental_and_external(641): Getting external variables for porosity submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,587 - [DEBUG] base_battery_model.build_fundamental_and_external(641): Getting external variables for negative active material submodel (Doyle-Fuller-Newman model)
2021-12-04 16:47:51,589 - [DEBUG] base_battery_model.build_fundamental_and_external(641): Getting external variables for positive active material submodel (Doyle-Fuller-Newman model)
[... cut ...]
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 984, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'pybamm.lithium_ion'
```

manjunathnilugal, 2021-12-05:
I tried moving the parameters to the C drive, it is showing the same error, here I am attaching the trace. 
[Trace.txt](https://github.com/pybamm-team/PyBaMM/files/7655560/Trace.txt)

Saransh-cpp, 2021-12-05:
@manjunathnilugal, my bad, I guess right now the parameters should be in your current working directory. For example -
If you are running the script in `C:/x/y/z` then the parameters can be anywhere in `z`, like in `C:/x/y/z/a/b/parameters.csv`

Can you install the branch again (with the latest commit) and try running your script?

manjunathnilugal, 2021-12-06:
I tried running the script from C drive. Here the instead of module names looks strange. instead of **LIONSIMBA**, it is showing **IONSIMBA**
```
2021-12-06 10:31:29,933 - [INFO] base_battery_model.build_model(741): Start building Doyle-Fuller-Newman model
2021-12-06 10:31:29,938 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for external circuit submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:29,939 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for porosity submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:29,945 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for negative active material submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:29,958 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for positive active material submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:29,964 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for electrolyte tortuosity submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:29,966 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for electrode tortuosity submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:29,967 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for transverse convection submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:29,973 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for through-cell convection submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:29,976 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for negative interface submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:29,978 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for positive interface submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:29,979 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for negative oxygen interface submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:29,980 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for positive oxygen interface submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:29,981 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for negative particle submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:29,987 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for positive particle submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:29,991 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for negative electrode potential submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:29,994 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for positive electrode potential submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:29,997 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for electrolyte diffusion submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:30,000 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for electrolyte conductivity submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:30,005 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for negative surface potential difference submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:30,006 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for separator surface potential difference submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:30,007 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for positive surface potential difference submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:30,008 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for thermal submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:30,012 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for current collector submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:30,013 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for sei submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:30,017 - [DEBUG] base_battery_model.build_fundamental_and_external(627): Getting fundamental variables for lithium plating submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:30,019 - [DEBUG] base_battery_model.build_fundamental_and_external(641): Getting external variables for external circuit submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:30,020 - [DEBUG] base_battery_model.build_fundamental_and_external(641): Getting external variables for porosity submodel (Doyle-Fuller-Newman model)
2021-12-06 10:31:30,020 - [DEBUG] base_battery_model.build_fundamental_and_external(641): Getting external variables for negative active material submodel (Doyle-Fuller-Newman model)
[... cut ...]
c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

c:\users\manjunath\appdata\local\programs\python\python39\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

ModuleNotFoundError: No module named 'IONSIMBA'
````

Saransh-cpp, 2021-12-06:
Thanks for trying this out! Apparently `os.getcwd()` returns `"C:\\"` when in the root drive and `"C:\\a\\b\\c"` when not (another bug🙂). Can you try again with the latest commit?

manjunathnilugal, 2021-12-06:
@Saransh-cpp i tried with the above mentioned commitment, the same error pops up.
ModuleNotFoundError: No module named 'IONSIMBA'
(The trace remains same as above)

Saransh-cpp, 2021-12-19:
It was working for me the last time I checked, I'll check again and let you know.

Saransh-cpp, 2021-12-19:
@manjunathnilugal running the following script - 
```py
import pybamm
params = pybamm.ParameterValues(r"C:\params\parameters.csv")
```
from the path `"C:\"`, where the parameter values are present at `"C:\params\parameters.csv"` gives me no error (using this branch). Could you re-install and try again?

manjunathnilugal, 2021-12-29:
I installed pybamm 0.4.0 version which was able to fetch the parameter details. Recently I installed the latest version of the pybamm and try to run the code. The new pybamm was able to fetch the parameter path which was running in 0.4.0 version.
I observed something strange, newer version of pybamm was not able to fetch the new path but it can trace the previous path which was executed by the 0.4.0.

Saransh-cpp, 2022-01-08:
@manjunathnilugal was this fixed by the commits present above?

manjunathnilugal, 2022-01-09:
Hi,

No actually it didn't solve the problem. But, I have tried to find a way to
execute
I installed 0.4.0, ran the code, then upgraded to the latest version. It
was able to recognise the path. But I don't know how it worked.

Regards,
Manjunath N

Saransh-cpp, 2022-01-11:
That is weird. The last time I checked, the bug was present. I'll take a look at it and finish the fix for a PR if the bug exists.

amirsh213, 2022-01-12:
@Saransh-cpp, Sorry, I just saw this in the issues, you can remove my post in the discussion. I was trying to follow [this tutorial](https://www.youtube.com/watch?v=oUhOL8ebOFw&t=329s) for parameterization and I got this error as soon as I created a lithium_ion folder with new parameters. It does not depend on my script because I got the same error when I simply want to run CC discharge. When I remove this folder everything works. The folder is in the same "C" drive as pybamm is and I am using version 21.08.

Saransh-cpp, 2022-01-12:
@amirsh213 can you uninstall the current version and install a different branch of `PyBaMM` using -
```
pip uninstall --yes pybamm
pip install git+https://github.com/pybamm-team/pybamm.git@issue-1804-parameterization
```
and add -
```py
pybamm.set_logging_level("DEBUG")
```
at the top of your script and try again?

amirsh213, 2022-01-12:
I reinstalled pybamm from the cloned repository and the subsequent line of code but still get the same error:
`ModuleNotFoundError: No module named 'C:'`

Saransh-cpp, 2022-01-13:
Please post the complete stacktrace

amirsh213, 2022-01-13:
`---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
~\AppData\Local\Temp/ipykernel_4732/2875356545.py in <module>
      2 #os.chdir(pybamm.__path__[0]+'/..')
      3 
----> 4 parameter_values = pybamm.ParameterValues(chemistry=chemistry)

~\Anaconda\PyBaMM-21.08\pybamm\parameters\parameter_values.py in __init__(self, values, chemistry)
     72         # First load chemistry
     73         if chemistry is not None:
---> 74             self.update_from_chemistry(chemistry)
     75         # Then update with values dictionary or file
     76         if values is not None:

~\Anaconda\PyBaMM-21.08\pybamm\parameters\parameter_values.py in update_from_chemistry(self, chemistry)
    215 
    216             # Update parameters, making sure to check any conflicts
--> 217             self.update(
    218                 component_params,
    219                 check_conflict=True,

~\Anaconda\PyBaMM-21.08\pybamm\parameters\parameter_values.py in update(self, values, check_conflict, check_already_exists, path)
    303             if isinstance(value, str):
    304                 if value.startswith("[function]"):
--> 305                     loaded_value = pybamm.load_function(os.path.join(path, value[10:]))
    306                     self._dict_items[name] = loaded_value
    307                     values[name] = loaded_value

~\Anaconda\PyBaMM-21.08\pybamm\util.py in load_function(filename)
    271     path = root_path.replace("/", ".")
    272     path = path.replace("\\", ".")
--> 273     module_object = importlib.import_module(path)
    274 
    275     return getattr(module_object, tail)

~\anaconda3\lib\importlib\__init__.py in import_module(name, package)
    125                 break
    126             level += 1
--> 127     return _bootstrap._gcd_import(name[level:], package, level)
    128 
    129 

~\anaconda3\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

~\anaconda3\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

~\anaconda3\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

~\anaconda3\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

~\anaconda3\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

~\anaconda3\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

~\anaconda3\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

~\anaconda3\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

~\anaconda3\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _call_with_frames_removed(f, *args, **kwds)

~\anaconda3\lib\importlib\_bootstrap.py in _gcd_import(name, package, level)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load(name, import_)

~\anaconda3\lib\importlib\_bootstrap.py in _find_and_load_unlocked(name, import_)

ModuleNotFoundError: No module named 'C:'
`

Saransh-cpp, 2022-01-13:
It looks like the installed version is still `21.08`. You should install the branch through `Anaconda prompt`, given that you are using `Anaconda`.

closed 2022-01-16 by tinosulzer

manjunathnilugal, 2022-01-28:
> Closed #1804 <https://github.com/pybamm-team/PyBaMM/issues/1804> via #1900

Dear Pybamm-Team/Pybamm,

I have recently updated to pybamm version 21.12, I am still facing the same
issue.
I pybamm is  not able to detect the path of the local csv file for
parameters. I have to install the 0.4.0 version initially and then have to
run the simulation once the path is traced, i have to update to a newer
version.

Kindly fix this major bug. I am always available, if anything is required
from my side.

Regards,
Manjunath N

manjunathnilugal, 2022-01-28:
Dear Pybamm-Team/Pybamm,

I have recently updated to pybamm version 21.12, I am still facing the same
issue.
Pybamm is not able to detect the path of the local csv file for parameters.
I have to install the 0.4.0 version initially and then have to run the
simulation to trace the path. I have to update to a newer version later on.

Kindly fix this major bug. I am always available, if anything is required
from my side.

Regards,
Manjunath N

Saransh-cpp, 2022-01-28:
@manjunathnilugal the fix will be available in the next release (version `22.1`). It should be available in the next couple of days (check #1922). For now you can try installing the develop branch using -
```
pip install git+https://github.com/pybamm-team/PyBaMM@develop
```

Let us know if you face any issues.

PR #1900 (2022-01-16): Fix parameter path bug in windows
# Description

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context. List any dependencies that are required for this change.

Fixes #1804 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-e [... cut]