pybamm-team/PyBaMM #1807: Add multiple advanced SEI and lithium plating submodels

ehtec, 2021-11-16:
I need to add the following submodels to PyBaMM:

**Coupling between Li plating and SEI growth:**
Possibilities: Arora, Doyle and White, Yang et al. and Zhao et al.
- Arora, Doyle and White assume that a fraction of the plated Li immediately reacts to form SEI instead of forming Li metal, but the simplicity of their model means this SEI can be stripped, as if it was plated Li.
- Yang et al. use two separate Tafel equations for the two reactions, in order to model a scenario where SEI-induced pore clogging causes rapid Li plating even at room temperature, causing the transition between the stabilisation and saturation stages of degradation
- Zhao et al. present an updated version of Arora, Doyle and White’s model that allows them to model both reversible and irreversible plating at the same time

**Combined model of Mn dissolution, SEI formation and Mn deposition on the NE:**
Possibilities: Lin et al.
- The model was able to qualitatively reproduce all three stages of degradation, but neither quantitative comparison with experiment nor temperature variation were included

**Combined model of Mn dissolution, SEI formation and Mn deposition on the NE + semi-empirical model of SEI formation on cracks in particles:**
Possibilities: Li, Landers and Park

**SEI growth on the NE, the same model applied to pSEI growth on the PE and an empirical model for loss of electrical contact due to particle cracking in both electrodes:**
Possibilities: Jin et al. 
- was found to be far superior to both equivalent circuit and purely empirical models in predicting capacity loss during an EV drive cycle

**SPM incorporating a NE SEI model:**
Possibilities: Reniers, Mulder and Howey
- a Tafel equation for irreversible lithium plating on the NE, an empirical expression for loss of electrical contact in the NE and a Tafel equation for acid dissolution of the PE

**P2D model combining SEI growth, SEI formation on cracks and partially reversible Li plating:**
Possibilities: Keil and Jossen
- Li plating could explain the transition from linear to nonlinear ageing, even at room temperature

From these combined models, three different possible explanations for the transition between stabilisation and saturation stages emerge.

-	Lin et al.’s 2013 model predicts:
stabilisation stage is dominated by SEI growth on the NE while the saturation stage is dominated by acid dissolution of the PE
-	Yang et al:
        the transition is caused by SEI-induced pore clogging, which in turn causes Li plating, neglecting the role of the PE completely
-	Keil and Jossen:
 	partially reversible Li plating could explain the transition
-	Reniers, Mulder and Howey
 	empirical model of electrical contact loss also reproduces the saturation stage

I have all the necessary formulas from the papers, but I have not yet added any submodel to PyBaMM before. I had a look at the documentation

https://pybamm.readthedocs.io/en/latest/tutorials/add-model.html

but it mainly focuses on adding a full new model, whereas I want to add submodels which can be used with SPM and DFN. Furthermore, I see there has already been some work done on this topic, and also used it successfully

https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/submodels/interface/sei/base_sei.py

https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/submodels/interface/lithium_plating/base_plating.py

Is there a way to use this work to simplify my submodels? I do not want to write redundant code.

Please give me some advice how to go forward here. There is not too much documentation on that topic, and it's a little bit difficultto try to understand everything from the source code.

Thanks in advance.

tinosulzer, 2021-11-16:
We're going to move all the SEI models to a single model with if statements, and same for both plating models. It should be clearer after that.

ehtec, 2021-11-16:
Ok, thanks. Any estimate when this will be done? I am targeting to get this finished by the end of next week, so if it takes longer, I will try to work with the currently existent code, but if you are modifying it in a matter of days anyways, please let me know, then I will delay my changes because I'd only generate merge conflicts and do redundant work.

tinosulzer, 2021-11-16:
Will be done in the next few days

tinosulzer, 2021-11-17:
Done. Take a look at https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/submodels/interface/sei/sei_growth.py and https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/submodels/interface/lithium_plating/plating.py. Hopefully it is clearer now how to add a new model, you just need to add a new option for sei/plating and the corresponding if statement in the submodel

ehtec, 2021-11-19:
Thanks a lot for the quick help, I will look into this and check how I can implement my own models.

ehtec, 2021-11-22:
Question: Where do I find the param property names mapped to strings from `ParameterValues`? For example, in the lithium plating file, there is a parameter `j0_plating = param.j0_plating`. However, I could not find where `param.j0_plating` is being set and how I would set it. I checked the `BasePlating` and `BaseInterface` classes. I would need to set parameters like this:

```python
pv = pybamm.ParameterValues(chemistry=my_chemistry)
pv["any parameter name"] = the_value
```

I don't know what `"any parameter name"` is for `param.j0_plating`.

rtimms, 2021-11-22:
The symbols for the parameters are all defined [here](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/parameters/lithium_ion_parameters.py).

ehtec, 2021-11-30:
I think there is a mistake in line 98 or 89 of https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/submodels/interface/sei/sei_growth.py. `j_sei` has once the unit `1/[C_sei]` and once `1/([C_sei] [L_sei_inner])`. This is impossible unless `L_sei_inner` has no unit, however, it should be a length.

ehtec, 2021-11-30:
And could you maybe link the papers where these formulas are from, apart from Yang 2017? That would make it a lot easier to understand `reaction limited`, `ec reaction limited`, etc.

PippaVossen, 2022-01-13:
Hi there @ehtec have you been able to find any more information on what `ec reaction limited` exactly entails?

rtimms, 2022-01-13:
@PippaVossen ec reaction limited uses eqns 10 and 11 from [Yang (2017)](https://www.sciencedirect.com/science/article/pii/S0378775317307619). 

@ehtec Re the question about units: note the equations in pybamm are written in dimensionless form. Would need to chase the refs for the other models, we have an issue for this #1260

PippaVossen, 2022-01-13:
Many thanks @rtimms !

satishrapol, 2023-01-09:
Hello, 
I have created a new parameter like this "param.update({'optim-parameter': 1.0},check_already_exists=False)" before the start simulation and I wish to access this parameter inside the sei  growth ageing model which I can change after simulation in a for loop. Any idea how can I access this value inside the sei.growth file. 
Let me know if there is another efficient way also.
Many thanks

rtimms, 2023-05-15:
Closing this. New issues can be opened for specific models.

closed 2023-05-15 by rtimms