pybamm-team/PyBaMM #1812: Using same naming convention for summed up and averaged variables

dion-w, 2021-11-18:
**Summary**
Requested is refactoring the variables starting with "Sum of" to "R-Sum of" and "X-Sum of" to better understand in which direction the values are summed up.  

 **Motivation**
The naming convention mentioned in the summary already holds for "Averaged" quantities in the code. Using the same naming convention for summed up variables would increase understanding of the code and less confusion.

**Additional context**
I got confused when i looked into the code and saw a variable named "Sum of electrode interfacial current densities". What is summed up here? Is it the R or the x-direction? In the context of  a model in "full_surface_form_conductivity" i guess it's R-direction, but i can't directly see it. The naming convention for the averaged values is used with "R-Averaged" or "X-Averaged" and the same convention being used for summed up variables would reduce confusion when looking into a code. A simple "R-sum" or "x-sum" might do it but i'm also open for better ideas.

tinosulzer, 2021-11-18:
In this context "sum of" means summing over all the different reactions (intercalation, sei, lithium plating, any future ones)

dion-w, 2021-11-19:
Maybe the question would have been more suitable for the Discussion then. I tried to think of a more intuitive name for this variable but i couldn't come up with one. May we close this issue then?

closed 2021-11-19 by dion-w