pybamm-team/PyBaMM #1816: DomainError for parameterSet Mohtat2020 when using surface form for electrolyte conductivity

dion-w, 2021-11-19:
- PyBaMM version - `$ python -m pip show pybamm`: v21.08 - 2021-08-26
- Python version - `$ python --version`: 3.10

**Describe the bug**
A DFN model with options={"electrolyte conductivity": "full", "surface form": "differential"} can't be solved with the parameterSet of Mohtat2020 because of the error:
`pybamm.expression_tree.exceptions.DomainError: Cannot concatenate child 'Sparse Matrix (20, 21) @ (0.0013992044174743195 * ((Sparse Matrix (21, 20) @ (1.24 / (y[1201:1221] / 0.3)) + Sparse Matrix (21, 19) @ ((Sparse Matrix (19, 20) @ (1.24 / (y[1201:1221] / 0.3))) * (Sparse Matrix (19, 20) @ (1.24 / (y[1201:1221] / 0.3))) / (Sparse Matrix (19, 20) @ (1.24 / (y[1201:1221] / 0.3)) + 1e-16))) * (Sparse Matrix (21, 20) @ y[1201:1221]) + Sparse Matrix (21, 20) @ y[1261:1281] + Column vector of length 21 + Sparse Matrix (21, 1) @ (0.6294267348410779 - ((Sparse Matrix (1, 20) @ (1.24 / (y[1201:1221] / 0.3))) * (Sparse Matrix (1, 20) @ y[1201:1221])) - 0.002809524352153561)) + 3.931098884534938e-06)' with empty domain`

**To Reproduce**
`import pybamm
import numpy as np
model = pybamm.lithium_ion.DFN(options={"electrolyte conductivity": "full", "surface form": "differential"})
params = [pybamm.ParameterValues(chemistry=pybamm.parameter_sets.Mohtat2020),
          pybamm.ParameterValues(chemistry=pybamm.parameter_sets.Marquis2019)]
for param in params:
    sim = pybamm.Simulation(model, parameter_values=param)
    sol = sim.solve(np.arange(0.001, 9000, 2))`

**Expected behaviour**
I expected the error to be independent of ParameterSet but instead the error occurs only for the parameter set of Mohtat2020 and not for Marquis2019. 

**Additional context**
I found the error when dealing with my own code and the above submodel. Since i didn't want to upload my code, luckily the same error happens with the parameterSet of Mohtat2020. Using a text comparison programm to compare the parameter csv files revealed no exceptional difference between the parameter sets of Mohtat2020 and Marquis2019.

dion-w, 2021-11-21:
Digging a bit deeper i found out that the electrolyte parameters mess something up. While Mohtat uses c_e - independent electrolyte conductivity and diffusivity Marquis use c_e dependent values. When using the parameterSet of Marquis and only using the electrolyte parameters from Mohtat the concatenation error happens again. When using the parameterSet of Mohtat and only using the electrolyte parameters from Marquis the simulation takes unusually long. When i exit the calculation early it says `rhs failed: KeyboardInterrupt
At t = 0.000210875, the right-hand side routine failed in an unrecoverable manner.`.

dion-w, 2021-11-21:
When i used electrolyte concentration dependent Diffusivity and conductivity in my own parameter Set up it also worked. SO final question: Why does a constant value of electrolyte diffusivity and conductivity produce a concatenation error in this setup?

tinosulzer, 2021-11-22:
Seems like a bug in the symbolic representation of the equations. Will look into it

PR #1831 (2021-12-02): Issue 1816 domain error
# Description

Some changes to the expression tree simplification logic. 
Fixes #1816 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No style i [... cut]
closed 2021-12-02 by tinosulzer