pybamm-team/PyBaMM #1846: How to implement a paramter function of electrolyte transport properties with interpolation?

thuwangming, 2021-12-08:
Hello, 

I try to implement a parameter function of electrolyte transport properties, e.g. diffusivity, conductivity, or cation transference number.  The function is temperature-dependent, and the values at reference temperature are evaluated by interpolation with respect to electrolyte concentration. So the function has the form: D(c_e, T) = interp(C_e) * exp[Ea/R * (1/T_ref - 1/T)], which is the production of an interpolation function and an Arrhenius term.

First, I use the scipy.interpolate.interp1d function, like this:
`electrolyte_diff_data = np.genfromtxt('EC3_EMC7_LiPF6_diff.csv', delimiter=',')
interp_diff = interp1d(electrolyte_diff_data[:, 0], electrolyte_diff_data[:, 1], kind='cubic')
def electrolyte_diffusivity(c_e, T):
    D_c_e = interp_diff(c_e)
    E_D_e = 15000
    arrhenius = exp(E_D_e / constants.R * (1 / 298.15 - 1 / T))
`
But it raises an error like: "ValueError: object arrays are not supported".

So next, I use the pybamm.Interpolant class, but the "children" parameter makes me confused.
In this case, what should be passed to the children parameter?
And then, can the pybamm.Interpolant class be multiplied by a temperature-dependent Arrhenius term in the diffusivity parameter function?
Or there is another way to implement this?
Thank you!

rtimms, 2021-12-08:
You can do things like this
```python3
from pybamm import Interpolant, exp, constants
import numpy as np

def graphite_diffusivity(sto, T):
    sto_ref = np.array([0, 4e-2, 1e-1, 1.5e-1, 1.9e-1, 5e-1, 6e-1, 9.7e-1, 1])
    diffusivity_ref = np.array(
        [6e-12, 2e-12, 1e-12, 3e-13, 1e-13, 1e-14, 1e-15, 2e-15, 4e-13]
    )
    
    D_ref = Interpolant(sto_ref, diffusivity_ref, sto)

    E_D_s = 42770
    arrhenius = exp(E_D_s / constants.R * (1 / 298.15 - 1 / T))

    return D_ref * arrhenius

```
but you might find you get better performance from fitting a function to your data and using that instead

brosaplanella, 2021-12-09:
Shall we convert this issue into a discussion?

rtimms, 2021-12-09:
yeah, we get asked this a lot