pybamm-team/PyBaMM #1862: Unable to initialize simulation with all_first_states

sravanpannala, 2021-12-16:
- PyBaMM version - `21.11`:
- Python version - `3.9.9`:

**Describe the bug**
I'm trying to initialize simulation by using all_first_states. I then chose the cycle I want to simulate and then initialize the simulation by using starting_solution=first_states[cycle_no].  

**To Reproduce**
Steps to reproduce the behaviour:
1. simulate N cycles. save the solution variable as `sol`
2. Get first states by the following step `first_states = sol.all_first_states`
3. Initialize another simulation for a different experiment using the first states of a particular cycle
```
sim = pybamm.Simulation(spm, experiment=experiment2, parameter_values=parameter_values, 
                            solver=pybamm.CasadiSolver("safe"))
sol2 = sim.solve(starting_solution=first_states[n])
```
4. See error:
```
AttributeError                            Traceback (most recent call last)
/tmp/ipykernel_804/3421577315.py in <module>
      1 sim_exp = pybamm.Simulation(spm, experiment=experiment, parameter_values=parameter_values, 
      2                             solver=pybamm.CasadiSolver("safe"))
----> 3 sol_exp = sim_exp.solve(starting_solution=first_states[1])

~/PyBaMM/pybamm/simulation.py in solve(self, t_eval, solver, check_model, save_at_cycles, calc_esoh, starting_solution, initial_soc, **kwargs)
    824                 starting_solution_cycles = starting_solution.cycles.copy()
    825                 starting_solution_summary_variables = (
--> 826                     starting_solution.all_summary_variables.copy()
    827                 )
    828                 starting_solution_first_states = (

AttributeError: 'Solution' object has no attribute 'all_summary_variables'
```
**Expected behaviour**
I expected to get a simulation with the first state from the nth cycle. But instead, I get an error.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

brosaplanella, 2021-12-17:
Hi @sravanpannala! You can't use it as a starting solution as this would append the new solution to the starting one and there is a mismatch in the classes. What you can do is to initialise the model with the corresponding first state:
```python3
model.set_initial_conditions_from(first_states[1])
```
and then create a simulation with this model. This should do what you want I think.

sravanpannala, 2021-12-21:
Yes, this worked. Thank You!!

closed 2021-12-21 by sravanpannala