pybamm-team/PyBaMM #1863: idklu solver for `convert_to_casadi=True`

martinjrobins, 2021-12-17:
**Summary**

The current idklu solver goes back to python whenever it needs to calculate the rhs, algebraic or event equations. It also runs very slow when using `convert_to_casadi=True`. This issue aims to develop a new idklu solver that directly calls casadi functions from C++, with ideally no memory copying to/from casadi arrays

 **Motivation**

We want to get the superior event handling capabilities of the ida solvers with efficient casadi evaluation of equations

martinjrobins, 2022-03-24:
this largely seems to be working ok now and scaling as expected. Its faster than the casadi solver with events, but there are still some things to do to speed it up further:
- serialising the casadi functions takes a fair bit of time, need to do this in the solver setup, not the solve itself
- we're allocating memory for return vectors in the solve, again this could be done in solver setup
- also we initialise ida_mem and everything every time, could do this once when solver is created, or only on the 1st solve etc
- need to play around with ida sundials settings: create an options dict to pass to the solver to choose various things. A major addition here would be to look at the iterative solvers

tinosulzer, 2022-03-24:
> we're allocating memory for return vectors in the solve, again this could be done in solver setup

On this subject, an idea to consider is to have the option to calculate some scalar variables (voltage would be the main one but could be anything) at each time step, and then not store the `y` vector.

tinosulzer, 2022-03-24:
> A major addition here would be to look at the iterative solvers

Iterative solvers only start to outperform direct solvers at relatively large matrix sizes, right? (don't know how large)

martinjrobins, 2022-03-25:
Yes, iterative solvers can give speed benefits at the very large matrix sizes, as long as you have a decent preconditioner (i.e. a good approximation to the jacobian that you can solve easily).

martinjrobins, 2022-03-31:
This is largely finished, see pr #2002, here is my todo list for subsequent issues:
- [ ] we're allocating memory for return vectors in the solve, this could be done in solver setup
- [ ] we initialise ida_mem and everything every time, could do this once when solver is created, or only on the 1st solve etc
- [ ] create an options dict to pass to the solver, e.g. option to print solver stats,  don't/do use jacobian
- [ ] add option to use iterative solvers
- [ ] add option to use other linear solvers in ida
- [ ] add option to use dense jacobian
- [ ] benchmarks for all of the above solver options
- [ ] investigate if we need to densify the rhs_algebraic casadi evaluation, could it return a sparse vector?
- [ ] investigate if constructing a new casadi function for the residual (rather than doing it manually in c++) is faster
- [ ] investigate if constructing a new casadi function for the sensitivity calculations (rather than doing it manually in c++) is faster
- [ ] we calculate consistent initial conditions at the start of the idklu solve, so we might not need to do it in the BaseSolver setup

PR #2002 (2022-04-20): I1863 idklu casadi
# Description

Adds a casadi version of the idklu solver, currently getting approx 2 x speed-up over current casadi solver with fast events, but this still needs to be properly benchmarked

As part of the above, the `process` function in the base solver now also returns a function that evaluates the jacobian times a vector (e.g. `casadi.jtimes`). This was neccessary to implement the sensitivity equations that IDA expects, but will also be useful when we add the option of using iterative solvers.

update: also fixed a discretisation bug that occurs when an input parameter is  [... cut]
closed 2022-04-20 by martinjrobins