pybamm-team/PyBaMM #1882: Failing tests

Saransh-cpp, 2022-01-03:
The tests are failing with an error -
```
File "/tmp/pip-build-env-uklwrsnk/overlay/lib/python3.9/site-packages/numpy/distutils/command/build_ext.py", line 380, in build_extension
      extra_cflags = ext.extra_c_compile_args or []
  AttributeError: 'Extension' object has no attribute 'extra_c_compile_args'
  ----------------------------------------
  ERROR: Failed building wheel for scikits.odes
Failed to build scikits.odes
ERROR: Could not build wheels for scikits.odes, which is required to install pyproject.toml-based projects
```

I tried looking it up and found an issue (https://github.com/bmcage/odes/issues/130) in the `odes` repository related to the `distutils` module of `numpy`. The latest `numpy` [release](https://github.com/numpy/numpy/releases/tag/v1.22.0) has a "Compatibility note" section that says -
### Distutils forces strict floating point model on clang
NumPy now sets the -ftrapping-math option on clang to enforce correct
floating point error handling for universal functions. Clang defaults to
non-IEEE and C99 conform behaviour otherwise. This change (using the
equivalent but newer -ffp-exception-behavior=strict) was attempted in
NumPy 1.21, but was effectively never used.

I am not sure if this is the root problem and I couldn't find a fix or any related issue on GitHub.

priyanshuone6, 2022-01-04:
This might be relevant https://github.com/pybamm-team/PyBaMM/pull/1355

tinosulzer, 2022-01-04:
Thanks for investigating, I commented on the issue asking if we should just skip scikit-odes tests for now (python 3.7 is still working, so we could just do the coverage using that)

PR #1901 (2022-01-14): Add a workaround for the failing tests
# Description

The tests are passing locally but are failing on GitHub Actions (I don't really know why). `cosh` function was behaving weirdly, others are working fine -

```
# func
# casadi.evalf(a), casadi.evalf(b), casadi.evalf(a) - casadi.evalf(b)
<ufunc 'sqrt'>
1.73205 1.73205 0
<ufunc 'tanh'>
0.995055 0.995055 0
<ufunc 'cosh'>
10.0677 10.0677 1.77636e-15
```

Another workaround would be to remove the `cosh` function completely from this test.

Fixes #1882

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github. [... cut]
closed 2022-01-14 by tinosulzer