pybamm-team/PyBaMM #1883: Drive cycle doesn't allow voltage function

Naroek56, 2022-01-04:
Hi everyone,

Here's my setup:
Windows 10
Python 3.8
Pybamm 21.12

When using the drive cycle function with voltage as follows:
`pb.Experiment(["Run Cyc1 (V)"],drive_cycles={Cyc1=vdata})`

I get the following error:
> Parameter provided for 'Voltage function [V]' is of the wrong type (should either be scalar-like or callable)

As an example, vdata looks like this:
> [[ 0.00000000e+00 3.74]
>  [ 1.00000000e+00 3.75]
>  [ 2.00000000e+00 3.76]]
This data format works fine for current, but not voltage. 

I'd like to use voltage cycling for data fitting, so I would really appreciate your help. Thanks in advance!

brosaplanella, 2022-01-06:
Just to double check, if you run 
```python3
pb.Experiment(["Run Cyc1 (A)"],drive_cycles={Cyc1=vdata})
```
it works, right?

Can you post the entire error log to trace where the issue is?

Naroek56, 2022-01-06:
> Just to double check, if you run
> 
> pb.Experiment(["Run Cyc1 (A)"],drive_cycles={Cyc1=vdata})
> it works, right?

Yes. Here is the full error log:

> ---------------------------------------------------------------------------
> KeyError                                  Traceback (most recent call last)
> c:\users\marti\appdata\local\programs\python\python38\lib\site-packages\pybamm\parameters\parameter_values.py in process_symbol(self, symbol)
>     592         try:
> --> 593             return self._processed_symbols[symbol.id]
>     594         except KeyError:
> 
> KeyError: 8839928289061507556
> 
> During handling of the above exception, another exception occurred:
> 
> KeyError                                  Traceback (most recent call last)
> c:\users\marti\appdata\local\programs\python\python38\lib\site-packages\pybamm\parameters\parameter_values.py in process_symbol(self, symbol)
>     592         try:
> --> 593             return self._processed_symbols[symbol.id]
>     594         except KeyError:
> 
> KeyError: 7949390961416403161
> 
> During handling of the above exception, another exception occurred:
> 
> TypeError                                 Traceback (most recent call last)
> <ipython-input-14-8a794a002c45> in <module>
>      13 print(dcycs["Cyc1"])
>      14 
> ---> 15 v=simfit(dcycs,3.7509e-12,1.437675e-12,0)
>      16 plt.plot(v)
>      17 plt.plot(pd.read_csv("Cyc1.csv",header=0)["V"])
> 
> <ipython-input-13-8eb939462fdb> in simfit(dcycs, Drefn, Drefp, R)
>      51                       ,output_variables=["Current [A]","Terminal voltage [V]","X-averaged positive electrode potential [V]","X-averaged positive particle concentration",
>      52                                      "X-averaged negative electrode potential [V]","X-averaged negative particle concentration"])
> ---> 53     sim.solve(calc_esoh=False
>      54               ,solver=pb.CasadiSolver(mode="fast")
>      55 
> 
> c:\users\marti\appdata\local\programs\python\python38\lib\site-packages\pybamm\simulation.py in solve(self, t_eval, solver, check_model, save_at_cycles, calc_esoh, starting_solution, initial_soc, **kwargs)
>     804 
>     805         elif self.operating_mode == "with experiment":
> --> 806             self.build_for_experiment(check_model=check_model)
>     807             if t_eval is not None:
>     808                 pybamm.logger.warning(
> 
> c:\users\marti\appdata\local\programs\python\python38\lib\site-packages\pybamm\simulation.py in build_for_experiment(self, check_model)
>     618                     # It's ok to modify the models in-place as they are not accessible
>     619                     # from outside the simulation
> --> 620                     model_with_set_params = parameter_values.process_model(
>     621                         unbuilt_model, inplace=True
>     622                     )
> 
> c:\users\marti\appdata\local\programs\python\python38\lib\site-packages\pybamm\parameters\parameter_values.py in process_model(self, unprocessed_model, inplace)
>     446                 "Processing parameters for {!r} (algebraic)".format(variable)
>     447             )
> --> 448             new_algebraic[variable] = self.process_symbol(equation)
>     449         model.algebraic = new_algebraic
>     450 
> 
> c:\users\marti\appdata\local\programs\python\python38\lib\site-packages\pybamm\parameters\parameter_values.py in process_symbol(self, symbol)
>     593             return self._processed_symbols[symbol.id]
>     594         except KeyError:
> --> 595             processed_symbol = self._process_symbol(symbol)
>     596             self._processed_symbols[symbol.id] = processed_symbol
>     597 
> 
> c:\users\marti\appdata\local\programs\python\python38\lib\site-packages\pybamm\parameters\parameter_values.py in _process_symbol(self, symbol)
>     720             # process children
>     721             new_left = self.process_symbol(symbol.left)
> --> 722             new_right = self.process_symbol(symbol.right)
>     723             # make new symbol, ensure domain remains the same
>     724             new_symbol = symbol._binary_new_copy(new_left, new_right)
> 
> c:\users\marti\appdata\local\programs\python\python38\lib\site-packages\pybamm\parameters\parameter_values.py in process_symbol(self, symbol)
>     593             return self._processed_symbols[symbol.id]
>     594         except KeyError:
> --> 595             processed_symbol = self._process_symbol(symbol)
>     596             self._processed_symbols[symbol.id] = processed_symbol
>     597 
> 
> c:\users\marti\appdata\local\programs\python\python38\lib\site-packages\pybamm\parameters\parameter_values.py in _process_symbol(self, symbol)
>     700                 function = function_name
>     701             else:
> --> 702                 raise TypeError(
>     703                     "Parameter provided for '{}' ".format(symbol.name)
>     704                     + "is of the wrong type (should either be scalar-like or callable)"
> 
> TypeError: Parameter provided for 'Voltage function [V]' is of the wrong type (should either be scalar-like or callable)

Naroek56, 2022-01-06:
Ah, sorry, I had a typo: It's pb.Experiment(["Run Cyc1 (V)"],drive_cycles={Cyc1:vdata}).
The problem is the same however.

brosaplanella, 2022-03-10:
Hi! Sorry for the delay in my response. Can you try updating to the most recent PyBaMM version and trying again? I tried your code with `pybamm 22.2` and it worked well.

Naroek56, 2022-03-11:
Hi! Yes, it works now, thank for this!

closed 2022-03-11 by Naroek56