pybamm-team/PyBaMM #1889: Scalar timescale

TomTranter, 2022-01-07:
I'd like to implement a way to set a scalar or typical timescale. This is because when solving a jellyroll model or liionpack model if I set electrode height to an input then this changes the timescale along with a few other parameters. It would simplify things to just have a constant value across all the simulations using the model. If I replace the model.timescale with a Scalar though I see different results when changing the scalar value. See below for example:
```
import pybamm

parameter_values = pybamm.ParameterValues("Chen2020")

model1 = pybamm.lithium_ion.SPM()
model2 = pybamm.lithium_ion.SPM()

symbolic_tau = pybamm.LithiumIonParameters().tau_discharge
tau = parameter_values.process_symbol(symbolic_tau)
tau_eval = tau.evaluate()
model2.timescale = pybamm.Scalar(2*tau_eval)

sim1 = pybamm.Simulation(model=model1, parameter_values=parameter_values)
sim2 = pybamm.Simulation(model=model2, parameter_values=parameter_values)

sol1 = sim1.solve([0, 1800])
sol2 = sim2.solve([0, 1800])

pybamm.QuickPlot(solutions=[sol1, sol2]).dynamic_plot()
```

TomTranter, 2022-01-07:
I tried delaying the build of the model until after the timescale is updated but this didn't help
```
import pybamm

parameter_values = pybamm.ParameterValues("Chen2020")

model1 = pybamm.lithium_ion.SPM()
model2 = pybamm.lithium_ion.SPM(build=False)

symbolic_tau = model2.param.tau_discharge
tau = parameter_values.process_symbol(symbolic_tau)
tau_eval = tau.evaluate()

# Double the timescale
# This shouldn't affect the solution as it's used for non-dimensionalising
# But it does
new_timescale = pybamm.Scalar(2*tau_eval)
model2.timescale = new_timescale
model2.param.timescale = new_timescale
model2.param.tau_discharge = new_timescale

model2.build_model()

sim1 = pybamm.Simulation(model=model1, parameter_values=parameter_values)
sim2 = pybamm.Simulation(model=model2, parameter_values=parameter_values)

sim1.build()
sim2.build()

# Makes no difference
sim2.built_model.timescale = new_timescale
sim2.built_model.param.timescale = new_timescale
sim2.built_model.param.tau_discharge = new_timescale

sol1 = sim1.solve([0, 1800])
sol2 = sim2.solve([0, 1800])

plot = pybamm.QuickPlot(solutions=[sol1, sol2])
plot.dynamic_plot()
```

TomTranter, 2022-01-07:
[image]

TomTranter, 2022-01-07:
Running a few more methods that get run in lithium_ion_parameters and base_lithium_ion_model also doesn't help
```
import pybamm

parameter_values = pybamm.ParameterValues("Chen2020")
parameter_values.update({'Current function [A]': "[input]"})

model1 = pybamm.lithium_ion.DFN()
model2 = pybamm.lithium_ion.DFN(build=False)

symbolic_tau = model2.param.tau_discharge
tau = parameter_values.process_symbol(symbolic_tau)
tau_eval = tau.evaluate()

# Double the timescale
# This shouldn't affect the solution as it's used for non-dimensionalising
# But it does
new_timescale = pybamm.Scalar(2*tau_eval)
model2.timescale = new_timescale
model2.param.timescale = new_timescale
model2.param.tau_discharge = new_timescale

model2.param._set_dimensionless_parameters()
model2.param._set_input_current()
model2.set_standard_output_variables()

model2.build_model()

sim1 = pybamm.Simulation(model=model1, parameter_values=parameter_values)
sim2 = pybamm.Simulation(model=model2, parameter_values=parameter_values)

sim1.build()
sim2.build()

sol1 = sim1.solve([0, 1800], inputs={'Current function [A]': 5.0})
sol2 = sim2.solve([0, 1800], inputs={'Current function [A]': 5.0})

plot = pybamm.QuickPlot(solutions=[sol1, sol2])
plot.dynamic_plot()
```

TomTranter, 2022-01-07:
Discussion with @rtimms and @tinosulzer might need to define a tau_discharge and a tau_typical and use the ratio of these in the model equations to make sure non-dimensionalization is consistent and correct parameter is being used.

TomTranter, 2022-01-07:
It's not really a bug as changing the timescale like I am trying to do already is quite hacky but it would be a useful feature to add and has highlighted some potential limitations in the code. Could be a real pain to fix though so might have to try some different hacks elsewhere

tinosulzer, 2022-01-07:
The following works as expected (both models give the same answer), so at least it's not something in the solver
```python
import pybamm

sols = []
for tau in [1, 10]:
    model = pybamm.BaseModel()
    tau = pybamm.Scalar(10)
    model.timescale = tau
    Q = pybamm.Variable("Q")
    model.rhs = {Q: tau} # non-dimensionalized with timescale
    model.initial_conditions = {Q: pybamm.Scalar(1)}
    model.variables = {"Q": Q}
    solver = pybamm.CasadiSolver()
    sol = solver.solve(model, [0, 1])
    sols.append(sol)

pybamm.dynamic_plot(sols, ["Q"])
```

tinosulzer, 2022-01-07:
> Discussion with @rtimms and @tinosulzer might need to define a tau_discharge and a tau_typical and use the ratio of these in the model equations to make sure non-dimensionalization is consistent and correct parameter is being used.

Yes, I think there is a factor "tau_discharge/timescale" (usually equal to 1) missing in the particle equations. The easiest fix might be to redefine `gamma_p` (which already appears in all the appropriate equations), and introduce an equivalent `gamma_n`.

tinosulzer, 2022-01-07:
Fixed, will open a PR later. "timescale" will be an option to pass to the model, and `model.timescale` will be a read-only variable

PR #1893 (2022-01-20): #1889 fix some parameter bugs and add timescale option
# Description

Fixes #1889 

`model.timescale` cannot be changed any more. `model._timescale` can in theory but users really shouldn't. Instead, the "timescale" option can be passed to the model. The simulation results are independent of this (new test added for this).

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [  [... cut]
closed 2022-01-20 by tinosulzer