pybamm-team/PyBaMM #1898: refactor `process` function in BaseSolver to produce "raw" functions with a common signature

martinjrobins, 2022-01-13:
**Summary**
The evaluate signature for expression trees is:
`evaluate(self, t=None, y=None, y_dot=None, inputs=None, known_evals=None) -> result, known_evals`

The call signature used for calls by the solvers are:
`__call__(t, y, inputs) -> result`

in `BaseSolver`, the `process` function is used to convert an pybamm expression tree into one of pybamm.EvaluatorJax, pybamm.EvaluatorPython or Casadi functions, and then wrap them in a `SolverCallable` so that the call signature for the solvers above are satisfied. It also generates "raw" functions (the 1st return value), which can be:
1. the `evaluate` function of a pybamm expression tree
2. the `evaluate` function of a EvaluatorJax or EvaluatorPython class
3. a casadi function

The main issue here is that the signature of 3 is the same as the call signature above, while 1 and 2 use the `evaluate` signature. This is mainly why `SolverCallable` is needed to wrap them in a common interface.

I propose that we:
- remove the 1st "raw" option above, so that these "raw" functions are either one of the evaluator functions, or a casadi function
- ensure that the `evaluate` function of the Evaluator classes use the call signature used by the solvers
- remove the `SolverCallable` class entirely, and have the solvers all use the raw functions directly (which are either the evaluate methods of one of the Evaluator classes, or a casadi function)

 **Motivation**

The motivation is to:
- speed up the calls by the solvers. The `SolverCallable` needs to be generic across all formats, but the raw casadi and evaluator functions have the potential to be faster. Note that this is currently done by the casadi solver anyway, which uses raw casadi functions rather than the `SolverCallable`
- simplify the BaseSolver code: it currently needs to keep track of raw functions (generally casadi function) and `SolverCallable` functions, this way it only keeps track of the raw functions
- allow all solvers access to the raw functions. At the moment only the casadi solver uses the raw casadi functions, but other solvers might need these as well (see #1863)

martinjrobins, 2022-01-13:
@tinosulzer, I think you wrote most of this, so let me know what you think

tinosulzer, 2022-01-13:
I agree this would be a good change for all the reasons specified. We can get rid of `known_evals` entirely at this stage. One additional functionality of the `SolverCallable` class is to call `.full()` after evaluation of casadi objects, but this is definitely sub-optimal in terms of performance so if we can find an alternative that would be great

martinjrobins, 2022-01-13:
great, I'll do this in preparation for #1863 then. 

I saw that call to `.full()`, why is this necessary for calls to rhs/algebraic/residuals? It it possible for casadi to return a sparse vector for these?

tinosulzer, 2022-01-13:
The output type for casadi functions is their own `DM` object, which I haven't found a way to pass directly to the non-casadi solvers

PR #1959 (2022-03-11): I1898 make evaluator functions consistent across casadi, python and jax
# Description

- changes the `process` function in `pybamm.BaseSolver` to return "raw" evaluator functions
- all evalators now use the api expected by a solver, rather than the `evaluate` function used by an expression tree.The evaluators return different types (e.g casadi returns DMs, python returns ndarrays), but this is expected to be handled in each solver.
- this brings in other evaluator functions in line with the casadi evaluator

Fixes #1898 

## Type of change

Please add a line in the relevant section of [CHANGEL [... cut]
closed 2022-03-11 by martinjrobins