pybamm-team/PyBaMM #1902: Events in drive cycles in experiments

rtimms, 2022-01-17:
It would be good to be able to e.g. run a drive cycle in an experiment until a given voltage, like so:

```python3
import pybamm
import pandas as pd
import os

os.chdir(pybamm.__path__[0] + "/..")

pybamm.set_logging_level("INFO")

drive_cycle_current = pd.read_csv(
    "pybamm/input/drive_cycles/US06.csv", comment="#", header=None
).to_numpy()

experiment = pybamm.Experiment(
    [
        "Run US06_A (A) until 3.7 V",
    ],
    drive_cycles={
        "US06_A": drive_cycle_current,
    },
)

model = pybamm.lithium_ion.SPMe()
sim = pybamm.Simulation(model, experiment=experiment, solver=pybamm.CasadiSolver())
sim.solve()

sim.plot()
```

You can do this by changing the min/max voltage parameters, but maybe you want a different stopping condition for the drive cycle. This was prompted by [this discussion](https://github.com/pybamm-team/PyBaMM/discussions/1639).

rtimms, 2022-01-17:
As pointed out by @tinosulzer , this is an issue with how casadi handles events during cycles. See

```python3
import pybamm
import pandas as pd
import os

os.chdir(pybamm.__path__[0] + "/..")

model = pybamm.lithium_ion.SPM()
param = model.default_parameter_values

drive_cycle = pd.read_csv(
    "pybamm/input/drive_cycles/US06.csv", comment="#", header=None
).to_numpy()

timescale = param.evaluate(model.timescale)
current_interpolant = pybamm.Interpolant(
    drive_cycle[:, 0], drive_cycle[:, 1], timescale * pybamm.t
)

param["Current function [A]"] = current_interpolant
param["Lower voltage cut-off [V]"] = 3.6

sim = pybamm.Simulation(model, parameter_values=param)

solvers = [pybamm.ScipySolver(), pybamm.CasadiSolver()]
sols = []
for solver in solvers:
    sol = sim.solve(solver=solver)
    sols.append(sol)

pybamm.dynamic_plot(sols, labels=["scipy", "casadi"])
```

rtimms, 2022-03-17:
Yes, that’s a good workaround for now

tinosulzer, 2022-12-13:
Might be possible to do this with IDAKLU solver now/soon