pybamm-team/PyBaMM #1905: Missing tests for `lithium_metal` full model

Saransh-cpp, 2022-01-20:
Currently, the file [`dfn.py`](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/full_battery_models/lithium_metal/dfn.py), located in `models/full_battery_models/lithium_metal`, is not being tested. The tests should be easy to add as of now but, the model is not available to the users (and I don't know if it has been implemented completely).

AnonymousCodes911, 2022-03-25:
Hey, I would like to work on this, can you briefly explain how, to begin with, it? thank you.

tinosulzer, 2022-03-26:
Not sure yet what to do with these models so will remove the "good first issue" for now

Saransh-cpp, 2024-11-17:
Adding here for the future:

The following line in the conda recipe should be uncommented whenever the module is made public again  - https://github.com/conda-forge/pybamm-feedstock/blob/45a2c99c1c70b9b84966ea88fe1b23ebe2223fd8/recipe/meta.yaml#L59