pybamm-team/PyBaMM #1926: Dockerize `PyBaMM`?

Saransh-cpp, 2022-01-30:
**Summary**
Adding a `Dockerfile` in `PyBaMM` to create a new isolated container for developers. This will come with an option to install all the optional features.

 **Motivation**
 - This will allow developers to set up the `develop` version of `PyBaMM` with a single command.
 - Adding optional features to the local installation will become easier.
 - Developers will be able to choose which optional features they want to be installed.
 - Windows users will be able to access all the optional features.

**Additional context**
- We could also create a wrapper around the `Dockerfile` and do something like -
```
python docker_install.py --install-jax  --install-odes
```
- We could also do this for the users (install `PyBaMM` from `pip` rather than the repository) but I am not sure if users will opt for this.
- Docker - https://www.docker.com/

tinosulzer, 2022-03-10:
The reason we went against docker originally was that, if docker is the only option for installation, it's not easy to combine pybamm with another package (e.g. if the other package also requires its own docker). However, I'd be open to providing a docker image, as long as the lightweight pip install still works well in most cases

ayeankit, 2023-02-20:
Hey @tinosulzer is this still open?

tinosulzer, 2023-02-20:
Yes

ayeankit, 2023-02-20:
Hi @tinosulzer, Please assign me this, I will like to work on this issue.

Saransh-cpp, 2023-02-20:
This might not be a "good first issue".

Also, to familiarize yourself with pybamm, please try to work on a single issue instead of taking up multiple issues (as mentioned here - https://github.com/pybamm-team/PyBaMM/issues/2265#issuecomment-1436625549).

srikanthallu, 2023-02-24:
here is a sample Dockerfile, I have stripped down other packages. 
We could cut down further to bare bones

```
ARG BASE_CONTAINER=jupyter/minimal-notebook
FROM $BASE_CONTAINER
MAINTAINER Srikanth Allu <allus@ornl.gov>

SHELL ["/bin/bash", "-o", "pipefail", "-c"]

USER root

ENV PATH="/opt/conda/bin/:$PATH"

# Install all OS dependencies for fully functional notebook server
RUN apt-get update --yes && \
    apt-get install --yes --no-install-recommends \
    git \
    nano-tiny \
    tzdata \
    unzip \
    vim-tiny \
    # git-over-ssh
    openssh-client \
    texlive-xetex \
    texlive-fonts-recommended \
    texlive-plain-generic \
    xclip && \
    apt-get clean && rm -rf /var/lib/apt/lists/*

# Create alternative for nano -> nano-tiny
RUN update-alternatives --install /usr/bin/nano nano /bin/nano-tiny 10

RUN conda install -c conda-forge pybamm

USER ${NB_UID}

WORKDIR "${HOME}" 
```

Saransh-cpp, 2023-02-25:
Thanks for attaching the Dockerfile, @srikanthallu!

arjxn-py, 2023-03-08:
@Saransh-cpp How about we use [miniconda3](https://hub.docker.com/r/continuumio/miniconda3) as the base image to create a seperate development environment with suitable dependencies? One thing to notice about miniconda is that it provides fast access to the packages which might be able to reduce the buildtime too.

Saransh-cpp, 2023-03-08:
I think a `jupyter` base image makes more sense because that will open up a jupyter notebook automatically inside the docker container. @srikanthallu uses the same above, and I used the same while I was developing the initial Dockerfile -

```Dockerfile
FROM jupyter/minimal-notebook

# ENV PATH=/usr/local/env/bin:"${PATH}"
COPY . pybamm-dev/
WORKDIR $HOME/pybamm-dev/
    
USER root

# install all the required dev dependencies
RUN apt-get -y update && \
    apt-get -y install graphviz

RUN python -m pip --no-cache-dir install --upgrade pip setuptools wheel && \
    python -m pip install -e .[dev,docs]

RUN apt-get -y install libopenblas-dev gfortran cmake && \
    python -m pip install wget && \
    pybamm_install_odes

RUN pybamm_install_jax

WORKDIR $HOME/
```

IIRC the image generated from this file did not work as I hoped it would. Here are the different iterations of this file - https://github.com/Saransh-cpp/PyBaMM/commits/dockerize/scripts/Dockerfile

arjxn-py, 2023-03-31:
```docker
# Base Image
FROM python:3.9-slim-buster

# Set the working directory
WORKDIR /app

# Install the necessary dependencies
RUN apt-get update \
    && apt-get install -y build-essential \
    && apt-get install -y libgmp3-dev libmpfr-dev libmpc-dev \
    && apt-get install -y libffi-dev libssl-dev

# Copy necessary files into the container
COPY setup.py  .
COPY CMakeBuild.py .
COPY README.md .
COPY pybamm/version.py ./pybamm/version.py

# Install PyBaMM
RUN pip install -e ".[dev]"

# Expose the default Jupyter notebook port
EXPOSE 8888

# Start Jupyter notebook on container start
CMD ["jupyter", "notebook", "--ip=0.0.0.0", "--port=8888", "--no-browser", "--allow-root"]
```

@Saransh-cpp @tinosulzer  This is a dockerfile with which I am able to setup the development environment for Pybamm. It has a build time of less than 5 minutes.

Command to build the Docker Image (You can give a name of your own choice) :
```docker
docker build -t pybamm .
```
Command to open up a Jupyter Notebook with Pybamm Development Environment:
```docker
docker run -it -p 8888:8888 pybamm
```

arjxn-py, 2023-03-31:
> The reason we went against docker originally was that, if docker is the only option for installation, it's not easy to combine pybamm with another package (e.g. if the other package also requires [...]

@tinosulzer Combining Pybamm with another packages would not be a big task even after dockerization as one would just have to specify new package in the `setup.py`. And yes `pip install` would still work in most of the cases.
The jupyter notebook in this image is running on the `Development Environment` that I have built using the Dockerfile above specified.
[image]

PR #3223 (2023-08-29): Dockerize PyBaMM
# Description

Please include a summary of the change and which issue is fixed. Please also include relevant motivation and context. List any dependencies that are required for this change.

Fixes #1926 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that sp [... cut]
closed 2023-08-29 by Saransh-cpp