pybamm-team/PyBaMM #1939: Option to pass cell-level thermal properties to thermal models

rtimms, 2022-02-11:
### Description

At the moment we require users to provide thermal properties for each of the cell components and then we compute lumped properties by taking appropriate averages. We should add an option to allow users to pass cell-level thermal properties (e.g. density, specific heat capacity). For the x-lumped pouch cell models, we can also provide the out-of-plane thermal conductivity for the pouch. 

At the moment we compute the parameter "Effective volumetric heat capacity [J.m-3.K-1]" from the densities, specific heat capacities, and thicknesses. We should let users provide this parameter directly.

tinosulzer, 2022-02-14:
I'm still not sure what is the best way to deal with cases where you could specify parameters differently and get the same result. Other examples:
- Current / C-rate
- Initial concentrations / Initial SOC

brosaplanella, 2022-02-15:
Is there an easy way to check which parameter values have been passed when assembling the model parametes? If so, we could have some if statements to choose how things are defined. If both parameters are passed, we could set a hierarchy and throw a warning, or throw an error directly.

rtimms, 2022-02-15:
I think this is how we used to handle C-rate and current. If both were provided we checked they agreed based on the nominal capacity, and raised an error if they didn't.

Maybe the safest option is to not let users provide both at once? At least then there is no confusion.

tinosulzer, 2022-02-15:
Yeah, we got rid of the current + C-rate because it was confusing. In this case, doesn't the user just need to set each component level to be equal to get the correct overall? I guess we could provide a helper function to do that?

rtimms, 2022-02-21:
Yeah, true. Maybe we just leave it as it is for now.