pybamm-team/PyBaMM #1940: [Bug]: Unable to install Scikit.ODEs

amirsh213, 2022-02-15:
### PyBaMM Version

21.12

### Python Version

3.8.10

### Describe the bug

I tried to install Scikit.ODEs via WSL (Ubuntu). The first line `apt install libopenblas-dev` went through but for the second line, it does not find the `/.local` directory:
```
$ pybamm_install_odes
/home/ash213/PyBaMM/pybamm/scikits_odes_setup.log
2022-02-08 20:53:53,552 - [INFO] install_odes.main(123): Starting scikits.odes setup
2022-02-08 20:53:53,556 - [INFO] install_odes.main(143): Looking for sundials at /home/ash213/.local
2022-02-08 20:53:53,556 - [INFO] install_odes.main(143): Looking for sundials at /usr/local
2022-02-08 20:53:53,557 - [INFO] install_odes.main(143): Looking for sundials at /usr
2022-02-08 20:53:53,558 - [INFO] install_odes.main(153): Could not find sundials libraries.
Traceback (most recent call last):
  File "/home/ash213/.local/bin/pybamm_install_odes", line 11, in <module>
    load_entry_point('pybamm', 'console_scripts', 'pybamm_install_odes')()
  File "/home/ash213/PyBaMM/pybamm/install_odes.py", line 154, in main
    logger.info("Installing sundials in {}".install_dir)
AttributeError: 'str' object has no attribute 'install_dir'
```
When I wanted to run the last line of the instruction, I got the error below:

<details>
<summary>error</summary>

```
$ pip install scikits.odes
Defaulting to user installation because normal site-packages is not writeable
Collecting scikits.odes
  Using cached scikits.odes-2.6.3.tar.gz (211 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: scipy in ./.local/lib/python3.8/site-packages (from scikits.odes) (1.7.3)
Requirement already satisfied: numpy<1.23.0,>=1.16.5 in ./.local/lib/python3.8/site-packages (from scipy->scikits.odes) (1.22.1)
Building wheels for collected packages: scikits.odes
  Building wheel for scikits.odes (pyproject.toml) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /usr/local/lib/python3.8/dist-packages/pip/_vendor/pep517/in_process/_in_process.py build_wheel /tmp/tmpk6w41t8n
       cwd: /tmp/pip-install-js_s6uni/scikits-odes_e262af391a8f46c5a2919763e7415d5c
  Complete output (560 lines):
  running bdist_wheel
  running build
  running config_cc
  INFO: unifing config_cc, config, build_clib, build_ext, build commands --compiler options
  running config_fc
  INFO: unifing config_fc, config, build_clib, build_ext, build commands --fcompiler options
  running build_src
  INFO: build_src
  INFO: building extension "scikits.odes.ddaspk" sources
  INFO: f2py options: []
  INFO: f2py: scikits/odes/ddaspk.pyf
  Reading fortran codes...
        Reading file 'scikits/odes/ddaspk.pyf' (format:free)
  Post-processing...
        Block: ddaspk__user__routines
                Block: ddaspk_user_interface
                        Block: res
                        Block: jac
        Block: ddaspk
                        Block: ddaspk
  In: scikits/odes/ddaspk.pyf:ddaspk:unknown_interface:ddaspk
  get_useparameters: no module ddaspk__user__routines info used by ddaspk
  Post-processing (stage 2)...
[... 514 lines cut ...]
  ########### EXT COMPILER OPTIMIZATION ###########
  INFO: Platform      :
    Architecture: x64
    Compiler    : gcc

  CPU baseline  :
    Requested   : 'min'
    Enabled     : SSE SSE2 SSE3
    Flags       : -msse -msse2 -msse3
    Extra checks: none

  CPU dispatch  :
    Requested   : 'max -xop -fma4'
    Enabled     : SSSE3 SSE41 POPCNT SSE42 AVX F16C FMA3 AVX2 AVX512F AVX512CD AVX512_KNL AVX512_KNM AVX512_SKX AVX512_CLX AVX512_CNL AVX512_ICL
    Generated   : none
  INFO: CCompilerOpt.cache_flush[817] : write cache to path -> /tmp/pip-install-js_s6uni/scikits-odes_e262af391a8f46c5a2919763e7415d5c/build/temp.linux-x86_64-3.8/ccompiler_opt_cache_ext.py
  error: extension 'scikits.odes.ddaspk' has Fortran sources but no Fortran compiler found
  ----------------------------------------
  ERROR: Failed building wheel for scikits.odes
Failed to build scikits.odes
ERROR: Could not build wheels for scikits.odes, which is required to install pyproject.toml-based projects
WARNING: You are using pip version 21.3.1; however, version 22.0.3 is available.
You should consider upgrading via the '/usr/bin/python3 -m pip install --upgrade pip' command.
```
</details>

Saransh-cpp, 2022-02-16:
@amirsh213 this has been fixed in the develop branch and should be available in the next PyBaMM version. For now, you can either install PyBaMM through the develop branch or, you can change the contents of your local install_odes.py file as in #1916.

Note: This script automates the installation process. You can also install the dependencies (SUNDIALS) manually. As for the second error, scikit.odes requires an environment variable SUNDIALS_INST pointing to the location of installed SUNDIALS (For ex: SUNDIALS_INST=/usr/local/lib/sundials-4.0.2/). See https://github.com/bmcage/odes/issues/90 for reference.

amirsh213, 2022-02-16:
I am using the develop version in Ubuntu but still getting the error. I used the instruction for sundials but got another error. The installation process for sundials is not straightforward and needs a lot of prerequisites. I prefer to wait for the next version of pybamm if it is anytime soon. Hopefully, it makes it work.

Saransh-cpp, 2022-02-17:
@amirsh213 you should not be getting the error -
```
$ pybamm_install_odes
/home/ash213/PyBaMM/pybamm/scikits_odes_setup.log
2022-02-08 20:53:53,552 - [INFO] install_odes.main(123): Starting scikits.odes setup
2022-02-08 20:53:53,556 - [INFO] install_odes.main(143): Looking for sundials at /home/ash213/.local
2022-02-08 20:53:53,556 - [INFO] install_odes.main(143): Looking for sundials at /usr/local
2022-02-08 20:53:53,557 - [INFO] install_odes.main(143): Looking for sundials at /usr
2022-02-08 20:53:53,558 - [INFO] install_odes.main(153): Could not find sundials libraries.
Traceback (most recent call last):
  File "/home/ash213/.local/bin/pybamm_install_odes", line 11, in <module>
    load_entry_point('pybamm', 'console_scripts', 'pybamm_install_odes')()
  File "/home/ash213/PyBaMM/pybamm/install_odes.py", line 154, in main
    logger.info("Installing sundials in {}".install_dir)
AttributeError: 'str' object has no attribute 'install_dir'
```
when using the develop branch. This was fixed and is visible in the `install_odes.py` file -
https://github.com/pybamm-team/PyBaMM/blob/214a4b1e8fc5da6dc42211572454f31671ac1af2/pybamm/install_odes.py#L158

You can try the latest version of `develop`, or you can wait for the next release which should be soon! :)

amirsh213, 2022-03-03:
Thanks for the update! But I'm still getting the error below. 
By the way, it still needs WSL and does not work on Windows, right?

<details>
<summary>error</summary>

```
pybamm_install_odes
/home/ash213/.local/lib/python3.8/site-packages/pybamm/scikits_odes_setup.log
2022-03-03 11:26:31,657 - [INFO] install_odes.main(125): Starting scikits.odes setup
2022-03-03 11:26:31,660 - [INFO] install_odes.main(147): Looking for sundials at /home/ash213/.local
2022-03-03 11:26:31,661 - [INFO] install_odes.main(147): Looking for sundials at /usr/local
2022-03-03 11:26:31,663 - [INFO] install_odes.main(147): Looking for sundials at /usr
2022-03-03 11:26:31,664 - [INFO] install_odes.main(157): Could not find sundials libraries.
2022-03-03 11:26:31,665 - [INFO] install_odes.main(158): Installing sundials in /home/ash213/.local
cmake version 3.16.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).
2022-03-03 11:26:31,764 - [INFO] install_odes.install_sundials(49): Downloading sundials
100% [........................................................................] 18496945 / 18496945
-
-
-
-
-
-
-
-
-
- Creating build dir ----------------------------------------
---------- Running CMake prepare ----------------------------------------
-- The C compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
CMake Warning at config/SundialsCMakeMacros.cmake:83 (message):
  ------------------------------------------------------------------------

  WARNING: EXAMPLES_ENABLE is deprecated and will be removed in the future.

  Copying value to EXAMPLES_ENABLE_C.

  ------------------------------------------------------------------------
Call Stack (most recent call first):
  config/SundialsDeprecated.cmake:20 (print_warning)
  config/SundialsDeprecated.cmake:25 (print_deprecated)
  CMakeLists.txt:166 (include)

-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of int32_t
-- Check size of int32_t - done
-- Using int32_t for indices
-- Looking for _POSIX_TIMERS
-- Looking for _POSIX_TIMERS - found
-- The Fortran compiler identification is unknown
CMake Error at config/SundialsFortran.cmake:38 (enable_language):
  No CMAKE_Fortran_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "FC" or the CMake cache entry CMAKE_Fortran_COMPILER to the full
  path to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
[... 616 lines cut ...]
      ########### EXT COMPILER OPTIMIZATION ###########
      INFO: Platform      :
        Architecture: x64
        Compiler    : gcc

      CPU baseline  :
        Requested   : 'min'
        Enabled     : SSE SSE2 SSE3
        Flags       : -msse -msse2 -msse3
        Extra checks: none

      CPU dispatch  :
        Requested   : 'max -xop -fma4'
        Enabled     : SSSE3 SSE41 POPCNT SSE42 AVX F16C FMA3 AVX2 AVX512F AVX512CD AVX512_KNL AVX512_KNM AVX512_SKX AVX512_CLX AVX512_CNL AVX512_ICL
        Generated   : none
      INFO: CCompilerOpt.cache_flush[817] : write cache to path -> /tmp/pip-install-inqe0hv1/scikits-odes_7891a9af0911414488be96f4e9b47721/build/temp.linux-x86_64-3.8/ccompiler_opt_cache_ext.py
      error: extension 'scikits.odes.ddaspk' has Fortran sources but no Fortran compiler found
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for scikits.odes
Failed to build scikits.odes
ERROR: Could not build wheels for scikits.odes, which is required to install pyproject.toml-based projects
```
</details>

Saransh-cpp, 2022-03-03:
Yes, this still requires wsl.

As for the error -
```
error: extension 'scikits.odes.ddaspk' has Fortran sources but no Fortran compiler found
```
the installation requires a Fortran compiler. Run the following command to install `gfortran` -
```
sudo apt-get install gfortran
```
and then try `pybamm_install_odes` again.

amirsh213, 2022-03-03:
Thanks, I installed it successfully but now I cannot use it. I'm getting the error below for a CC discharge. Am I using it for the wrong application?
I'm also wondering what the advantage of ScipySolver is over the CasadiSolver?

<details>
<summary>error</summary>

````
---------------------------------------------------------------------------
SolverError                               Traceback (most recent call last)
Input In [6], in <module>
      9 # create our dictionary 
     10 #var_pts = {
     11  #   "x_n": 10,  # negative electrode
   (...)
     15  #   "r_p": 10,  # positive particle
     16 #}
     17 sim = pybamm.Simulation(model, parameter_values=parameter_values, solver=solver)
---> 19 sim.solve([0, 1540])
     20 sim.plot()

File ~/.local/lib/python3.8/site-packages/pybamm/simulation.py:803, in Simulation.solve(self, t_eval, solver, check_model, save_at_cycles, calc_esoh, starting_solution, initial_soc, **kwargs)
    788             if dt_eval_max > dt_data_min + sys.float_info.epsilon:
    789                 warnings.warn(
    790                     """
    791                     The largest timestep in t_eval ({}) is larger than
   (...)
    800                     pybamm.SolverWarning,
    801                 )
--> 803     self._solution = solver.solve(self.built_model, t_eval, **kwargs)
    805 elif self.operating_mode == "with experiment":
    806     self.build_for_experiment(check_model=check_model)

File ~/.local/lib/python3.8/site-packages/pybamm/solvers/base_solver.py:903, in BaseSolver.solve(self, model, t_eval, external_variables, inputs, initial_conditions, nproc, calculate_sensitivities)
    896 timer = pybamm.Timer()
    897 if model not in self.models_set_up:
    898     # It is assumed that when len(inputs_list) > 1, model set
    899     # up (initial condition, time-scale and length-scale) does
    900     # not depend on input parameters. Thefore only `ext_and_inputs[0]`
    901     # is passed to `set_up`.
    902     # See https://github.com/pybamm-team/PyBaMM/pull/1261
--> 903     self.set_up(model, ext_and_inputs_list[0], t_eval)
    904     self.models_set_up.update(
    905         {model: {"initial conditions": model.concatenated_initial_conditions}}
    906     )
    907 else:

File ~/.local/lib/python3.8/site-packages/pybamm/solvers/base_solver.py:115, in BaseSolver.set_up(self, model, inputs, t_eval, ics_only)
    113 # Check model.algebraic for ode solvers
    114 if self.ode_solver is True and len(model.algebraic) > 0:
--> 115     raise pybamm.SolverError(
    116         "Cannot use ODE solver '{}' to solve DAE model".format(self.name)
    117     )
    118 # Check model.rhs for algebraic solvers
    119 if self.algebraic_solver is True and len(model.rhs) > 0:

SolverError: Cannot use ODE solver 'Scipy solver (BDF)' to solve DAE model
```
</details>

Saransh-cpp, 2022-03-04:
No idea about this. I think @rtimms might be able to help (as you were having a similar discussion in #1639)

rtimms, 2022-03-04:
The `ScipySolver` can only solve ODE models (e.g. SPM, SPMe). To use the scikits dae sovler you need to you `pybamm.ScikitsDaeSolver`

closed 2022-03-05 by amirsh213