pybamm-team/PyBaMM #1970: ScipySolver issue

amirsh213, 2022-03-08:
### PyBaMM Version

22.2

### Python Version

3.8.10

### Describe the bug

I installed the ScipySolver hoping to get rid of the CasadiSolver error on events but the solver still struggles. I was wondering if there was any update on this issue on the new release? If not, what other options do I have to handle events when the voltage hits upper/lower cutoffs? 
I am simulating an HPPC test (using current interpolant) that hits both upper and lower voltages and the solver stops at the lower cutoff. Rebuilding the input current as "Experiment" is not an option because it's very time taking and I need the exact current from my experiment.
Thanks.

tinosulzer, 2022-03-09:
Please post the script you are using to help diagnose this error

amirsh213, 2022-03-10:
Below is the script when I use "Experiment" for defining HPPC. I defined my own parameters using Chen2020. 
```
# Defining HPPC via experiment
experiment = pybamm.Experiment(
    [
        (
            "Rest for 1 hour",
            "Discharge at 1C for 30 seconds or until 2.5 V",
            "Rest for 40 seconds ",
            "Charge at 0.75C for 10 seconds",
            "Rest for 10 seconds",
            "Discharge at C/3 for 1014.5 seconds or until 2.5 V ",
        ),
    ]
    * 10
)

# Solving for experimental current
parameter_values["Initial concentration in negative electrode [mol.m-3]"] = 0.8*31329
parameter_values["Initial concentration in positive electrode [mol.m-3]"] = 0.15*49956

model = pybamm.lithium_ion.SPMe()

#solver= pybamm.CasadiSolver(mode="safe")
solver= pybamm.ScipySolver()
#solver= pybamm.ScikitsDaeSolver()

simulation = pybamm.Simulation(model, experiment=experiment, parameter_values=parameter_values, solver=solver)

# simulate 
simulation.solve()

# plot
simulation.plot(["Current [A]", "Terminal voltage [V]"])
```

tinosulzer, 2022-03-10:
I ran this
```python3
import pybamm

experiment = pybamm.Experiment(
    [
        (
            "Rest for 1 hour",
            "Discharge at 1C for 30 seconds or until 2.5 V",
            "Rest for 40 seconds ",
            "Charge at 0.75C for 10 seconds",
            "Rest for 10 seconds",
            "Discharge at C/3 for 1014.5 seconds or until 2.5 V ",
        ),
    ]
    * 10
)

# Solving for experimental current
parameter_values = pybamm.ParameterValues("Chen2020")
parameter_values["Initial concentration in negative electrode [mol.m-3]"] = 0.8 * 31329
parameter_values["Initial concentration in positive electrode [mol.m-3]"] = 0.15 * 49956

model = pybamm.lithium_ion.SPMe()

# solver= pybamm.CasadiSolver(mode="safe")
solver = pybamm.ScipySolver()
# solver= pybamm.ScikitsDaeSolver()

simulation = pybamm.Simulation(
    model, experiment=experiment, parameter_values=parameter_values, solver=solver
)

# simulate
simulation.solve()

# plot
simulation.plot(["Current [A]", "Terminal voltage [V]"])
```
and got a different error
```
2022-03-09 19:31:06,290 - [WARNING] simulation.solve(929): 

        Experiment is infeasible: 'event: Maximum voltage' was triggered during 'Discharge at 1C for 30 seconds or until 2.5 V'. The returned solution only contains the first 1 cycles. Try reducing the current, shortening the time interval, or reducing the period.
```
This seems to indicate the initial conditions you chose make the initial voltage too high.
Using the normal initial conditions the script works fine.
If you can post the parameters you used I will be able to see why you got that error

amirsh213, 2022-03-10:
It works fine with default Chen parameters. My initial conditions result in voltage lower than 4.2 V. I set the upper cutoff to 4.4 V and tried a different initial condition but it still raises the same error. If I use the drive cycle I don't get this error and I see that the initial voltage is lower than 4.2 V but it seems that it is not reading the current. I changed almost all parameters and functions. 

[image]

tinosulzer, 2022-03-10:
Please could you provide a [MWE](https://stackoverflow.com/help/minimal-reproducible-example) so that I can see what's causing the error? Otherwise it won't be possible to fix.

amirsh213, 2022-03-10:
How can I share all of the parameters? because you already have the script.

tinosulzer, 2022-03-10:
Thanks, I've found the bug and opened a PR fixing it

PR #1976 (2022-03-11): #1970 fix bug
# Description

Fixes #1970 . Bug was introduced in #1805 but not caught because of the way this part of the code was being tested.

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [x] Bug fix (non-breaking change which fixes an issue)

 [... cut]
closed 2022-03-11 by tinosulzer