pybamm-team/PyBaMM #1973: print_(evaluated)_parameters doesn't work

dion-w, 2022-03-09:
### Discussed in https://github.com/pybamm-team/PyBaMM/discussions/1972

<div type='discussions-op-text'>

<sup>Originally posted by **dion-w** March  9, 2022</sup>
Hello everyone,
I know when saving simulation results there are different options. You can either save as pkl (which can be up to a few gigabytes quite easily) or export only a couple of variables with matlab.  The export of the variables is great and works perfectly.

For the parameters though i have difficulty understanding what needs to be done here. I see there are some methods like `param.print_parameters(sol.all_models[0].param,'my_output.csv')` 
for param being from pybamm.parameters.parameter_values.py but when i try this it fails to evaluate the spatialvariable x_p as it says this method is not implemented. 

When i try to use a parameter to be evaluated like 
`param.print_evaluated_parameters(param,'my_output.csv')`  
and let it be exported it says it "cannot unpack non-iterable function object". 

The only thing that works is
param.export_csv('my_output.csv')
but sadly this does not evaluate function parameters so this is no use for a parameter study where for example the particle radius changes during runs, and you want to know which run has which particle size.

I mean, maybe i'm doing this wrong but i found nothing helpful in the source code or the documentation for this matter.</div>

tinosulzer, 2022-03-09:
If you just care about a single parameter, the easiest thing to do would be: `parameter_values.evaluate(model.param.R_n_typ)`. For some parameters this will not work because they are not constant scalars but have spatial distributions, and the `parameter_values` don't know about the discretization. 
An alternative is to add the parameter to the variables dictionary at the start:
`model.variables["Radius [m]"] = model.param.R_n_dimensional`.

Also, for me, `parameter_values.print_parameters(model.param, "output.txt")` works fine. Does that work for you?

dion-w, 2022-03-10:
makes sense, that parameter_values doesn't know about discretization. `parameter_values.evaluate(model.param.R_n_typ)` works for scalars. This is nice, so thank you! 
About your suggestion with declaration of variables: this works like a charm for scalars, unfortunately not for FunctionParameters (tried with `model.variables["my negative electrode conductivity"] = model.param.sigma_n_dimensional` )since it gives the error 

> Traceback (most recent call last):
>   File "...\PyBaMM\pybamm\parameters\parameter_values.py", line 603, in process_symbol
>     return self._processed_symbols[symbol.id]
> AttributeError: 'function' object has no attribute 'id'

This would have been nice but it's ok for now. Thank you very much. :) 

`parameter_values.print_parameters(model.param, "output.txt")` sadly doesn't work me as it still fails to evaluate the spatialvariable x_p. It says this method is not implemented.. I will see if this is due to my version. I changed the source code to make the initial utilisation a FunctionParameter since i thought this would greatly benefit graded electrodes with different particle sizes along the thickness. I'll see if i messed something up when i check out a new clean version.

tinosulzer, 2022-03-10:
Can you post the code that fails `.print_parameters`?

tinosulzer, 2022-03-20:
I'd like to fix this bug but need the failing code to make sure it's properly fixed @dion-w

tinosulzer, 2022-08-02:
@dion-w bump (if you still have the code)

tinosulzer, 2022-10-07:
Closing due to no reply

closed 2022-10-07 by tinosulzer