pybamm-team/PyBaMM #1986: Remove zero-concentration cut-off event from DFN

tinosulzer, 2022-03-17:
- add `cutoff_event` kwarg (boolean, default False) to the full electrolyte model
- in `set_events` in the full electrolyte model, don't add the event if `cutoff_event` is `False`
- make `True` for the SPMe (for cut-off, see discussion in #1982 )
- make the cut-off tolerance smaller when `cutoff_event` is True

Easy issue, assuming the models still work without the event

brosaplanella, 2022-03-17:
I think the lithium plating still might need some fixes to ensure that `c_e` doesn't go negative in the reaction.

tinosulzer, 2022-03-17:
Ah yeah and the `log(c_e)` in the electrolyte potential too?

dion-w, 2022-03-20:
Can i work on this? 

Also since it's my first time working with other people on a public repo: is there a timespan where results are expected? I want to see this problem resolved and would work on it in the next few weeks.

tinosulzer, 2022-03-20:
Yes, go for it. There is no expected timespan, it is just down to whoever is motivated to do it first

dion-w, 2022-03-29:
I found a rather neat solution for this. It involves exactly what is mentioned in #1982 . In `base_electrolyte_diffusion` heaviside functions are introduced for the c_e variable setting every value zero to 1e-8. This way the jacobians won't have NaN-values, the calculation is therefore faster and a lot of issues mentioned in #1992 don't happen. I guess these were edge case for the exact szenario that a variable becomes zero. I will soon perform the tests and create a PR on this.

brosaplanella, 2022-03-30:
I think the standard DFN should work just by dropping the zero concentration event, so that might be a good benchmark to see if these changes are required.

PR #2207 (2022-08-05): #1986 remove zero electrolyte concentration cut-off
# Description

Removes the "Zero electrolyte concentration cut-off" event.
Adds some bounds to functions so that the solver still works even if we step into electrolyte depletion

Fixes #1986 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization  [... cut]
closed 2022-08-05 by tinosulzer