pybamm-team/PyBaMM #1994: Add more doctests to PyBaMM

Saransh-cpp, 2022-03-23:
### Description

It would be nice to have examples in docstrings that are tested periodically. For example, the `Simulation` class could have something like this -

```
>>> spm = pybamm.lithium_ion.SPM()
<some output>
>>> sim = pybamm.Simulation(spm)
<some output>
>>> sim.solve([0, 10, 2])
<some output>
```

### Motivation

Would make minor examples available in the documentation itself and would make sure that the docs are not outdated.

parthxtripathi, 2022-03-23:
Hey @Saransh-cpp , Can i work on this issue?

Saransh-cpp, 2022-03-23:
Yes! Go ahead!

tinosulzer, 2022-09-27:
Close or keep open @Saransh-cpp ?

Saransh-cpp, 2022-09-27:
With time I have realized that maybe doctests are not that feasible for PyBaMM. Most of PyBaMM's functionality is designed to integrate together (if I am not wrong), which makes it a bit tough to write standalone (and short) doctests. For instance, writing doctests for `solve` would require building a model first. Also, the doctests added after this issue was created do not test anything substantial, so I think it should be okay to close this issue (?)

tinosulzer, 2022-09-27:
Yeah, I agree

closed 2022-09-27 by tinosulzer