pybamm-team/PyBaMM #2001: Step method in simulation class with experiment load

presvetli, 2022-03-31:
### PyBaMM Version

22.2

### Python Version

3.9.5

### Describe the bug

I tried run analysis by step method in Simulation class. 

1st error:
In simulation.py script line 960, self.build_model is 'None' for my case because method self.build() is not run(line 953)
        self._solution = solver.step(
            starting_solution, self.built_model, dt, npts=npts, save=save, **kwargs
        )

Solution : change self.build_model into self.model or put in list  ''with experiment'' in line 952.

2nd issue:
When analysis is done, on plots I can not see experiment load.
[image]

If I increase dt from 3600[s] to true duration of cycle I got [SolverError: Maximum number of decreased steps occurred at t=0.0. Try solving the model up to this time only or reducing dt_max (currently, dt_max=2272.5).]()

### Steps to Reproduce

import pybamm
lithium_plating = "okane2020_Li_plating"
param = pybamm.ParameterValues('Chen2020')

model = pybamm.lithium_ion.DFN({"SEI": "reaction limited","lithium plating": "reversible"},name="reaction limited")
chemistry = getattr(pybamm.parameter_sets, 'Chen2020')
if lithium_plating:
    chemistry["lithium plating"]  = lithium_plating
    param = pybamm.ParameterValues(chemistry)
# Experiment
experiment = pybamm.Experiment(
    [
        (
            "Discharge at 0.5 C until 3.2 V",
            "Charge at 1 C until 4.2 V",
        )
    ]
)

# create geometry
geometry = model.default_geometry

# load parameter values and process model and geometry
param.process_model(model)
param.process_geometry(geometry)
# set mesh
mesh = pybamm.Mesh(geometry, model.default_submesh_types, model.default_var_pts)
# discretise model
disc = pybamm.Discretisation(mesh, model.default_spatial_methods)
disc.process_model(model)

# Solver
dt_max = 2
solver = pybamm.CasadiSolver(mode="safe", dt_max=dt_max)
sim = pybamm.Simulation(model, experiment=experiment, parameter_values=param)
dt = 3600     
for cycle in range(10):
    step_solution = sim.step(dt, solver=solver)

### Relevant log output

```shell
AttributeError                            Traceback (most recent call last)
/mnt/e/AU01007/electrochemical-degradation-model/Test_files/Simulation_step.ipynb Cell 4' in <cell line: 6>()
      5 dt = 360     
      6 for cycle in range(10):
----> 7     step_solution = sim.step(dt, solver=solver)
     10 if step_solution:
     11     sim.plot(
     12         [
     13             "Current [A]",
   (...)
     27             ],
     28     ])

File /usr/local/lib/python3.9/dist-packages/pybamm/simulation.py:960, in Simulation.step(self, dt, solver, npts, save, starting_solution, **kwargs)
    957 if starting_solution is None:
    958     starting_solution = self._solution
--> 960 self._solution = solver.step(
    961     starting_solution, self.built_model, dt, npts=npts, save=save, **kwargs
    962 )
    964 return self.solution

File /usr/local/lib/python3.9/dist-packages/pybamm/solvers/base_solver.py:1189, in BaseSolver.step(self, old_solution, model, dt, npts, external_variables, inputs, save)
   1186     return old_solution
   1188 # Make sure model isn't empty
-> 1189 if len(model.rhs) == 0 and len(model.algebraic) == 0:
   1190     if not isinstance(self, pybamm.DummySolver):
   1191         raise pybamm.ModelError(
   1192             "Cannot step empty model, use `pybamm.DummySolver` instead"
   1193         )

AttributeError: 'NoneType' object has no attribute 'rhs'
```

tinosulzer, 2022-03-31:
Stepping isn't supported with experiments. Trying to figure out if we should support it: why are you wanting to step and not solve?

presvetli, 2022-04-01:
ok, thank you for your fast replay.
Reason why I want use step is because I want to change some parameter  in some equation. 
This parameter depends of  "Negative electrode surface potential difference" variable.

I would rather use solve, but I didn't found way how to check "Negative electrode surface potential difference" variable **value** in parameter function to check is it lower or higher from 0.
Or there is another way to adjust some equation based on **some variable sign** with boundary condition (<,>)?

tinosulzer, 2022-04-01:
> Or there is another way to adjust some equation based on some variable sign with boundary condition (<,>)

Yes you can just multiply by e.g. `(delta_phi > 0)`. This is much more robust than step and check

presvetli, 2022-04-04:
I tested it, and it works well. Thank you very much.

closed 2022-04-04 by tinosulzer