pybamm-team/PyBaMM #2012: pybamm_install_odes fails

HOLL95, 2022-04-07:
### PyBaMM Version

22.3

### Python Version

3.9.10

### Describe the bug

Running pybamm_install_odes on fedora 34 correctly install sundials, but fails at pip install scikits.odes, even when $SUNDIALS_INST is set to the correct path

```
echo $SUNDIALS_INST  ~/.local/include/sundials/

### Steps to Reproduce

```pip3 install --user scikits.odes

### Relevant log output

```shell
pip3 install scikits.odes --user
Collecting scikits.odes
  Using cached scikits.odes-2.6.3.tar.gz (211 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: scipy in /usr/lib64/python3.9/site-packages (from scikits.odes) (1.6.2)
Requirement already satisfied: numpy<1.23.0,>=1.16.5 in /usr/lib64/python3.9/site-packages (from scipy->scikits.odes) (1.20.1)
Building wheels for collected packages: scikits.odes
  Building wheel for scikits.odes (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for scikits.odes (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [2379 lines of output]
      /tmp/pip-build-env-0w1s9h1b/overlay/lib/python3.9/site-packages/setuptools/dist.py:757: UserWarning: Usage of dash-separated 'description-file' will not be supported in future versions. Please use the underscore name 'description_file' instead
        warnings.warn(
      running bdist_wheel
      running build
      running config_cc
      INFO: unifing config_cc, config, build_clib, build_ext, build commands --compiler options
      running config_fc
      INFO: unifing config_fc, config, build_clib, build_ext, build commands --fcompiler options
      running build_src
      INFO: build_src
      INFO: building extension "scikits.odes.ddaspk" sources
      creating build
      creating build/src.linux-x86_64-3.9
      creating build/src.linux-x86_64-3.9/scikits
      creating build/src.linux-x86_64-3.9/scikits/odes
      INFO: f2py options: []
      INFO: f2py: scikits/odes/ddaspk.pyf
      append_needs: unknown need 'int'
      append_needs: unknown need 'double'
      append_needs: unknown need 'double'
      append_needs: unknown need 'double'
      append_needs: unknown need 'double'
      append_needs: unknown need 'double'
      append_needs: unknown need 'int'
      append_needs: unknown need 'double'
      append_needs: unknown need 'int'
      append_needs: unknown need 'int'
      append_needs: unknown need 'double'
      append_needs: unknown need 'double'
      append_needs: unknown need 'double'
      append_needs: unknown need 'double'
      append_needs: unknown need 'double'
      append_needs: unknown need 'double'
[... 384 lines cut ...]
      error: Command "gcc -pthread -Wno-unused-result -Wsign-compare -DDYNAMIC_ANNOTATIONS_ENABLED=1 -DNDEBUG -O2 -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -O2 -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -O2 -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -fPIC -I/users/henney/.local/include/sundials/include -I/tmp/pip-build-env-0w1s9h1b/overlay/lib64/python3.9/site-packages/numpy/core/include -Ibuild/src.linux-x86_64-3.9/numpy/distutils/include -I/usr/include/python3.9 -c scikits/odes/sundials/common_defs.c -o build/temp.linux-x86_64-3.9/scikits/odes/sundials/common_defs.o -MMD -MF build/temp.linux-x86_64-3.9/scikits/odes/sundials/common_defs.o.d -msse -msse2 -msse3" failed with exit status 1
      INFO:
      ########### EXT COMPILER OPTIMIZATION ###########
      INFO: Platform      :
        Architecture: x64
        Compiler    : gcc
      
      CPU baseline  :
        Requested   : 'min'
        Enabled     : SSE SSE2 SSE3
        Flags       : -msse -msse2 -msse3
        Extra checks: none
      
      CPU dispatch  :
        Requested   : 'max -xop -fma4'
        Enabled     : SSSE3 SSE41 POPCNT SSE42 AVX F16C FMA3 AVX2 AVX512F AVX512CD AVX512_KNL AVX512_KNM AVX512_SKX AVX512_CLX AVX512_CNL AVX512_ICL
        Generated   : none
      INFO: CCompilerOpt.cache_flush[817] : write cache to path -> /tmp/pip-install-6f0e2qao/scikits-odes_0fc3d1c477914388b7caea9450bec839/build/temp.linux-x86_64-3.9/ccompiler_opt_cache_ext.py
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for scikits.odes
Failed to build scikits.odes
ERROR: Could not build wheels for scikits.odes, which is required to install pyproject.toml-based projects
```

Saransh-cpp, 2022-04-07:
@HOLL95, could you please attach the output of `pybamm_install_odes` and not `pip3 install scikits.odes --user`

HOLL95, 2022-04-07:
pybamm_install_odes 
/users/henney/.local/lib/python3.9/site-packages/pybamm/scikits_odes_setup.log
2022-04-07 17:11:58,992 - [INFO] install_odes.main(125): Starting scikits.odes setup
2022-04-07 17:11:58,993 - [INFO] install_odes.main(147): Looking for sundials at /users/henney/.local
2022-04-07 17:11:58,993 - [INFO] install_odes.main(147): Looking for sundials at /usr/local
2022-04-07 17:11:58,993 - [INFO] install_odes.main(147): Looking for sundials at /usr
2022-04-07 17:11:58,993 - [INFO] install_odes.main(157): Could not find sundials libraries.
2022-04-07 17:11:58,993 - [INFO] install_odes.main(158): Installing sundials in /users/henney/.local
cmake version 3.20.5

CMake suite maintained and supported by Kitware (kitware.com/cmake).
2022-04-07 17:11:59,007 - [INFO] install_odes.install_sundials(49): Downloading sundials
100% [........................................................................] 18496945 / 18496945---------- Running CMake prepare ----------------------------------------
CMake Warning at config/SundialsCMakeMacros.cmake:83 (message):
  ------------------------------------------------------------------------

  WARNING: EXAMPLES_ENABLE is deprecated and will be removed in the future.

  Copying value to EXAMPLES_ENABLE_C.

  ------------------------------------------------------------------------
Call Stack (most recent call first):
  config/SundialsDeprecated.cmake:20 (print_warning)
  config/SundialsDeprecated.cmake:25 (print_deprecated)
  CMakeLists.txt:166 (include)

-- Using int32_t for indices
CMake Warning at config/SundialsCMakeMacros.cmake:83 (message):
  ------------------------------------------------------------------------

  WARNING: Fortran compiler does not support F90

  F90 support will not be provided

  ------------------------------------------------------------------------
Call Stack (most recent call first):
  config/SundialsFortran.cmake:48 (print_warning)
  CMakeLists.txt:795 (include)

-- Determining Fortran name-mangling scheme... OK
-- Looking for LAPACK libraries... OK
CMake Deprecation Warning at /auto/users/henney/.local/lib/python3.9/site-packages/sundials/build_sundials/LapackTest/CMakeLists.txt:1 (CMAKE_MINIMUM_REQUIRED):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Checking if LAPACK works... OK
-- Added NVECTOR_SERIAL module
-- Added NVECTOR_MANYVECTOR module
-- Added SUNMATRIX_BAND module
-- Added SUNMATRIX_DENSE module
-- Added SUNMATRIX_SPARSE module
-- Added SUNLINSOL_BAND module
-- Added SUNLINSOL_DENSE module
-- Added SUNLINSOL_PCG module
-- Added SUNLINSOL_SPBCGS module
-- Added SUNLINSOL_SPFGMR module
-- Added SUNLINSOL_SPGMR module
-- Added SUNLINSOL_SPTFQMR module
-- Added SUNLINSOL_LAPACKBAND module
-- Added SUNLINSOL_LAPACKDENSE module
-- Added SUNNONLINSOL_NEWTON module
-- Added SUNNONLINSOL_FIXEDPOINT module
-- Added CVODE module
-- Added CVODES module
-- Added IDA module
-- Added IDAS module
-- Added KINSOL module
[... 2494 lines cut ...]
      Warning: Fortran 2018 deleted feature: DO termination statement which is not END DO or CONTINUE with label 440 at (1)
      scikits/odes/daepack/stodi.f:353:72:
      
        353 |         do 470 i = 1,n
            |                                                                        1
      Warning: Fortran 2018 deleted feature: Shared DO termination label 470 at (1)
      scikits/odes/daepack/stodi.f:354:72:
      
        354 |  470      yh(i,j) = yh(i,j) + eljh*acor(i)
            |                                                                        1
      Warning: Fortran 2018 deleted feature: DO termination statement which is not END DO or CONTINUE with label 470 at (1)
      scikits/odes/daepack/stodi.f:360:72:
      
        360 |  490    yh(i,lmax) = acor(i)
            |                                                                        1
      Warning: Fortran 2018 deleted feature: DO termination statement which is not END DO or CONTINUE with label 490 at (1)
      scikits/odes/daepack/stodi.f:376:72:
      
        376 |  510      yh1(i) = yh1(i) - yh1(i+nyh)
            |                                                                        1
      Warning: Fortran 2018 deleted feature: DO termination statement which is not END DO or CONTINUE with label 510 at (1)
      scikits/odes/daepack/stodi.f:396:72:
      
        396 |  530    savf(i) = acor(i) - yh(i,lmax)
            |

HOLL95, 2022-04-07:
Apologies that this is gigantic, although you're correct that the error is different using pip and pybamm_install_odes

Saransh-cpp, 2022-04-07:
No worries! Which Fortran compiler are you using? I see a warning in your logs which I don't see locally when using `gfortran` -
```
-- Using int32_t for indices
CMake Warning at config/SundialsCMakeMacros.cmake:83 (message):
  ------------------------------------------------------------------------

  WARNING: Fortran compiler does not support F90

  F90 support will not be provided

  ------------------------------------------------------------------------
```

Saransh-cpp, 2022-04-07:
Adding my logs here -

<details>
<summary>Logs:</summary>

```console
┌──(saransh㉿LAPTOP-V7LNSS4H)-[~]
└─$ pybamm_install_odes
/home/saransh/.local/lib/python3.9/site-packages/pybamm/scikits_odes_setup.log
2022-04-07 21:38:43,528 - [INFO] install_odes.main(125): Starting scikits.odes setup
2022-04-07 21:38:43,529 - [INFO] install_odes.main(147): Looking for sundials at /home/saransh/.local
2022-04-07 21:38:43,529 - [INFO] install_odes.main(147): Looking for sundials at /usr/local
2022-04-07 21:38:43,529 - [INFO] install_odes.main(147): Looking for sundials at /usr
2022-04-07 21:38:43,529 - [INFO] install_odes.main(157): Could not find sundials libraries.
2022-04-07 21:38:43,530 - [INFO] install_odes.main(158): Installing sundials in /home/saransh/.local
cmake version 3.22.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).
2022-04-07 21:38:43,640 - [INFO] install_odes.install_sundials(49): Downloading sundials
100% [........................................................................] 18496945 / 18496945---------- Running CMake prepare ----------------------------------------
CMake Warning at config/SundialsCMakeMacros.cmake:83 (message):
  ------------------------------------------------------------------------

  WARNING: EXAMPLES_ENABLE is deprecated and will be removed in the future.

  Copying value to EXAMPLES_ENABLE_C.

  ------------------------------------------------------------------------
Call Stack (most recent call first):
  config/SundialsDeprecated.cmake:20 (print_warning)
  config/SundialsDeprecated.cmake:25 (print_deprecated)
  CMakeLists.txt:166 (include)

-- Using int32_t for indices
-- Determining Fortran name-mangling scheme... OK
-- Looking for Fortran sgemm
-- Looking for Fortran sgemm - found
-- Found BLAS: /usr/lib/x86_64-linux-gnu/libopenblas.so
-- Looking for Fortran cheev
-- Looking for Fortran cheev - found
-- Found LAPACK: /usr/lib/x86_64-linux-gnu/libopenblas.so;-lpthread;-lm;-ldl
-- Looking for LAPACK libraries... OK
CMake Deprecation Warning at /home/saransh/.local/lib/python3.9/site-packages/sundials/build_sundials/LapackTest/CMakeLists.txt:1 (CMAKE_MINIMUM_REQUIRED):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Checking if LAPACK works... OK
-- Added NVECTOR_SERIAL module
-- Added NVECTOR_MANYVECTOR module
-- Added SUNMATRIX_BAND module
-- Added SUNMATRIX_DENSE module
-- Added SUNMATRIX_SPARSE module
-- Added SUNLINSOL_BAND module
-- Added SUNLINSOL_DENSE module
-- Added SUNLINSOL_PCG module
-- Added SUNLINSOL_SPBCGS module
-- Added SUNLINSOL_SPFGMR module
-- Added SUNLINSOL_SPGMR module
-- Added SUNLINSOL_SPTFQMR module
-- Added SUNLINSOL_LAPACKBAND module
-- Added SUNLINSOL_LAPACKDENSE module
-- Added SUNNONLINSOL_NEWTON module
-- Added SUNNONLINSOL_FIXEDPOINT module
-- Added CVODE module
-- Added CVODES module
-- Added IDA module
-- Added IDAS module
[... 797 lines cut ...]
-- Installing: /home/saransh/.local/lib/libsundials_kinsol.so
-- Installing: /home/saransh/.local/include/kinsol/kinsol.h
-- Installing: /home/saransh/.local/include/kinsol/kinsol_bbdpre.h
-- Installing: /home/saransh/.local/include/kinsol/kinsol_direct.h
-- Installing: /home/saransh/.local/include/kinsol/kinsol_ls.h
-- Installing: /home/saransh/.local/include/kinsol/kinsol_spils.h
Adding /home/saransh/.local/lib to LD_LIBRARY_PATH in /home/saransh/.bashrc
Defaulting to user installation because normal site-packages is not writeable
Collecting scikits.odes
  Using cached scikits.odes-2.6.3.tar.gz (211 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: scipy in ./.local/lib/python3.9/site-packages (from scikits.odes) (1.8.0)
Requirement already satisfied: numpy<1.25.0,>=1.17.3 in ./.local/lib/python3.9/site-packages (from scipy->scikits.odes) (1.22.3)
Building wheels for collected packages: scikits.odes
  Building wheel for scikits.odes (pyproject.toml) ... done
  Created wheel for scikits.odes: filename=scikits.odes-2.6.3-cp39-cp39-linux_x86_64.whl size=3453616 sha256=f737ecef6b4fec11fc2b5c07b60f57f63a61739812a9fc05a96288421c799589
  Stored in directory: /home/saransh/.cache/pip/wheels/ed/7f/b3/5b2cb0061e37013fc64daa92f47b56b07039e2760ae13ecc26
Successfully built scikits.odes
Installing collected packages: scikits.odes
Successfully installed scikits.odes-2.6.3
```

</details>

HOLL95, 2022-04-11:
Ok, this has been resolved. There were files necessary for compilation in both ```lib```, and ```lib64```, and so running the following (hack) from within ```lib```

```ln -s ../lib64/*.```
Allowed the install to go ahead (once ```install_odes.py``` had been modified with a ```--user``` flag)