pybamm-team/PyBaMM #2028: Use `deprecation` package for deprecation warnings

brosaplanella, 2022-04-20:
Replace existing deprecation warnings/errors with [`deprecation` package](https://pypi.org/project/deprecation/#description) so tests fail after a few releases.

brosaplanella, 2022-04-25:
I have been looking into this and I have a (probably very basic) question. The `deprecation` package uses decorators and it seems designed to deprecate entire methods directly. How should I use them when we want to deprecate either a few lines of code, or some arguments? For an example, see l84 of [simulation.py](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/simulation.py).

Saransh-cpp, 2022-06-03:
The [deprecat](https://github.com/mjhajharia/deprecat) package (built on top of the `deprecate` package) allows us to put deprecation warnings for arguments - https://github.com/mjhajharia/deprecat#deprecated-kwargs. Not sure how we can deprecate some particular lines of code.

agriyakhetarpal, 2023-03-10:
@brosaplanella I was looking through some solutions for this and while I couldn't find anything to deprecate some lines of code, there are indeed ways for deprecating arguments: in addition to [deprecat](https://github.com/mjhajharia/deprecat#deprecated-kwargs) mentioned by @Saransh-cpp, there's also [pyDeprecate](https://github.com/Borda/pyDeprecate) – a similar library with many more customisations available, applicable on both classes and methods along with forwarding inputs to newer arguments.  

Both of them will come at the cost of an extra dependency, so here is a purely Pythonic way which could perhaps be implemented as helper functions in `pybamm/doc_utils.py`? (see #2597). It transfers parameter values too but can be modified to just raise a warning on deprecation: https://gist.github.com/ZoomQuiet/f0113711181750e56ddfb51756d5c7fc

Saransh-cpp, 2023-03-14:
`pyDeprecate` looks really nice. We could also implement a new helper function, but having an extra dependency doesn't sound too bad to me (if it fulfills all our requirements).

RohitP2005, 2025-01-07:
hey @Saransh-cpp  i would like to work on it , if this issue isnt resolved yet .

Saransh-cpp, 2025-01-07:
HI @RohitP2005, please feel free to take this up. You might have to experiment a bit with the solutions/deprecation packages you find on the internet.

RohitP2005, 2025-01-08:
Hey @Saransh-cpp, I think I might have found a purely Pythonic way, suggested by @agriyakhetarpal, to handle those deprecation warnings using the warnings module: https://docs.python.org/3/library/warnings.html

### Example:

```import warnings

def old_function(x, y):
    warnings.warn(
        "old_function is deprecated and will be removed in future versions. Use new_function instead.",
        DeprecationWarning,
        stacklevel=2
    )
    return x + y

def new_function(x, y):
    return x + y
```

It also supports argument deprecation:

### Example:
```def some_function(a, b, deprecated_arg=None):
    if deprecated_arg is not None:
        warnings.warn(
            "The 'deprecated_arg' argument is deprecated and will be removed in future versions.",
            DeprecationWarning,
            stacklevel=2
        )
    # Function implementation
    return a + b
```
I also experimented with deprecating specific lines of code in a similar fashion:

### Example:
```import warnings

def example_function(x):
    # Line-level deprecation for a specific operation
    if x == 10:
        warnings.warn(
            "Using `x == 10` is deprecated and will be removed in future versions.",
            DeprecationWarning,
            stacklevel=2
        )
        # Deprecated operation
        result = x * 2
    else:
        result = x + 2

    return result
```

Kindly forgive me if I have mentioned anything irrelevant or incorrect. I'm still learning and appreciate any corrections or feedback!

agriyakhetarpal, 2025-01-08:
Thanks, @RohitP2005, but this is the standard approach, and we've been using it already. What we want here is to build on top of it, so that we can deprecate arguments to functions (or functions themselves), and the helper function to deprecate them can, additionally, handle the task of handling the deprecation warning automatically based on PyBaMM's version.

Our deprecation policy is for one year (three releases), but IIRC, we update PyBaMM's version in release branches only and not in the `develop` branch immediately after a release, and that will mean we will need to think about "tests fail after a few releases" requirement more carefully – since a deprecation warning implemented in version $$n$$ would fail only when we update to version $$n + 3$$, and that would happen in a release branch.

I would appreciate some more depth in the approach, but for now, Astropy's solution here: https://docs.astropy.org/en/stable/api/astropy.utils.decorators.deprecated_renamed_argument.html#deprecated-renamed-argument seems like a good approach to me. You could copy the code for this (and attribute it to to Astropy based on its license) as a start and adapt it for PyBaMM's codebase.

Another thing could be to turn `DeprecationWarning`s and other warnings into errors in our test suite – `pytest` can do that with the `-W` test flag, and is a common approach adopted by libraries.