pybamm-team/PyBaMM #2029: Dimensionality option

Mykle99, 2022-04-20:
### PyBaMM Version

22.3

### Python Version

3.9.12

### Describe the bug

When setting options:
```
options = {"dimensionality" : 1}
```
or

```
options = {"dimensionality" : 2}
```
and running:
```
custom_dfn = pybamm.lithium_ion.DFN(options = options)
```
there is a keyerror:
```
File pybamm\util.py:115, in FuzzyDict.__getitem__(self, key)
    109     raise KeyError(
    110         f"'{key}' not found. All lithium plating parameters have been "
    111         "renamed from '...negative electrode lithium plating...' "
    112         "to '...lithium plating...'"
    113     )
    114 best_matches = self.get_best_matches(key)
--> 115 raise KeyError(f"'{key}' not found. Best matches are {best_matches}")
```

### Steps to Reproduce

-

### Relevant log output

```shell
-
```

Mykle99, 2022-04-20:
Same goes for:

```
options = {"intercalation kinetics" : "asymmetric Butler-Volmer"}
options = {"intercalation kinetics" : "Marcus"}
options = {"intercalation kinetics" : "Marcus-Hush-Chidsey"}
```

`symmetric Butler-Volmer` & `linear `works

Mykle99, 2022-04-20:
and for:

```
options = {"loss of active material" : "reaction-driven"}
options = {"loss of active material" : "stress and reaction-driven"}
```

`reaction-driven` works

tinosulzer, 2022-04-20:
These all work fine for me. Please post your full script and error. It's probably a missing parameter.

tinosulzer, 2022-04-22:
Some of the parameter sets are missing parameters for some of the submodels. You can add them manually. See the error message at the bottom of the stacktrace.

Mykle99, 2022-04-22:
Ok, so the parameter_values I am using does not include all parameters needed for the submodels I want to use. 
And I would have to do something like this:
```
custom_param_set = pybamm.parameter_sets.Chen2020
custom_param_set["positive electrode"] = "nmc_ORegan2021"
```
Just for the parameters that are missing? Is there a simple way to see which ones are missing other than inspecting the stacktrace?

tinosulzer, 2022-04-22:
You can compare `model.print_parameter_info()` with `params.keys()`. And you can either update the custom param set or individual parameters like `params["..."] = ...`

closed 2022-04-22 by Mykle99