pybamm-team/PyBaMM #2032: local variable 'j_sei' referenced before assignment

Mykle99, 2022-04-22:
### PyBaMM Version

22.3

### Python Version

3.9.12

### Describe the bug

```
submodel_options = {
    "thermal" : "lumped",
    "cell geometry" : "pouch",
    "electrolyte conductivity" : "full",
    "particle" : "Fickian diffusion", # default
    "intercalation kinetics" : "asymmetric Butler-Volmer",
    "loss of active material" : "stress and reaction-driven",
    "SEI" : "solvent-diffusion limited",
    "SEI porosity change" : "true",
    "SEI film resistance" : "distributed", # default
    "particle mechanics" : "swelling and cracking",
    "lithium plating" : "irreversible"
}

dfn = pybamm.lithium_ion.DFN(options = submodel_options, build = False)
dfn.submodels["sei"] = pybamm.sei.SEIGrowth(dfn.param, "full electrode")

dfn.build_model()
```

```
UnboundLocalError                         Traceback (most recent call last)
Input In [55], in <cell line: 22>()
     19 dfn.submodels["sei"] = pybamm.sei.SEIGrowth(dfn.param, "full electrode")
     21 # finish by building model
---> 22 dfn.build_model()

File ~\AppData\Local\Programs\Python\Python39\lib\site-packages\pybamm\models\full_battery_models\base_battery_model.py:850, in BaseBatteryModel.build_model(self)
    847 if self._built_fundamental_and_external is False:
    848     self.build_fundamental_and_external()
--> 850 self.build_coupled_variables()
    852 self.build_model_equations()
    854 pybamm.logger.debug("Setting voltage variables ({})".format(self.name))

File ~\AppData\Local\Programs\Python\Python39\lib\site-packages\pybamm\models\full_battery_models\base_battery_model.py:777, in BaseBatteryModel.build_coupled_variables(self)
    770 pybamm.logger.debug(
    771     "Getting coupled variables for {} submodel ({})".format(
    772         submodel_name, self.name
    773     )
    774 )
    775 try:
    776     self.variables.update(
--> 777         submodel.get_coupled_variables(self.variables)
    778     )
    779     submodels.remove(submodel_name)
    780 except KeyError as key:

File ~\AppData\Local\Programs\Python\Python39\lib\site-packages\pybamm\models\submodels\interface\sei\sei_growth.py:140, in SEIGrowth.get_coupled_variables(self, variables)
    137 else:
    138     alpha = 0.5
--> 140 j_inner = alpha * j_sei
    141 j_outer = (1 - alpha) * j_sei
    143 variables.update(self._get_standard_reaction_variables(j_inner, j_outer))

UnboundLocalError: local variable 'j_sei' referenced before assignment
```

### Steps to Reproduce

b

### Relevant log output

```shell
b
```

tinosulzer, 2022-04-23:
You need to do
```python
dfn.submodels["sei"] = pybamm.sei.SEIGrowth(dfn.param, "full electrode", dfn.options)
```
What are you trying to do here?

tinosulzer, 2022-04-23:
The reason I ask is because this code:
```python
dfn = pybamm.lithium_ion.DFN(options = submodel_options, build = False)
dfn.submodels["sei"] = pybamm.sei.SEIGrowth(dfn.param, "full electrode", dfn.options)
dfn.build_model()
```
gives the same behavior as this code
```python
dfn = pybamm.lithium_ion.DFN(options = submodel_options)
```
So I assume you're preparing your own SEI submodel or something?

Mykle99, 2022-04-23:
Thanks for the explanation :) I am just looking through all the functionality of PyBaMM atm. Do you have any good examples to when options can not be used and submodels have to be added/swapped with something like this `dfn.submodels["sei"] = pybamm.sei.SEIGrowth(dfn.param, "full electrode", dfn.options)` instead?

tinosulzer, 2022-04-25:
> Is it just for when you are defining your own custom submodels (which do not exist as options)?

Yes

> Which option changes electrolyte diffusion <pybamm.models.submodels.electrolyte_diffusion.full_diffusion.Full object at 0x0000027CF5007070> and negative (& positive) electrode potential <pybamm.models.submodels.electrode.ohm.full_ohm.Full object at 0x0000027CF6537AC0>?

There are no options to change those, since there are no other submodels for those