pybamm-team/PyBaMM #2039: [Bug]: parameter "Inner SEI reaction proportion" not used in any models

martinjrobins, 2022-04-27:
### PyBaMM Version

develop branch

### Python Version

3.8

### Describe the bug

The parameter "Inner SEI reaction proportion" appears in:

- [ ] `pybamm/input/parameters/lithium_ion/seis/example/parameters.csv`
- [ ] `pybamm/input/parameters/lithium_ion/seis/ramadass2004/parameters.csv`

but is not used anywhere else in pybamm. Perhaps it should be removed?

### Steps to Reproduce

n/a

### Relevant log output

```shell
n/a
```

DrSOKane, 2022-04-27:
Hi Martin, please keep this parameter. I am currently working on an SEI model that does use it.

tinosulzer, 2022-04-27:
The degradation parameters are a mess (some duplicate but named differently in different models, some values way off)

closed 2022-04-29 by martinjrobins