pybamm-team/PyBaMM #2065: [Bug]: Scaling by 2x in SEI growth?

bardsleypt, 2022-05-18:
### PyBaMM Version

22.3

### Python Version

3.9.12

### Describe the bug

I believe there may be a scaling bug present in the SEI growth model by a factor of 2x, at least when considering the 'solvent-diffusion limited' growth model for SEI. Following is my reasoning (apologies for clunky equations as it doesn't appear TeX is supported here).

As determined from PyBaMM source code, the ODE for the *dimensional* SEI growth takes the form

dL/dt = (1/2) (c_sol * D_sol * V_sei) / L

where, by default there is an L_inner, and L_outer which satisfy the same growth law. I am fine with this growth law, crucially the scaling by (1/2), **when there is only a single SEI layer considered**. My current understanding of the (1/2) factor is from the stoichiometry of the SEI reaction requiring 2 moles of lithium. 

However, **when there is a double-layer SEI considered** (which is the behavior in models/submodels/interface/sei/sei_growth.py)., it seems the law should read

dL/dt = (1/4) (c_sol * D_sol * V_se) / L

My understanding of this scaling by (1/4) is due to the same stoichiometric scaling of (1/2), and an additional scaling of (1/2) coming from assuming the growth of the SEI layers is the same for both inner/outer. Thus half of the SEI growth comes from the inner layer growth, and the other half from the outer layer growth.

I have verified that the scaling in both the L_inner and L_outer growth laws does use the 1/2 scaling, as opposed to the 1/4 scaling. Thus, it would seem the default PyBaMM SEI growth model implements a 2x speed up of the SEI layer growth. My reference for this scaling comes directly from one of the papers using the Chen2020_plating parameter set:

O’Kane, Simon E. J., et al. “Lithium-Ion Battery Degradation: How to Model It.” Physical Chemistry Chemical Physics, vol. 24, no. 13, 2022, pp. 7909–22. DOI.org (Crossref), https://doi.org/10.1039/D2CP00417H.

As well as the supplemental information for this article.

### Steps to Reproduce

[pybamm_sei_scaling.pdf](https://github.com/pybamm-team/PyBaMM/files/8720394/pybamm_sei_scaling.pdf)

PDF  is an exported jupyter notebook (with execution displays) with the following commands:

```python
import pybamm
import numpy as np

model = pybamm.lithium_ion.DFN({'SEI': 'solvent-diffusion limited', 
                                'SEI porosity change': 'true'})
params = pybamm.ParameterValues('Chen2020_plating')

dLdt = list(model.submodels['sei'].rhs.values())[0]
dLdt = params.process_symbol(dLdt)
dLdt.render()

def rhs_from_model(L_outer):
    A = dLdt.left.evaluate()
    B = dLdt.right.left.evaluate()
    nondim_rhs = A*B/L_outer
    return nondim_rhs

def rhs_from_okane(L_outer, scl=0.5, tau_s=1, L0=1):
    dim_rhs = params['Bulk solvent concentration [mol.m-3]']*params['Outer SEI solvent diffusivity [m2.s-1]']*\
        params['Outer SEI partial molar volume [m3.mol-1]'] / L_outer
    nondim_scl = tau_s / L0 / L0 # tau_s for time scaling, one copy L0 for LHS ODE, one copy for RHS ODE
    return scl*nondim_scl*dim_rhs

tau_s = params.evaluate(model.timescale)
L0 = params.evaluate(model.param.L_sei_0_dim)

model_rhs = rhs_from_model(np.logspace(-9, -8, 10))
model_okane2 = rhs_from_okane(np.logspace(-9, -8, 10), scl=0.5, tau_s=tau_s, L0=L0)
model_okane4 = rhs_from_okane(np.logspace(-9, -8, 10), scl=0.25, tau_s=tau_s, L0=L0)

relerr2 = np.linalg.norm(model_rhs - model_okane2) / np.linalg.norm(model_rhs)
relerr4 = np.linalg.norm(model_rhs - model_okane4) / np.linalg.norm(model_rhs)
print('Rel. Error (1/2 scaling) = {:.4g}'.format(relerr2))
print('Rel. Error (1/4 scaling) = {:.4g}'.format(relerr4))
```

bardsleypt, 2022-05-20:
I believe it will depend on the chemical reaction that is taking place in the SEI reaction (i.e., how many moles of Li are required to form one mole of SEI). Most of the literature I have encountered, derives the SEI reaction from Safari et al. [1], where they assume the SEI product is lithium ethylene dicarbonate (CH<sub>2</sub>OCO<sub>2</sub>Li)<sub>2</sub>, thus requiring 2 lithium ions to form the product. My understanding, is this requirement of 2 lithium ions for SEI is why the growth of the SEI layer uses 1/2 of the  SEI-lithium current density (see equation 12 in Safari et. al[1]). 

I assume if the SEI product includes only one mole of lithium, then the scaling is fine as-is. In that case, the equations in PyBaMM are fine as-is, but the diffusivity of the solvent (EC typically) in some of the parameter sets (e.g., Chen2020_plating) would be physically 2x faster than expected. 

Now for the caveat, I am not an electro-chemist by training. My training is in applied mathematics, so I can definitely interpret the differential equations, and the rates they imply. However, my understanding of the electro-chemical reactions is not expert-opinion. Thus, if you'd like (and in fact I recommend it) to get a second opinion on this, by all means do so. I'll be happy to further collaborate and share what I do/don't know.

Here are some of the references where I have encountered SEI growth, and thus my understanding of the 1/2 scaling due to stoichiometry. It seems we agree that the 1/2 present in PyBaMM accounts for the dual-layer SEI, and thus its a matter of resolving this possible issue with chemical diffusivities and reaction stoichiometry.

References

1. Safari, M., et al. “Multimodal Physics-Based Aging Model for Life Prediction of Li-Ion Batteries.” Journal of The Electrochemical Society, vol. 156, no. 3, 2009, p. A145. DOI.org (Crossref), https://doi.org/10.1149/1.3043429.
2. O’Kane, Simon E. J., et al. “Lithium-Ion Battery Degradation: How to Model It.” Physical Chemistry Chemical Physics, vol. 24, no. 13, 2022, pp. 7909–22. DOI.org (Crossref), https://doi.org/10.1039/D2CP00417H.
3. Yang, Xiao-Guang, et al. “Modeling of Lithium Plating Induced Aging of Lithium-Ion Batteries: Transition from Linear to Nonlinear Aging.” Journal of Power Sources, vol. 360, Aug. 2017, pp. 28–40. DOI.org (Crossref), https://doi.org/10.1016/j.jpowsour.2017.05.110.

brosaplanella, 2022-07-04:
I am not totally sure on how the inner/outer SEI models work (I have mostly worked with the EC reaction limited model). The best reference on such models is @Scottmar93 thesis (available [here](https://ora.ox.ac.uk/objects/uuid:8afdcc34-cc42-48ba-b316-96a6d0f33a45), and actually Scott might be able to provide some more insight).

As I understand, the `alpha` coefficient in `j_inner` and `j_outer` should take care of that. Isn't that the case?

tinosulzer, 2022-07-05:
Yes the inner / outer is fine. The question is whether there should be an extra factor of 2 because each mole of SEI contains two moles of lithium. We currently have this for the EC reaction limited model, but not for the others.

bardsleypt, 2022-07-05:
Yes, the inner / outer looks fine to me as well. To try and clarify, using the 'sovlent-diffusion limited' model (I haven't checked the scalings for the other modes), and then backing out the PyBaMM non-dimensionalizations, I end up with two dimensional DEs describing the growth of the SEI layer:

$$
\begin{aligned}
\frac{\partial L_\text{inner}}{\partial t}& = \frac{1}{2}\frac{c_\text{bulk}D_\text{sei}V_\text{sei}}{L_\text{outer}}\\
\frac{\partial L_\text{outer}}{\partial t}& = \frac{1}{2}\frac{c_\text{bulk}D_\text{sei}V_\text{sei}}{L_\text{outer}}\\
\end{aligned}
$$

where, as I understand it, the 1/2 scalings here come exactly from the `alpha` used here: https://github.com/pybamm-team/PyBaMM/blob/212c306c0a1c55638cb77a25f31be08a2214e953/pybamm/models/submodels/interface/sei/sei_growth.py#L138

This gives the dynamics for the overall SEI layer thickness $L_\text{inner} + L_\text{outer}$ to be

$$
\frac{\partial (L_\text{inner} + L_\text{outer})}{\partial t} = \frac{c_\text{bulk}D_\text{sei}V_\text{sei}}{L_\text{outer}}.\\
$$

However, when comparing with the forms of SEI growth in the above listed references (see in particular the supplementary info in this article: https://pubs.rsc.org/en/content/articlelanding/2022/cp/d2cp00417h), there would ultimately be a missing factor of 1/2. I believe, though I'm not certain, this additional scaling factor of 1/2 comes from the stoichiometric coefficient of lithium in the SEI reaction. 

If this is indeed a bug and I didn't miss some other scaling elsewhere, an easy workaround is to simply adjust the solvent diffusivity (I guess bulk solvent concentration may work as well) by a factor of 1/2. Alternatively, this could be interpreted as the diffusivities kept in the PyBaMM parameter sets would be scaled by 2x. 

Hopefully this helps clear up my question on this issue, but let me know if additional comments are necessary. And thanks for everyone's time on this already.

Best,
Patrick

Scottmar93, 2022-08-06:
Hey my apologies for seeing this late but hopefully I can still be helpful. Thank you for tracking down this issue and looking into it deeply as it is really helpful for improving PyBaMM! 

### Source of the difference
I agree with you that the source of this difference is the stoichiometric coefficient of lithium in the SEI reaction. Looking back on some of my [old notes (see sections 1.1 and 1.3)](https://github.com/pybamm-team/PyBaMM/files/9275299/SEI_Models.pdf) I used a reaction of the form: $\text{Li}^+ + \text{Sol} + \text{e}^{-} \rightarrow  \text{SEI}$. This is different to the more typically used $2\text{Li}^+ + 2\text{Sol} + 2\text{e}^{-} \rightarrow  \text{SEI}$. My justification was that there are many compounds in the SEI and I felt the exact reaction was a little unclear so I chose to just use the simplest one. I also thought that any such model will need to be fitted to data so this stoichiometry coefficient would also be absorbed into the diffusion fitting parameter. As you say this can be seen by comparing the prefactors in our implementation to those in (S16) and (S17) of the supplementary material of the paper that you shared. 

### Way forward
Going forward I think it is probably best to use the more standard reaction used in the literature. So as you suggest adding an additional factor of two somewhere would help prevent errors if someone takes some SEI parameters from the literature. I am also not against refactoring the SEI models to not use the two-layer structure if there is a feeling that this overcomplicates things and makes the models less standard and easy to use. 

### Additional differences
I think you already touched upon this but in addition to the stoichiometric factor, there is also a second difference between our implementation and equation (6) of the main paper you shared https://pubs.rsc.org/en/content/articlelanding/2022/cp/d2cp00417h which for reference is:
$$\frac{dL_{sei}}{dt} = \frac{c_{sol}D_{sol}V_{sei}}{2L_{sei}}$$
Having the two-layer structure also introduces a different prefactor. By setting $\alpha=0.5$ in the two-layer SEI model, only half of the mass produced by the SEI reaction goes into growing the outer layer. Therefore a two-layer SEI model should grow faster than a single-layer model since the solvent is only limited by the growth of the outer layer. This can be seen by considering the case where initially the inner and outer layers are the same thickness so we can take $L_{SEI}=2L_{outer}$ in the equations in the supplementary material of the paper you shared https://pubs.rsc.org/en/content/articlelanding/2022/cp/d2cp00417h. Doing this, I get
$$\frac{dL_{sei}}{dt} = \frac{c_{sol}D_{sol}V_{sei}}{L_{sei}}$$
so the growth is twice as fast as a single-layer model in this case. To recover the single-layer model though one could set $\alpha=0$ and set the SEI to consist completely of outer SEI initially. 

Hopefully, this is helpful and I am happy to chat if there are any other questions. 

Best, 
Scott

DrSOKane, 2022-08-08:
Hi all,

Having thought about this long and hard, it's a genuine bug. How to fix it?

The simplest fix would be to divide by 2 in SEIgrowth.set_rhs, then change the variable li_mols_per_sei_mols in BaseSEI._get_standard_concentration_variables from 1 to 2 in all cases, not just EC reaction limited.

However, my paper and at least one other degradation paper were made with this bug in place, so there needs to be a setting to be able to reproduce the flawed data. I therefore propose that was make li_mols_per_sei_mols a user-defined parameter. This also opens up scope for future upgrades to include SEI compounds beyond LEDC or Li2CO3.

If the moderators agree this is the best approach, I can implement it immediately.

Many thanks, Simon

bardsleypt, 2022-08-08:
Hi all, thank you Scott and Simon for your thorough responses. It is especially useful knowing the genesis of the scaling issue comes from implementing a different SEI reaction. Regarding an overall fix, I am in favor of what Simon proposes: an immediate fix by dividing by the stoichiometric coefficient. My suggestion would be to plumb in a new parameter from the top-level down, something of the flavor "stoichiometric coefficient in SEI reaction", which gets a default value of 2 but can be overridden to a value of 1 to recover legacy results. I am happy to assist in this (or any other decided upon) change as well.

Best,
Patrick

tinosulzer, 2022-08-08:
Sounds good, go for it

Scottmar93, 2022-08-12:
I also agree that this sounds like a good way forward for recovering previous results and allowing the default behaviour to be in line with the majority of the literature. 

My apologies to Simon that I didn't document this implementation so well, resulting in the mentioned errors in the papers.

DrSOKane, 2022-08-12:
Please don't worry. The paper itself points out that a factor of two difference, whether it be from having one vs. two layers or the valancy of the SEI compound(s), is negligible compared to the reported range of the SEI growth rate itself.

PR #2222 (2022-08-15): Added new parameter z_sei to allow the user to set SEI valancy
# Description

A new parameter "Ratio of lithium moles to SEI moles" (short name z_sei) has been added. This variable was previously hardcoded to 1 for all SEI growth options except "ec reaction limited". As pointed out in issue #2065, z_sei=1 is incorrect for the two most common SEI compoints - LEDC and Li2CO3, both of which require z=2.

The SEI parameter sets have been updated accordingly. The new parameter z_sei is set to 2 by default for all parameter sets except OKane2022, which used the old code for which it was set to 1.

F [... cut]
closed 2022-08-15 by tinosulzer